esp8266/Arduino #2724: AP -> STA server problem

ChrisXenon, 2016-12-02:
I'm working on ESP826-12 in Arduino IDE.
I am implementing a WiFi management system. As usual, initially a softAP is created which the user can interact with to set access point SSID and password. These are saved in EEPROM and connection in STA mode attempted. If it connects OK then the AP is disconnected and the STA server is in charge.

It all works, except that when the AP disconnect code has run and the new STA server is running, the old AP pages are still being served, and where names coincide, the AP server tales preference. There is code to close the AP and server, but it's not working. The three relevant routines are below.

void startAccessPoint()
{ Serial.println("Starting access point.");
  Serial.setDebugOutput(true);
//  Serial.setDebugOutput(false);
  APList = listAccessPoints();
  WiFi.mode(WIFI_AP_STA);
  if   (WiFi.softAP(APssid, APpassword, strongestChannel))
       {IPAddress myIP = WiFi.softAPIP();
        Serial.print("AP IP address: ");
        Serial.println(myIP);
        server.on("/",          APhandleRoot      );
        server.on("/root",      APhandleRoot      );
        server.on("/password",  APhandlePassword  );
        server.on("/connect",   APhandleConnect   );
        server.on("/server",    APhandleServer    );
        server.on("/notlisted", APhandleNotlisted );
        server.on("/disconnect",APhandleDisconnect);
        server.begin();
        Serial.println("Access point started.");
        Serial.println(listAccessPoints()); 
       }
  else {Serial.println("SoftAP failed to start.");
       }
}

void APhandleDisconnect()
{ // re-connect but this time disconnect the AP
  WIFIconnect(WIFInameChars, WIFIpasswordChars, true);
}

boolean WIFIconnect(String APname, String APpassword, boolean disconnectAP)
{Serial.print("WIFIconnect: ");
if  (disconnectAP)
     {Serial.println("Disconnecting AP");
      WiFi.softAPdisconnect(true);
      server.close();
      WiFi.disconnect();
     };

// WiFi.mode(WIFI_STA);

 APname.toCharArray(WIFInameChars,31);
 APpassword.toCharArray(WIFIpasswordChars,31);
 Serial.print(WIFInameChars); Serial.print(", ");
 Serial.println(WIFIpasswordChars);
 
 WiFi.begin(WIFInameChars, WIFIpasswordChars);
 //WiFi.config(ip, gateway, subnet);
 
 for (int i=20; i>0; i--)
     {if   (WiFi.status() == WL_CONNECTED)
           {Serial.println("Connected in station mode. Start web server.");
            server.on("/", handleRoot);
            server.on("/notlisted", handleNotlisted);
            server.begin();
            return true;
           }
      else {delay(1000);
            Serial.print(i);Serial.print(". ");
           }
     }
 Serial.println("\nFailed to connect in station mode");
 return false;
};

tablatronix, 2017-02-20:
try this instead

```
WiFi.softAPdisconnect(false);
WiFi.mode(STA);
```

ChrisXenon, 2017-02-21:
Thanks, I'll give that a go.

tablatronix, 2017-02-21:
Add code blocks to your code

ChrisXenon, 2017-02-21:
I have no idea what that might mean.

tablatronix, 2017-02-21:
Your code you pasted is not in code blocks or formatted, so is hard to read.
add the markdown code fences around it.

https://guides.github.com/features/mastering-markdown/

ChrisXenon, 2017-02-22:
Oh I see. I'll do that next time. Thanks again.

tablatronix, 2017-02-22:
I meant edit your post, so we can check your code, or duplicate it.

devyte, 2017-10-06:
@ChrisXenon is this issue still valid? did @tablatronix 's suggestion work for you?

ChrisXenon, 2017-10-07:
The suggestion didn't fix it.

devyte, 2017-10-07:
Ok, please provide a MCVE.

ChrisXenon, 2017-10-07:
It's nearly a year old and ancient history for me. I don't know what a MCVE is and I no longer have the code anyway.

devyte, 2017-10-07:
MCVE: minimal complete verifiable example. If you google it, you'll find a more in-depth explanatiion.
If it's ancient history, and you don't have the code anymore, I will close. If you think the problem is still relevant, please open a new issue and provide the details requested in the isssue template, so that somebody can look into it.
Closing due to lack of info.

closed 2017-10-07 by devyte