esp8266/Arduino #2725: Nodemcu reading encoder and transmitting signal via WiFi to another nodemcu

aletrifo, 2016-12-02:
Good morning everyone as said in the title i'm programming a Nodemcu to communicate with another one the position of the encoder, i took an example sketch and modified it. It all goes well until i move the encoder very slowly, if i move it "fast" i get the message displayed under. I think that the problem here is the interrupt routine that disturbs the WiFi part of the code but i'm not an expert on coding so i don't know how to solve it.
If you please give me some advice it would be awesome.
Thanks 
Alex

#### Hardware
Hardware:			Nodemcu v1+Quadrature encoder 2048cpr
Core Version:      	2.3.0

### Description

Problem description

### Settings in IDE

Module:  			Nodemcu v1
Flash Size: 		4MB
CPU Frequency:		80Mhz
Flash Mode:			ArduinoIDE
Flash Frequency:	Baud 115200 
Upload Using:		SERIAL
Reset Method:		nodemcu

### Sketch
```
  #include <Wire.h>           // To Be Able To Use Sensors Like IMU9050
  #include <EEPROM.h>         // To Be Able To Store & Read Data Between Power Off
  #include <ESP8266WiFi.h>    // The Basic Function Of The ESP NOD MCU
  #include <Encoder.h>        // Encoder library
  #define       LED0      2         
  #define       LED1      D0        
  Encoder knobLeft(5, 4);
  int           LEDState          = LOW;
  unsigned long CurrMillis        = 0;
  unsigned long PrevMillis        = 0;
  unsigned long Interval          = 1000;
  char*         TKDssid;           
  char*         TKDpassword;        
  int             TKDServerPort  = 9001;
  IPAddress       TKDServer(192,168,4,1);
  WiFiClient      TKDClient;

  void setup() 
  {
    Serial.begin(115200);           // Computer Communication
    pinMode(LED0, OUTPUT);          // WIFI OnBoard LED Light
    pinMode(LED1, OUTPUT);          // Indicate Encoder moving #1
    digitalWrite(LED0, !LOW);       // Turn WiFi LED Off
    Serial.println("\nI/O Pins Modes Set .... Done");
    WiFi.mode(WIFI_STA); // To Avoid Broadcasting An SSID
    WiFi.persistent(false);
    WiFi.begin("GOLIATH");          // The SSID That We Want To Connect To
    Serial.println("!--- Connecting To " + WiFi.SSID() + " ---!");
    CheckWiFiConnectivity();        // Checking For Connection
  }
  long positionLeft  = -999;
  long positionRight = -999;  
  long newLeft;
  float giri;
  String Message;
  long timer=millis();
  int wait=500;
  
  void loop()
  {
    newLeft = knobLeft.read();
    giri=float(newLeft)/2048;  
    positionLeft = newLeft;
    if ((millis()-timer)>wait){
      Serial.println(newLeft);
      TKDClient.print("T1 E");
      TKDClient.println(newLeft);
      Message = TKDClient.readStringUntil('\r');
      Serial.println(Message);
      TKDClient.flush();
      timer=millis();
    }
  }
  void CheckWiFiConnectivity()
  {
    while(WiFi.status() != WL_CONNECTED)
    {
      for(int i=0; i < 10; i++)
      {
        digitalWrite(LED0, !HIGH);
        delay(250);
        digitalWrite(LED0, !LOW);
        delay(250);
        Serial.print(".");
      }
      Serial.println("");
    }
    digitalWrite(LED0, !HIGH);
    Serial.println("!-- Client Device Connected --!");
    Serial.println("Connected To      : " + String(WiFi.SSID()));
    Serial.println("Signal Strenght   : " + String(WiFi.RSSI()) + " dBm");
    Serial.print  ("Server IP Address : ");
    Serial.println(TKDServer);
    Serial.print  ("Device IP Address : ");
    Serial.println(WiFi.localIP());
    TKDRequest();
  }        
  void TKDRequest()
  {
    TKDClient.stop();
    if(TKDClient.connect(TKDServer, TKDServerPort))
    {
      Serial.println    ("< Tower 1 connected >");
      TKDClient.println ("< Tower 1 connected >");
    }
  }

```

### Debug Messages

```
0x40106b58: interrupt_handler at C:\Users\ale\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_wiring_digital.c line 119
0x401012bd: ppEnqueueRxq at ?? line ?
0x40100f73: ppProcessTxQ at ?? line ?
0x40106b20: interrupt_handler at C:\Users\ale\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_wiring_digital.c line 113
0x4022f269: pm_open_rf at ?? line ?
0x40106bbc: interrupt_handler at C:\Users\ale\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_wiring_digital.c line 135
0x40106b20: interrupt_handler at C:\Users\ale\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_wiring_digital.c line 113
0x40106ba6: interrupt_handler at C:\Users\ale\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_wiring_digital.c line 131
0x40105101: ets_timer_disarm at ?? line ?
0x40106b20: interrupt_handler at C:\Users\ale\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_wiring_digital.c line 113
0x402086ac: pm_open at ?? line ?
0x40105666: spi_flash_read at ?? line ?
0x40106ff8: pvPortZalloc at C:\Users\ale\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/heap.c line 33
0x4020817d: pm_set_sleep_type_from_upper at ?? line ?
0x40208612: pm_shutdown at ?? line ?
0x402240e0: udp_input at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/core/udp.c line 360
0x402086bf: pm_open at ?? line ?
0x40219da1: system_restart_local at ?? line ?
0x40219de6: system_restart at ?? line ?
```

devyte, 2017-10-06:
@aletrifo is this issue still valid with latest git? have you tried different pins?

devyte, 2017-12-29:
Closing due to no feedback in over 2 months, and age.

closed 2017-12-29 by devyte