esp8266/Arduino #2728: Dependency of soft AP DHCP addresses on selected debug level, 'Generic ESP8266 Module'

mrwgx3, 2016-12-03:
### Basic Infos

#### Hardware

```
Hardware:     ESP12, Adafruit Huzzah Breakout and Feather Boards  
Core Version: 2.3.0
```

#### OS, IDE, and SDK

``` 
OS:   Windows 10 Professional and Linux Ubuntu 16.04LTS   
IDE:  1.6.12  
SDK:  1.5.3_16_04_18
```

### Description

While in AP+STA mode, it was observed that DHCP addresses assigned by the AP side sometimes differed from those specified by the [network parameters passed in Wifi.softAPConfig()][wifi_softAPConfig]. These variations were traced to the [debug level][debugging_md] chosen for the 'Generic ESP8266 Module' board setting, with erroneous addresses being generated for levels **not** containing a 'Wifi' entry. Incorrect addresses were also seen for the 'Adafruit HUZZAH ESP8266' board setting, which doesn't expose any debug levels at all.

Further tests revealed that when a [specific DEBUG\_WIFI() call is NOT made within Wifi.softAPConfig()][softAP_crit_debugwifi], the results of a [puzzling (perhaps incomplete) DCHP address calculation][softAP_addrcalc] are used. With the 'Wifi' flag set, however, the critical DEBUG_WIFI() call is made, and the correct DHCP addresses are somehow calculated and set; _where and how is not readily apparent_.

Also noted was that some address ranges caused Wifi.softAPConfig() to fail due to the calculated starting/ending DHCP address range being inverted (end > start); an example is included in the attached sketch.

Additional items:

* Tried the same tests for just AP mode; the results were the same.
* The title for library file [ESP8266WiFiAP.cpp][title_line] should be changed from "ESP8266WiFiSTA.cpp" to "ESP8266WiFiAP.cpp".

[debugging_md]: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.md  "Debug Features Tutorial"
[wifi_softAPConfig]: https://github.com/esp8266/Arduino/blob/2.3.0/libraries/ESP8266WiFi/src/ESP8266WiFiAP.cpp#L173#L179  "Sets IP, Gateway, Subnet"
[softAP_crit_debugwifi]: https://github.com/esp8266/Arduino/blob/2.3.0/libraries/ESP8266WiFi/src/ESP8266WiFiAP.cpp#L180  "1st DEBUG_WIFI()"
[softAP_addrcalc]: https://github.com/esp8266/Arduino/blob/2.3.0/libraries/ESP8266WiFi/src/ESP8266WiFiAP.cpp#L202#L215  "DCHP address calculation"
[title_line]: https://github.com/esp8266/Arduino/blob/2.3.0/libraries/ESP8266WiFi/src/ESP8266WiFiAP.cpp#L2 "Incorrect title line"

### Settings in IDE

```
Module:          Adafruit HUZZAH ESP8266  
Flash Size:      4MB (3M SPIFFS)  
CPU Frequency:   80Mhz  
Debug Settings:  None presented by IDE  
```

```
Module:          Generic ESP8266 Module  
Flash Size:      4MB (3M SPIFFS)  
CPU Frequency:   80Mhz  
Flash Mode:      qio  
Flash Frequency: 40Mhz  
Upload Using:    Disabled or SERIAL  
Reset Method:    nodemcu  
Debug Port:      None or Serial
Debug Level:     None, Core, or Core+Wifi
```

### Sketch Outline

```
The sketch included at the end of this post is used to:
1. Place the ESP12 in AP + STA mode
2. Configure and start the AP side
3. Statically connect to the home network on the STA side
4. Start a minimal web server to check for connectivity
```

### History

This issue was noticed while developing the attached sketch in 'Generic ESP8266 / Core+Wifi' mode. The debug messages from the soft-AP configuration and startup are:
[... 300 lines cut ...]

//    No network connect
      Serial.println ( "No STA connection " );

   } //end-else

// DEBUG: Set auto-connect flags
  WiFi.setAutoConnect( true );
  WiFi.setAutoReconnect( true );

// Connect handlers and start server
	server.on ( "/", handleRoot );
	server.begin();
	Serial.println ( "HTTP server started" );
 
} //setup

//---------------------------------------------------------------------------
void loop ( void ) {
	server.handleClient();
} //loop

//---------------------------------------------------------------------------
//---------------------------------------------------------------------------
```

mrwgx3, 2016-12-03:
Errata: In _Description_,  'DHCP address range being inverted (end > start)' should read 'DHCP address range being inverted (start > end)'

eketjall, 2016-12-21:
I have noticed that even when I set Wifi.softAPConfig() before softAP mode is started (I have it in STA mode before that), the debug information states that the dhcp server is started with ip 192.168.4.1 even if I set a different IP in the config.
The error seems to be only in the debug message though. The dhcp server is in fact started at the adress specified in the softAPConfig.

tablatronix, 2017-02-20:
The sdk docs clearly state
> This API can be called only if the ESP8266 SoftAP is enabled. 

Not sure how the debug comes into play, but  I had similar issues with softap_disconnect, until I read the same disclaimer on that also.

devyte, 2017-10-06:
@mrwgx3 I can't reproduce this behavior with latest git, but then again I'm not sure I fully understand the entire description. 
Is this issue still valid?

mrwgx3, 2017-10-07:
Hi devyte
```
@mrwgx3 I can't reproduce this behavior with latest git,
but then again I'm not sure I fully understand the entire description.
Is this issue still valid?
```
Yes, I believe so, but we both need to get on the same page to crosscheck results. Again here are the tools and part I tested with; please list yours:
```
Hardware:     ESP12, Adafruit Huzzah Breakout and Feather Boards  
Core Version: 2.3.0
OS:   Windows 10 Professional and Linux Ubuntu 16.04LTS   
IDE:  1.6.12  
SDK:  1.5.3_16_04_18
```
Next, could you verify the STAT+AP initialization within 'setup()' show in my SKETCH codeblock: 
```
void setup ( void ) {
// LED and serial setup
// ......

  Serial.setDebugOutput(true);     // ? Cryptic WIFI info

// DEBUG: Disable auto-connect, then disconnect WIFI
  WiFi.setAutoConnect( false );
  WiFi.setAutoReconnect( false );
  WiFi.disconnect( true );           // Wifi OFF
  WiFi.softAPdisconnect( false );    // Wifi already OFF  

  Serial.println( "Configuring as AP+STA" );
  WiFi.mode( WIFI_AP_STA );

  Serial.println("Setting soft-AP configuration ... ");
  Serial.println( WiFi.softAPConfig(local_IP, gateway, subnet) ? "Ready" : "Failed!");

  Serial.println("Setting soft-AP ... ");
  bool ok = WiFi.softAP( apsf_ssid, apsf_password );
  Serial.println( ok ? "Ready" : "Failed!" );
//.....
} //setup
```
Aside from the concerns expressed by @eketjall  and @tablatronix about the ordering of the 'WiFi.softAPdisconnect()' statement, I see no issues (copy @igrr  as well)

I'll return later with further clarifications after I re-protoboard my hardware setup.

mrwgx3, 2017-10-30:
Note that in the original post, the 'debug level' link under the 'Description' header is broken; here is the [corrected link][debugging_md_ok].

[debugging_md_ok]: https://github.com/esp8266/Arduino/blob/2.3.0/doc/Troubleshooting/debugging.md  "Debug Features Tutorial, Corrected Link"

devyte, 2017-10-31:
@mrwgx3 I don't remember what I tested back when I said I couldn't reproduce it, and I also don't remember the details currently contained in the description.
Anyways, the SoftAP currently has dhcp parameters hardcoded, and it assumes that the served network is /24.
So yes, the start is the SoftAP IP +99, and the end is the SoftAP IP +199. If you specify an IP for the SoftAP >= 55, you have trouble, and same for the mask if it's not the default for a /24 network.
There was an attempt #3637 to improve this, but there were some obstacles encountered, and the merge was backed out.

About the debug message having an impact, I suspect you're facing some mem corruption, as it should have no impact whatsoever on the behavior.

You're welcome to give this a shot, I wouldn't mind seeing this improved.

mrwgx3, 2017-11-01:
Hi again, devyte

Got my hardware set back up, and finished working through my original notes. Surprisingly, I could reproduce my initial results using the Arduino IDE under Linux Ubuntu 16.04LTS, but not so running under Windows 10 Pro (baffling?!). To eliminate the possibility of an accidently modified SDK and/or ESD damaged hardware, I again re-tested using both the lastest Arduino IDE (version 1.8.5) and new ESP Huzzah hardware, again with the same results.

Now for some backstory...

My initial task was to connect and address some 30+ ESP's together for data collection purposes. Rather than jumping into the complexities of mesh-networking, I explored the possibility of chaining STA+AP configured ESP's into a statically addressed hierarchy. This required that I work in Linux in order to easily modify and recompile the required SDK libraries.

I turned on debugging using IDE's generic ESP settings and got things partially working, but everything broke when I turned debugging OFF. Testing via 'ifconfig' and 'ping' showed that the DCHP addressing being handed out by the ESP access point were nothing close to those requested, hence this issues post.

Now back to the present...

I agree with you that the status of the debug flags should make no difference on how the code performs. Apparently this is now the case for the IDE running under Windows 10; the code generates erroneous DCHP addresses regardless of debug flag state. This makes sense given what the code contains!

Running under Ubuntu Linux, however, the assigned DCHP address DO depend on debug flag status. Even more disturbing is that

* Correct addresses are calculated when debug 'printf' statements are enabled; how and where is this happening?
* One can cause the erroneous DCHP addresses to reappear by changing the number, hence argument depth of the debug printf statements being used.

This odd behavior does imply a memory corruption issue, but at how deep a level is it occuring at?

Given that...

* The debug printf statements seem to ultimately call a 'vnsprint()' function in the closed SDK, and
* The interrupt intensive nature of the Wifi stack,

I suspect that only detailed knowledge of the closed SDK plus a true hardware emulator will resolve the problem.
I think this issue should be forwarded onto Espressif, as it potentially impacts so many code modules.

schlaegerz, 2018-02-03:
I am having the same issue, but I have tried the same code on multiple different ESP-12E chips, and it will work without fail on some of them, on others it seems to break very randomly, and it appears more likely to work the more logging I add to my sketch, which makes this very hard to figure out.

IMAN4K, 2018-04-08:
The same failure for me.
I got managed to fix this by moving my AP config code section to main loop (was inside a task and callback)
Have no idea about the reason!