esp8266/Arduino #2729: telnet example is not telnet -- but just serial / wifi

blacktigersoftware, 2016-12-03:
### Basic Infos

#### Hardware
Hardware:			?ESP-12?
Core Version:      	?2.1.0-rc2?

### Description

Problem description

### Settings in IDE

Module:  			?Generic ESP8266 Module?
Flash Size: 		?4MB/1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
messages here
```
Your example misleads people into believing that they have a telnet server, but when I connect with a telnet client, I get junk characters that are part of the telnet protocol, IAC characters, etc.  Does not support DO,DONT,WILL,WONT commands.  Please make it clear that this is not TELNET, even though you changed the port number.

penfold42, 2016-12-04:
Maybe we should rename this to WiFiTCPToSerial ?

Is it worth implementing some basic telnet protocol support. Just enough to parse and swallow the chars and reply WONT to everything

blacktigersoftware, 2016-12-04:
Probably a good idea.  I've spent the day coding up a RFC compatible telnet for the ESP8266.  Only supports a single client, but extensible.  I'm  headed to adding in full RFC 2217 for a WiFI/UART bridge.  http://www.ietf.org/rfc/rfc2217.txt.  Let me clean up the comments and I'll push it somewhere on GitHub later today.

blacktigersoftware, 2016-12-04:
Okay, just finished up packaging the library for Arduino and did testing with example sketch.
https://github.com/blacktigersoftware/ESP8266RFCTelnet

devyte, 2017-10-06:
@blacktigersoftware raw telnet is still telnet, and the example says transparent telnet to serial, which is the same thing.
Closing.

closed 2017-10-06 by devyte