esp8266/Arduino #2730: Reading data from specific location in memeoty

MisterBE2, 2016-12-03:
### Basic Infos
Based on this thread: https://github.com/esp8266/Arduino/issues/1537
i did my own text to eeprop writer. But i quite dont understand how to write and read at specific addresses. As in example below i try to save "Test" to memory starting at address 20. But when I read it it appears as text saved itself from 0 address.

Can you help me with this?

Module:  			ESP8622MOD - lolin nodemcu v3
Flash Size: 		4MB
CPU Frequency:	80Mhz
Flash Mode:		?
Flash Frequency:	?
Upload Using:		SERIAL
Reset Method:		nodemcu

#include <EEPROM.h>

void SaveString(const char* id, int startAt)
{
  int i;
  int l = startAt+strlen(id);
  for(i=startAt; i<=l; i++) 
  {
    EEPROM.write(i, (uint8_t) id[i]); 
  }
  EEPROM.commit();
}

void ReadString(int bufor, int startAt)
{
  int i;
  int l = startAt+bufor;
  char buf[bufor];
  for(i=startAt; i<=l; i++)
  {
  buf[i] = (char)EEPROM.read(i);
  Serial.print(buf[i]);
  } 
  Serial.println(" ");
}

void setup() {
  EEPROM.begin(512);
  Serial.begin(9600);

  SaveString("Test", 20);
}

void loop() {
  ReadString(4, 20);
  delay(5000);
}

igrr, 2016-12-03:
You've got a bug in this line at least:

```c++
EEPROM.write(i, (uint8_t) id[i]);
```
Loop index runs from `startAt` but the array index should run from 0. Same problem in ReadString function.

MisterBE2, 2016-12-03:
You are wright. Thank you, now it works :)

closed 2016-12-03 by MisterBE2

nonozor, 2018-05-09:
Here is a cdoe I've wrote that works for write/read String in Eeprom : 
`void memSaveStr(String str0, int startAt){
  char str[32];
  String(str0+'\0').toCharArray(str,32);
  int i=0;
  while (str[i]!='\0'){
    EEPROM.put(startAt + i, (uint8_t) str[i++]);
    Serial.println("i="+String(i));
  }
  EEPROM.write(startAt + i, '\0');
  EEPROM.commit();
}

String memReadStr(int startAt){
  char buf[32];
  int i=0;
  EEPROM.get(startAt + i,buf[i]);
  while (buf[i]!='\0'){
    EEPROM.get(startAt + i,buf[i++]);
    Serial.println("i="+String(i));
  }
  return buf;
}`