esp8266/Arduino #2732: getChipId() return wrong value

fubax, 2016-12-04:
Hello everybody
I'm using ESP-07 and arduino 1.6.9 .my ESP's default  AP MAC address is:
5E-CF-7F-14-6A-C0
and it's STA MAC is :
5C-CF-7F-14-6A-C0
But when i run `ESP.getFlashChipId()` it return this:
1440e0
And when I run `ESP.getChipId()`it return this:
146a18
I'm really confused that MAC address is not the same as chip ID also it's not 64 bit!
please help

lrmoreno007, 2016-12-04:
#include <ESP8266WiFi.h>

uint8_t MAC_array[6];
char MAC_char[18];

void setup() {
  Serial.begin(115200);
  WiFi.macAddress(MAC_array);
  for (unsigned int i = 0; i < sizeof(MAC_array); ++i){
    sprintf(MAC_char,"%s%02x:",MAC_char,MAC_array[i]);
  }
Serial.print("MAC= ");
  Serial.println(MAC_char);
}

void loop(){

}

fubax, 2016-12-04:
Thank you bro.but I need to read chip ID to protect my code.as you know it's really easy to change ESP's MAC address.
By the way when I run your code I see this:
MAC= 5c:cf:7f:14:6a:**18**: 
That is very odd !

fubax, 2016-12-05:
HELP PLEASE!

vicnevicne, 2016-12-05:
I'm afraid shouting won't help... 
Giving more detail on what you want to achieve could help on the other hand.
You seem to say MAC address is not what you want but *you* asked about the Mac address in your first post...

fubax, 2016-12-06:
Dear vicnevicne thank you for reply.
I want to use ESP's unique chip ID to protect my source code.OK?
But now I'm faced with two problem:
1:when I read chip ID with `ESP.getChipId()` it doesn't  return 64 bit,I get only last 32 bit.
2:As you know the ESP's default MAC address will be the same as chip ID and now I'm really confused that this is not correct for me.
I use nodemcu flash programmer to burn code to module and it shows the AP and STA MAC.with this feature I can simply see the chip ID and use it for my source code protection purpose,but because my module chip ID is not the same as MAC ,it will be necessary to read the chip ID first and then use it in source code that will not be easy!
Any idea can be appreciated. :)

leonyuhanov, 2016-12-12:
@fubax The ESP.getChipId() returns the ID of the ESP as a 32Bit value 
from https://github.com/esp8266/Arduino/blob/master/doc/libraries.md
> ESP.getChipId() returns the ESP8266 chip ID as a 32-bit integer.

`

devyte, 2017-10-06:
Closing due to off-topic (usage question), see #3655 .

commit 2f65f5fd (2015-05-18): Provide Hooks before and after the build process

Closes #2732
closed 2017-10-06 by devyte