esp8266/Arduino #2733: Is there any support for HTTPS (SSL) in esp8266?

jmiguels, 2016-12-04:
I have a domotic project and I want to access from the internet in a secure way. Currently I am using HTTP just for testing.
I have a gateway between internet and my RF24 home network buided using a WEMOS.

Is there any support for HTTPS (SSL) in esp8266?
Can anyone help me with the correct library to use?

Thanks in advance

jmiguels

olileger, 2016-12-04:
Hi

Fortunately you could use the `WiFiClientSecure` class instead of the `WiFiClient` one.
This should allow you to connect to a secure endpoint and to check the certificate thumbprint by using the `verify` function.

Olivier.

jjssoftware, 2016-12-04:
@jmiguels when you say:

> "I want to access from the internet in a secure way"

..are you saying you're running a HTTP server on your widget and you want to run it as HTTPS / over TLS to enable you to access it securely from a client over the internet?

jmiguels, 2016-12-04:
Thank you Olivier (olileger) and jjssoftware for your reactions to my post, 
What I am running in my WEMOS device is a web server and I want to access from the Internet using my App (client). Intead of accessing from my App (or from a browser) via HTTP I want to access to my server via HTTPS.
This is my issue at this moment.

Best regards
José Miguel Santos

jjssoftware, 2016-12-04:
ok cool that's clear thanks, please see open issue https://github.com/esp8266/Arduino/issues/1740

There is TLS support built into ESPAsyncTCP which would suggest that ESPAsyncWebServer also has TLS support since it's built on ESPAsyncTCP. I must say I've no experience of implementing either of these so I'm unable to comment on stability. @me-no-dev might have more info about how to get this going / if it's possible.

Another option might be to stick with HTTP but offload TLS onto some other box / device external to your WEMOS device. This option does sort of swerve the problem completely but it does work and it's also stable. If this sounds like a viable option, take a look here:

- https://jjssoftware.github.io/secure-your-esp8266/
- https://jjssoftware.github.io/esp8266-password-complexify/

me-no-dev, 2016-12-04:
I do not recommend using TLS in a server. While I have it implemented and working in my async libs, it's turned off by default and requires a define switch in ESPAsyncTCP to turn TLS on. What I can tell you is that it's not at all possible to serve more than one client at a time because the ESP does not have enough memory and you will get into trouble quite fast. I have even artificially limited the clients that get handled by the server to one if TLS is on.

jjssoftware, 2016-12-04:
@me-no-dev thanks for that, it sounds limited but great work nonetheless. Perhaps TLS offload to some other box/device is indeed the best option to solve this problem. It's what I do anyway ;)

olileger, 2016-12-04:
thanks for these clarification.
In that case my suggestion is out of scope as I was considering you trying to reach a TLS endpoint from your WeMoS.

+1 for offloading SSL from a public endpoint exposed by your network. You should also implement a little bit of routing (port forwarding, path-based routing...depeding on your needs).
You could easily do this using a layer 7 load balancer/reverse proxy like HAProxy or NGinx.

Olivier.

me-no-dev, 2016-12-04:
I personally do it with nginx :) If you need only one client to connect to remote SSL endpoint (that is not ESP8266), you can surely use TLS directly on the device, but still lots of memory will be eaten and you could easily trigger WDT or memory issue

jjssoftware, 2016-12-04:
[nginx FTW!](https://jjssoftware.github.io/secure-your-esp8266/)

jmiguels, 2016-12-04:
Thank you Joe Lippa for your article about nginx!
I am a little disappointed since I thought I would be able to implement all this in my WEMOS (a little naïf too…), anyway I will try to run nginx probably in my Raspberry Pi since it is a nice workaround to this issue.
Thanks to all contributors
Regards
José Miguel Santos

jjssoftware, 2016-12-04:
You're welcome I hope it helps. 

As an aside but sort of related, I've seen wdt reset issues when attempting to use `WifiClientSecure` to make secure outbound calls in the past and I'm sure the problems I saw were related to being right at the edge of available heap memory on my ESP12.

Everything TLS related seems to be right on the edge of being able to work reliably on the ESP8266. I hope we'll get there one day with the ESP32 :)

davisonja, 2016-12-08:
Interestingly I've just run into issues with outbound secure calls. Looked like memory corruption in the heap and only appeared when the code started 'doing more' (I've got it running perfectly happily on a simple task)

devyte, 2017-10-06:
Closing due to off-topic (question instead of issue).

closed 2017-10-06 by devyte

rickyzhang82, 2018-06-07:
@me-no-dev Thanks for you work. But I checked ESP32 implementation. I don't see SSL enabled even it is considered more powerful chip than ESP8266. Any reason why? Thanks!

igrr, 2018-08-07:
Since the time this issue was opened, some things have changed. ESP8266 Arduino includes BearSSL library, which is much more feature complete than axTLS. It is also more robust because it doesn't rely on heap memory allocations. If you don't have enough memory to handle the request, it will fail gracefully instead of panic-ing. Please check the included BearSLL based examples, client and server side.

ESP32 Arduino uses mbedTLS library which is also very feature complete and regularly updated.

Misiu, 2018-08-07:
@igrr thanks for quick update. 
I'll definitively look into those samples :)

jmiguels, 2018-08-07:
Thanks Ivan! I will take a look at that library.

mehotkhan, 2020-08-15:
read this : 
https://github.com/cwi-dis/iotsa/issues/27

rucko24, 2022-07-26:
> [nginx FTW!](https://jjssoftware.github.io/secure-your-esp8266/)

https://bebo-dot-dev.github.io/secure-your-esp8266/