esp8266/Arduino #2741: Get the wrong SD card information

zhulin1573, 2016-12-07:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.md

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.md

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			ESP-12E
Core Version:      	2.3

### Description

Get the wrong SD card information,the Card size is not my real SD size(16 or 8GB)

follow is print info:
    Initializing SD card...Wiring is correct and a card is present.
   Card type: SDHC
   Volume type is FAT32
   Card size (bytes): 30318592
   Volume size (bytes): 2629828608
  Volume size (Kbytes): 2568192
[CardInfo.txt](https://github.com/esp8266/Arduino/files/635941/CardInfo.txt)

  Volume size (Mbytes): 2508

### Settings in IDE

Module:  			NodeMcu1.0  ESP-12E Module
Flash Size: 		4MB (3MB SPIFSS)
CPU Frequency:		80Mhz
Flash Mode:			dio
Flash Frequency:	?40Mhz?
Upload Using:		 SERIAL
Reset Method:		?ck / nodemcu?

### Sketch

```cpp

#include <SPI.h>
#include <SD.h>

// set up variables using the SD utility library functions:
Sd2Card card;
SdVolume volume;
SdFile root;

// change this to match your SD shield or module;
// Arduino Ethernet shield: pin 4
// Adafruit SD shields and modules: pin 10
// Sparkfun SD shield: pin 8
const int chipSelect = 4;

void setup()
{
  // Open serial communications and wait for port to open:
  Serial.begin(9600);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for Leonardo only
  }

  Serial.print("\nInitializing SD card...");

  // we'll use the initialization code from the utility libraries
  // since we're just testing if the card is working!
  if (!card.init(SPI_HALF_SPEED, chipSelect)) {
    Serial.println("initialization failed. Things to check:");
    Serial.println("* is a card inserted?");
    Serial.println("* is your wiring correct?");
    Serial.println("* did you change the chipSelect pin to match your shield or module?");
    return;
  } else {
    Serial.println("Wiring is correct and a card is present.");
  }

  // print the type of card
  Serial.print("\nCard type: ");
  switch (card.type()) {
    case SD_CARD_TYPE_SD1:
      Serial.println("SD1");
      break;
    case SD_CARD_TYPE_SD2:
      Serial.println("SD2");
      break;
    case SD_CARD_TYPE_SDHC:
      Serial.println("SDHC");
      break;
    default:
      Serial.println("Unknown");
  }

  // Now we will try to open the 'volume'/'partition' - it should be FAT16 or FAT32
  if (!volume.init(card)) {
    Serial.println("Could not find FAT16/FAT32 partition.\nMake sure you've formatted the card");
    return;
  }

  // print the type and size of the first FAT-type volume
  uint32_t volumesize;
  uint32_t cardsize;
  Serial.print("\nVolume type is FAT");
  Serial.println(volume.fatType(), DEC);
  Serial.println();

  cardsize = card.cardSize();
  Serial.print("Card size (bytes): ");
  Serial.println(cardsize);
  volumesize = volume.blocksPerCluster();    // clusters are collections of blocks
  volumesize *= volume.clusterCount();       // we'll have a lot of clusters
  volumesize *= 512;                            // SD card blocks are always 512 bytes
  Serial.print("Volume size (bytes): ");
  Serial.println(volumesize);
  Serial.print("Volume size (Kbytes): ");
  volumesize /= 1024;
  Serial.println(volumesize);
  Serial.print("Volume size (Mbytes): ");
  volumesize /= 1024;
  Serial.println(volumesize);

  Serial.println("\nFiles found on the card (name, date and size in bytes): ");
  root.openRoot(volume);

  // list all files in the card with date and size
  root.ls(LS_R | LS_DATE | LS_SIZE);
}

void loop(void) {

}

me-no-dev, 2016-12-07:
SDCard lib supports partitions up to 3GB, so if you have one more than that, it will not work

devyte, 2017-10-06:
@zhulin1573 considering that volumesize is uint32_t, your sketch can't represent a volume size greater than 4GB. Maybe the example should be updated, and volumesize should be a long int or something.
Having said that I can't find any reference online that says that the SD card must be 3GB max.
If SD.h doesn't work, try SDfat lib. It seems to have greater capabilitites vs. the Arduino one.
Given that there is an alternate lib, and @me-no-dev 's comment, I'm going to close this.

closed 2017-10-06 by devyte