esp8266/Arduino #2743: WiFiServer::write() write to all clients is not implemented

MeFisto94, 2016-12-07:
### Basic Infos
So I tried searching if this issue is already present but due to the mass it could be that I overlooked it, so apologies if this is a duplicate report.

### Description
The WiFiServer currently doesn't support writing to all clients. Instead one has to keep track of the client instances and do that manually which leads to a bit more ugly code.

See: [WiFiServer.cpp#L135](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/src/WiFiServer.cpp#L135)

I would file a pull request but since I am no C++ expert, the list code is hard to read.
In general I would recommend refactoring the code to be in line with the [Arduino's EthernetServer](https://github.com/arduino-libraries/Ethernet/blob/master/src/EthernetServer.cpp).

The key difference is that it handles the client array internally, hiding it from the user.
This means that `available` always returns the first readable client instead of only returning new connections like `WiFiServer` does.

I agree that this is a design decision and that it would break exisiting code, so this is soley a proposal, but since I had to rewrite some code depending on the Platform (Arduino Mega vs ESP it would be awesome to have that consistent.

devyte, 2017-10-06:
@igrr is this something that we want to do for compatibility or consistency?

igrr, 2017-10-06:
I don't see a problem with adding a function to write to all clients. It wasn't implemented in the first place because that looked like an odd thing to do (no widely used protocol over TCP does that, AFAIK). Still, implementing this would not hurt and would improve compatibility.

The other part, about `available` returning WiFiClient which has data available for reading, rather than the first newly connected WiFiClient, would indeed break existing code, for example in the case when a remote client connects to WiFiServer and doesn't send any data (expecting the Server to send something). So for this part, we probably don't want to change the implementation.

PR #6338 (2019-07-25): Update WiFiServer docs for ::write(all clients)
Fixes #5116
Fixes #2743

The Arduino WiFiClient object allows sending the same packet to all
connected clients of a WiFiServer.  In their implementation it may make
sense, but on the 8266 with things like SSL it doesn't.

Update the docs to note that WiFiServer::write() is a no-op, and that
the app should use the WiFiCliebnt::write() on all connected clients as
appropriate.
commit 76ed52c1 (2019-07-25): Update WiFiServer docs for ::write(all clients) (#6338)

Fixes #5116
Fixes #2743

The Arduino WiFiClient object allows sending the same packet to all
connected clients of a WiFiServer.  In their implementation it may make
sense, but on the 8266 with things like SSL it doesn't.

Update the docs to note that WiFiServer::write() is a no-op, and that
the app should use the WiFiCliebnt::write() on all connected clients as
appropriate.
closed 2019-07-25 by d-a-v