esp8266/Arduino #2744: Environment Variables?

JohnAntonusMaximus, 2016-12-07:
Hey guys, quick question...so I'm storing a variable as a JSON object in a sketch, but I may need to update firmware going forward (security patches, etc.), but I just realized that specific JSON object will be different on each device I'm building and if I upload new firmware to each board, all of those unique JSON objects will then be erased. 

My question is...is there any equivalent to 'environment variables' using NodeMCU/ESP8266? So that I can make sure those unique JSON objects are not overwritten when uploading a new version of firmware to the board? That way I can just have the code be uniform for each device to access their unique environment variables?

I spend most of my time doing web programming, so any help would be appreciated. 

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:		NodeMCU ESP-12E
Core Version:

SloMusti, 2016-12-08:
This project https://github.com/ESP8266nu/ESPEasy is able to store configuration that is re-flash persistent, you can have a look at how they have done it.

devyte, 2017-10-06:
@JohnAntonusMaximus use the SPIFFS.
Closing as off-topic, see #3655 .

closed 2017-10-06 by devyte