esp8266/Arduino #2746: HTTPClient downloads wrong data

holgerlembke, 2016-12-08:
Using the StreamHTTPClient example I download an URL providing binary data generated by a php script. Script shall send 7504 bytes (3 * 50 * 50 +5) with no Content-length: header. It is verifiable with IE and FF, they both receive 7504 bytes.

StreamHTTPClient fails. There are some extra bytes at the beginning and the end of  the saved stream.

After adding the Content-length: header into the php script, it works for the StreamHTTPClient, too.

devyte, 2017-10-06:
@holgerlembke no sketch, no basic info, no IDE settings.
Is this issue still valid with latest git?

holgerlembke, 2017-10-07:
I don't know, I use the latest release version 2.3.0-rc2.

I tried to make a demo, but thus is not downloaded at all but gives HTTPC_ERROR_READ_TIMEOUT.

without content-length: http://atomosys.de/lalatest/lala1.php

with content-length: http://atomosys.de/lalatest/lala2.php

There is only a sleep(10) inside the php stuff.

Codequotes:
```
void lalatest(void) {
  Serial.println("alala");
 
  HTTPClient http;

  http.begin("http://atomosys.de/lalatest/lala2.php");
  int httpCode = http.GET();
  Serial.print("httpCode: ");
  Serial.println(httpCode);
  if (httpCode > 0) {
    if (httpCode == HTTP_CODE_OK) {
      String answer = http.getString();
      Serial.print("lala1: ");
      Serial.println(answer.length());
    }
  }
  http.end();

  http.begin("http://atomosys.de/lalatest/lala1.php"); 
  httpCode = http.GET();
  Serial.print("httpCode: ");
  Serial.println(httpCode);
  if (httpCode > 0) {
    if (httpCode == HTTP_CODE_OK) {
      String answer = http.getString();
      Serial.print("lala2: ");
      Serial.println(answer.length());
    }
  }
  http.end();

  Serial.println("blala");
}
```

devyte, 2017-12-05:
@holgerlembke 2.3.0-rc2 is beyond ancient, it's a release candidate prior to 2.3.0. Please try latest git, preferrably building with lwip2 (you can select that from the Arduino IDE menu after installing git).

devyte, 2018-02-28:
@holgerlembke is this still valid with 2.4.0? with latest git?

holgerlembke, 2018-02-28:
How to get 2.4.0?

The on https://github.com/esp8266/Arduino labeled with 2.4.0 only offers 2.3.0 in the Arduino ide 1.8.5.

[image]

devyte, 2018-07-12:
Closing due to lack of feedback.

closed 2018-07-12 by devyte