esp8266/Arduino #2748: HTTPSConnect Connection Failed

maurbeck, 2016-12-09:
### Basic Infos

#### Hardware
Hardware:			Wemos D1 mini

### Description

Running HTTPSConnect and successfully connecting to api.github. Switch the paths to point towards my own server and connections fail. I get a "connection failed" message.

How do I get specific failure codes to understand better why it's not working?

maurbeck, 2016-12-09:
Anybody know what TLS/SSL version this library uses?

gjt211, 2016-12-09:
Hi maurbeck.
I am having other issues with TLS/SSL, but from what I (think) worked out today, I believe it must be TLSv1.1, with RC4-SHA. I added `RC4-SHA` and removed `!RC4` in my apache ciphers list configuration which got me going, but now my site's security only rates B- rather than an A because RC4 is not considered very secure. From my searching through the issues here, it seems like the TLS is using the code from axTLS which only supports TLSv1.1
Hope that helps you.

jjssoftware, 2016-12-09:
axTLS has been updated to v2 which has support for TLS1.2. See https://github.com/esp8266/Arduino/issues/2537
This change is not in the latest 'official' 2.3.0 release however because it's a change that was implemented some time after when 2.3.0 was released on 23/06 so if you wanted to give this a try, you'd need to directly use the current git code.

maurbeck, 2016-12-09:
Great news. I'll try that out, thanks.

Would a typical server be able to send an intelligible error code for this? It was quite a pain to debug what was going on without any way of knowing why the connection failed and I didnt see a way of exposing an error code in the library.

jjssoftware, 2016-12-09:
If you're trying to workout what goes wrong at TLS handshake time, the thing you're probably interested in is the int response from an ssl_handshake_status function call. The official axTLS docs are here: http://axtls.sourceforge.net/dox/

Inside the WiFiClientSecure class this is called at connect time inside WiFiClientSecure::_connectSSL but unfortunately the value isn't surfaced out when something goes wrong. See here:
https://github.com/esp8266/Arduino/blob/4897e0006b5b0123a2fa31f67b14a3fff65ce561/libraries/ESP8266WiFi/src/WiFiClientSecure.cpp#L319

You could change that locally yourself so the value is returned back in the connect call..that might give you what you need. If that works as expected you could submit a pull request as a potential improvement :)

maurbeck, 2016-12-09:
I have uninstalled the stable ESP 2.3.0 and installed 2.3.0-rc2. This did not solve my problem, so I'm wondering if that was my issue in the first place. 

So now I'm moving on to trying to return some information in _connectSSL. I see that the status is returned as type auto. It's simply checked against SSL_OK. 

What's the reason for it being auto? How many different types can be expected to return from ssl_handshake_status()?

I haven't played with C++ for a while. I'm having trouble figuring out how to turn that into something I can pass back to WiFiClientSecure::_connect to return to my Arduino implementation.

jjssoftware, 2016-12-09:
2.3.0 rc2 was a pre-release version of 2.3.0

If you're aiming for TLS1.2 support in axTLS then 2.3.0.anyVersion just won't do it. At all. If TLS1.2 is what you want, you need to use the git code as I previously mentioned.

maurbeck, 2016-12-09:
I see ssl_handshake_status() is either in libaxtls.a or libssl.a - which I don't know how to look at.

I feel like I need to define a specific filetype to accept the status returned, but I think I need to look at the function return itself to understand that, right?

maurbeck, 2016-12-09:
Oh, I see. I thought that the git code itself was accessible from the staging directory.

I'll go back and look at how to do what you've suggested once again.

maurbeck, 2016-12-09:
Ok, can you hold my hand a little while longer please?

I've pulled the code from git into the esp8266com dir according to [your link](https://github.com/esp8266/Arduino#using-git-version). Then I downloaded the binary tools by running get.py and restarting Arduino. 

I tried running again and got the same issue as before where the connection fails. Do I need to switch to using esp8266com rather than the standard one in some manner or something? I don't understand what is next.

maurbeck, 2016-12-09:
I uninstalled the esp8266 2.3.0 lib again and think that I should be finding a way to add the esp8266com files to my boards. I'm wondering if I should just copy the contents of this over top of the existing esp8266 2.3.0 lib contents.

Trying that next.

maurbeck, 2016-12-09:
Did that. Did not work. Wondering if I should try to use DEBUGV to log that response so that I can understand what's going on. Should I try using DEBUGV in such a manner? I wouldn't know how to begin down that journey...

jjssoftware, 2016-12-09:
Are you developing your stuff with the arduino IDE / trying to get this projects' git cloned code working in the arduino IDE by any chance?

maurbeck, 2016-12-10:
Yes, I'm using the Arduino IDE. Is that possible?

jjssoftware, 2016-12-10:
Yes I'm sure it's possible but I'm afraid I won't be able to help you determine if it's setup correctly with cloned github code because I gave up on it long ago myself.

davisonja, 2016-12-15:
@maurbeck Still having trouble making the git clone work? :)

eiannone, 2017-05-20:
Hi @maurbeck , I'm having your same issue. When trying to connect with WiFiClientSecure,  ssl_handshake_status() function returns -1 (SSL_NOT_OK)
I'm using Arduino IDE with latest esp8266 library cloned from github, and Wemos D1 mini hardware.
Did you manage to solve the problem?

maurbeck, 2017-05-20:
Sorry guys, I have up on trying to make it work shortly before I ended the project that needed it. 

The solution I was eyeing though was to have a pi relay the info with the proper SSL stuff. Won't work for many projects though :/

eiannone, 2017-05-20:
Hi, after some investigation I found that the issue is due to lack of ciphers compatibility between ESP8266 and the server.
ESP8266 uses the axTLS library (http://axtls.sourceforge.net/), built in from Expressif SDK, which supports only the following ciphers:
TLS_RSA_WITH_AES_128_CBC_SHA256
TLS_RSA_WITH_AES_256_CBC_SHA256
TLS_RSA_WITH_AES_256_CBC_SHA
TLS_RSA_WITH_AES_128_CBC_SHA

My server instead supports only TLS_ECDHE_RSA_* and TLS_DHE_RSA_* ciphers, so they cannot complete the handshake phase.

igrr, 2017-05-21:
@eiannone we don't use the library which comes from the Espressif SDK, instead this port is used:
https://github.com/igrr/axtls-8266
You're correct about the list of supported cipher suites though :)

eiannone, 2017-05-21:
I didn't know that @igrr , thank you for pointing it out!

devyte, 2018-05-29:
BearSSL is merged in #4273 , with alternate BearSSL::WiFi* classes. Although axtls-based classes are still available and even the default, they are planned for deprecation and then retirement, hence won't be fixed. Any issues with BearSSL-based classes should be reported in new issues.
Closing.

closed 2018-05-29 by devyte