esp8266/Arduino #2751: warning: espcomm_sync failed error: espcomm_open failed error: espcomm_upload_mem failed error: espcomm_upload_mem failed

tanyp97, 2016-12-10:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.md

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.md

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			?ESP-12?
Core Version:      	?2.1.0-rc2?

### Description

Problem description

### Settings in IDE

Module:  			?Generic ESP8266 Module?
Flash Size: 		?4MB/1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
messages here
```

jyotirajsharma, 2017-01-03:
Did you get it working now?

jyotirajsharma, 2017-01-03:
Should TX from Arduino not be connected to Rx of ESP8266 and vice versa ??

kumarmaxwell, 2017-01-08:
Hi all,
By Connecting RESET,CH_PD pins of E01 module to 3.3V and GPIO0 to GND. These error can resolved..A brief post on Errors are [here](https://esp8266e01.wordpress.com/2016/12/05/346/)

jyotirajsharma, 2017-01-09:
Do we need to supply 3.3v supply from external power source or we can use arduino power source only ?

jyotirajsharma, 2017-01-09:
Hi,
I am still getting same error after using configuration suggested by you:
C:\Users\Jyoti\AppData\Local\Arduino15\packages\esp8266\tools\esptool\0.4.9/esptool.exe -vv -cd ck -cb 115200 -cp COM3 -ca 0x00000 -cf C:\Users\Jyoti\AppData\Local\Temp\arduino_build_39553/dht_wifi.ino.bin 
esptool v0.4.9 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to ck
	setting baudrate from 115200 to 115200
	setting port from COM1 to COM3
	setting address from 0x00000000 to 0x00000000
	espcomm_upload_file
	espcomm_upload_mem
	setting serial port timeouts to 1000 ms
opening bootloader
resetting board
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	serialport_receive_C0: 08 instead of C0

Thanks,

jyotirajsharma, 2017-01-09:
I have nearly spent more than 20 days trying to run esp8266 but ended up doing nothing. It appears to me that esp8266 has lots of bugs and there is no exact connection diagram to fix sync error problem. I am getting same error everytime even after following almost all the suggestions mentioned in the esp8266 community. I am frustrated.

Is there noway to flash the esp8266 ? I see that almost 99% people are facing this problem and seems no exact solution to this.

vicnevicne, 2017-01-09:
> Do we need to supply 3.3v supply from external power source or we can use arduino power source only ?

jyotirajsharma, you're asking that question again and again. You even opened #2825 explicitely about it.
This is not a bug, it's a usability question, so I already suggested that you post that question to the esp8266.com forums, but I searched for your name on the forums and couldn't find any match. Did you post it ?
Moreover, I notice you already got an answer by hackernikhilraj in #2772 : 
> arduino didn't supply that much of current that is required for ESP8266 to work. arduino o/p current is less than that.

And common sense just dictates that if you are still wondering after a week, well, just try with a separate source. If you don't have a power supply, use 3 rechargeable batteries (3 x 1.2V = 3.6V) for example and try it for yourself.
Plus, I couldn't find anywhere the exact setup you're using. Is it a stand-alone module ? And if so, which one ? Do you use a USB to serial converter? If so, which one ?
But please don't answer here: head to the forums, ask this question or complete it with as much detail as possible and post a link here so that we continue the discussion there.

Once again: only open an issue if you were able to confirm you have found a bug.

Oh, and no, 99% of ESP8266 users are not facing this problem... but 99% of users that ask vague questions at the wrong place never get a solution indeed... ;-)

jyotirajsharma, 2017-01-10:
I have raised a question regarding this by creating a new forum but it says it needs a approval from moderator so do not have a link to post it now. I posted my concern here because the suggestion posted above where it looked they used arduino power supply only and i tried same config but did not get the correct result. I am using 
LAPTOP USB<---->ARDUINO(3.3v)<------->esp8266(Vcc)
Anyway, please accept post from forum i just raised now and please do not get upset about my last post.
Thank you for your anticipated co-operation :-)

jyotirajsharma, 2017-01-12:
I am still getting same error after using external power supply (3 AA 1.2v = 3.6v battery).
Please reply on http://www.esp8266.com/viewtopic.php?f=6&t=13279#p60775.
Please help to resolve this. I am not finding anything wrong with the connection.

proddy, 2017-02-18:
on my boards you cannot upload when there is something connected to TX

truebucha, 2017-06-20:
Just my 2 cents
Generally this error means there are no running esp8266 on the other side
So Check 
0 rx connected to tx and vise versa
1 the gpio0 and gpio15, gnd connected to ground
2 the vcc and en connected to external (not to programmer itself) power source
in my case the board doesn't work from programmer power supply
3 if you use breadboard please check carefully power line from power supply to esp, especially to en pin. the en pin should be connected right to the source without resistor.
if you measure some voltage on connected esp it is still not clear that connection is good enough.
in my case using dupont connectors on this lines leads to errors that appears by random from time to time﻿
if you want read some from console my esp12s has default rate 74880, maybe yours to.
some of UARTs want high CTS signal (CTS<->10k resistor to power source)

devyte, 2017-10-06:
The espcomm open failed error is documented. If what the doc says doesn't fix your problem, you have either bad wiring, bad power supply, or a bad board.
Closing.

closed 2017-10-06 by devyte