esp8266/Arduino #2752: Proprietary license on example code

Gavekort, 2016-12-10:
I'm a little unsure how to interpret the use of the example code provided with this library. There are two main issue that I'm having, and that is the complete lack of consistency between copyright notices and what I assume are different licenses. The second issue is that some of the provided examples are what I can only presume as proprietary code, which is not fitting as example code at all.

Most examples, like WiFiScan and SSDP is not explicitly licensed, and does not contain a proper header, which I can only default to as proprietary code:

>  This sketch demonstrates how to scan WiFi networks. 
>  The API is almost the same as with the WiFi Shield library, 
>  the most obvious difference being the different file you need to include:
> #include "ESP8266WiFi.h"

Some examples, like AdvancedWebServer have some burdening restraints:

> Copyright (c) 2015, Majenko Technologies
> All rights reserved.

> Redistribution and use in source and binary forms, with or without modification,
> are permitted provided that the following conditions are met:

> * Redistributions of source code must retain the above copyright notice, this
>   list of conditions and the following disclaimer.

> * Redistributions in binary form must reproduce the above copyright notice, this
>   list of conditions and the following disclaimer in the documentation and/or
>   other materials provided with the distribution.

> * Neither the name of Majenko Technologies nor the names of its
>   contributors may be used to endorse or promote products derived from
>   this software without specific prior written permission.

> THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND CONTRIBUTORS "AS IS" AND
ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT LIMITED TO, THE IMPLIED
WARRANTIES OF MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE ARE
DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT HOLDER OR CONTRIBUTORS BE LIABLE FOR
 ANY DIRECT, INDIRECT, INCIDENTAL, SPECIAL, EXEMPLARY, OR CONSEQUENTIAL DAMAGES
 (INCLUDING, BUT NOT LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS OR SERVICES;
 LOSS OF USE, DATA, OR PROFITS; OR BUSINESS INTERRUPTION) HOWEVER CAUSED AND ON
 ANY THEORY OF LIABILITY, WHETHER IN CONTRACT, STRICT LIABILITY, OR TORT
 (INCLUDING NEGLIGENCE OR OTHERWISE) ARISING IN ANY WAY OUT OF THE USE OF THIS
 SOFTWARE, EVEN IF ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.

Unfortunately the proprietary intellectual property of others is not something I can modify and create a pull request on. I am however proposing that the maintainer either purge all of this proprietary code and replace it with free alternatives, or that the maintainer reaches out to the respective authors and asks them to release their examples under a free license.

igrr, 2016-12-10:
The license in the example you have quoted is 3-clause BSD license, which is a free software license according to the FSF. What would be the specific license you propose to use instead?

Gavekort, 2016-12-10:
Sorry about that. I updated my comment just as you typed this. But the inconsistency is still an issue, and most of the examples I interpret as proprietary.

me-no-dev, 2016-12-10:
Sorry guys but I can not wrap my head around "proprietary Arduino example". How could an example in Arduino be proprietary at all? I get why libs should be licensed, but examples?

Gavekort, 2016-12-10:
Unless the author is explicitly licensing his code for you to use as free software it is proprietary software. This means that modifying and redistributing some of the examples bundled in this framework may be a copyright violation.

http://choosealicense.com/no-license/

me-no-dev, 2016-12-10:
from what I read in that link, a general license can be added to README and be done.

Gavekort, 2016-12-10:
Yes. But the author is still the owner of the code, and since there is no header in the source file, nor any explicit rights granted to these example-snippets we can't just edit the README and call it a day. In some cases it is very clear that underlying code without a header is still covered under the license. Unfortunately there is nothing that implies that the example-code is granted the same rights as the core-code in this case.

What needs to be done is to either get the consent from the respective authors and put all example code under the same license as the rest of the code, or to get the respective authors to _explicitly_ license their own code in a way that allows modification and redistribution. Right now you are not allowed to build upon the provided example code, as this would be a copyright violation.

devyte, 2017-10-06:
While examples can be considered a bit of a gray area, the repo clearly states LGPL, and the rules of github state that once you choose a license for your repo, the license applies to the entire repo. The rules are not all-encompassing, which means that it is also possible to state exceptions, so you can use a different license for specific files.
Therefore, the example files here that don't have a specific license stated in them fall under LGPL, and don't default to proprietary.
Closing as resolved.

closed 2017-10-06 by devyte