esp8266/Arduino #2754: serial rx interrupt

xhdix, 2016-12-11:
### Basic info
program use 4 timer , telnet in ap or sta, serial with atmega, i2c with ds1307, and more..

#### Hardware
Hardware:			ESP-07
Core Version:      	2.1.0-rc2

### Description

in one time.. i went give some command from telnet and serial .. analyse and excute function  without any lose data

and every 200ms 3 different packet from serial analyse and every 1s send to telnet..

### similar : 

```cpp

void setup() {
 initalize rtc1307time, timer, wifi , telnet, serial and more..
}

void loop1() {
 event/interrupt recive telnet
 if complate : analyse command
 if command == a
  dosomting();
 }else if command==b 
  dosomting();
 }else if ...
 }
}

void loop2() {
 event/interrupt recive serial
 if complate : analyse command
 if command == a
  dosomting();
 }else if command==b 
  dosomting();
 }else if ...
 }
}

```

### Problem

arduino esp8266 have some code similar uart.on() and net.socket:on() ?

uart(serial) interrupt(event) very very very important for me.. ( data lose is a bad thing.. )

https://nodemcu.readthedocs.io/en/master/en/modules/uart/#uarton

https://nodemcu.readthedocs.io/en/master/en/modules/net/#netsocketon

Sorry for my poor English :|

kendo55, 2016-12-13:
hi,
look at this:
https://github.com/esp8266/Arduino/issues?utf8=%E2%9C%93&q=serialEvent

my solution is:

```

String inputString = "";         // a string to hold incoming data
boolean stringComplete = false;  // whether the string is complete

void loop() {
//--------------------------------  
  #ifdef __AVR__
    // nothing , USART Interrupt is implemented
  #else
    // ESP8266  polling 
     ESPserialEvent();
  #endif
  
 // print the string when a newline arrives and do something...:
  if (stringComplete) {
    Serial.println(inputString);
    
     doCMD();   // CMD interpreter
  }
}

#ifdef __AVR__

void serialEvent() {
  while (Serial.available()) {
    // get the new byte:
    char inChar = (char)Serial.read();
    // add it to the inputString:
    inputString += inChar;
    // if the incoming character is a newline, set a flag
    // so the main loop can do something about it:
    if (inChar == '\n') {
      stringComplete = true;
    }
  }
}
#elif defined(ARDUINO_ARCH_ESP8266)

void ESPserialEvent() {
  while (Serial.available()) {
    // get the new byte:
    char inChar = (char)Serial.read();
    // add it to the inputString:
    inputString += inChar;
    // if the incoming character is a newline, set a flag
    // so the main loop can do something about it:
    if (inChar == '\n') {
      stringComplete = true;
    }
  }
}
#endif
```

xhdix, 2016-12-13:
hi.
thanks for your comment..
i saw issuses before
this code use loop function and risk for use 

i went uart0 rx interrupt for my compatible code.. uart data very important and never lose eny characters..
string have important values.. 
my code:
atmega read 3 adc in 20ms .. analyse then measuring 200ms .. then send 3 values to esp serial port..  then esp analyse 3 values and measuring  then store ... some values send to mobile with telnet.. and give command from telnet.. analyse and some command send to atmega..  some times atmega report with some other values.. (timmer, and more synchronic executing)

i see espressif rtos and non os sdk api refrence.. 
both pdfs have uart0 rx interrupt 

rtos sdk api : 
4.27.2.1 UART_ClearIntrStatus(UART_Port uart_no, uint32 clr_mask) . . . . . . . . . . 135
4.27.2.2 uart_init_new(void) . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . 136
4.27.2.3 UART_intr_handler_register(void fn, void arg) . . . . . . . . . . . . . . . . . 136
4.27.2.4 UART_IntrConfig(UART_Port uart_no, UART_IntrConfTypeDef pUARTIntrConf) 136

non os sdk api : 
8.2.3. uart0_rx_intr_handler 196
.....................................................................................................

other ide like  espruino or nudemcu have command "event/on" for handle uart interrupt

@me-no-dev @igrr @Links2004  
how to append uart interrupt from sdk to library till works good? 

i must use nudemcu??       

Sorry for my poor English :|

devyte, 2017-10-06:
@xhdix I'm not aware of any serial object implemented with async callbacks. The general usage here is to use the HardwareSerial object class, and poll Serial.available() in the loop to see if bytes have arrived, and if they have, then read them out. There is a buffer under the hood, which I understand gets filled from the serial ISR, so when you read out bytes, you are really pulling them from the buffer. Unless you have a very high incoming data rate over the serial, it should be enough for you.
To be honest, I wouldn't mind seeing an async serial object, but given that the current Arduino-style Serial class works, it's not high priority.
If you need further assistance, please discuss at a forum.
Closing due to off-topic, see #3655 .

closed 2017-10-06 by devyte