esp8266/Arduino #2756: Exception 28 when  trying to print system info.

kommancs96, 2016-12-11:
I get exception 28 when trying to print general information about my hardware.
My code is the following:

```
extern "C"{
  #include "user_interface.h"
  #include "ets_sys.h"
  #include "osapi.h"
}

void setup(){
  Serial.begin(115200);
  delay(2000);
  Serial.printf("Chip ID: %ld\n", system_get_chip_id());
  Serial.printf("SDK version: %s\n", system_get_sdk_version());
  Serial.printf("Boot version: %s\n", system_get_boot_version());  // Fails
  Serial.printf("SPI FLASH ID: %s\n", spi_flash_get_id()); // Fails
  Serial.printf("CPU frequency: %s\n", system_get_cpu_freq()); // Fails
  Serial.printf("User bin address %x\n", system_get_userbin_addr()); // Fails
}

void loop() {
  while(1){
     delay(100);
  }
}

```

The 4 failing lines in setup throw an exception like this:
```
Exception (28):
epc1=0x402086bc epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000050 depc=0x00000000

ctx: cont
sp: 3ffef3b0 end: 3ffef7c0 offset: 01a0
```
My board is a nodeMCU board with esp 12-e or 12-f

pipi61, 2016-12-11:
Hi!
Try https://github.com/esp8266/Arduino/tree/master/libraries/esp8266/examples/TestEspApi

liquidfalcon, 2016-12-12:
You're passing integers to a function that's expecting a character pointer, so, of course this code will fail. It will try and dereference the passed integers and crash, instead of printing them as numbers like you expet.

devyte, 2017-10-06:
Closing per previous comment (usage error). Also, the ESP object has an interface better than direct SDK usage.

closed 2017-10-06 by devyte