esp8266/Arduino #2759: Can't connect to Server with client on another ESP

league55, 2016-12-12:
----------------------------- Remove above -----------------------------

#### Hardware
Hardware: ESP-8266 x2

### Description

I have 2 Esp-8266, created Soft AP and Server one the first one and Client on the second but can't astablish connection. Client connects to Wi-fi AP, but doesn't respond. I'm pretty sure that server is OK, because it works with my android App, and i can do requests via computer after connect to WIFI.
I provide code. I also tried WebServer library and I get -1 what means "connection refused". 

TCPServer
`  #include <ESP8266WiFi.h>

  WiFiServer server(80);
  const char* ssid = "esp8266";
  const int RED = 15;
  
  void setup() {
  Serial.begin(115200);
  Serial.println();
  pinMode(RED, OUTPUT);

  Serial.print("Setting soft-AP ... ");
  Serial.println(WiFi.softAP(ssid) ? "Ready": "Failed");
  Serial.printf("Soft-AP IP address = ");
  Serial.print(WiFi.softAPIP());
  
  server.begin();
  Serial.printf("Web server started, open %s in a web browser\n", WiFi.localIP().toString().c_str());
}

void loop() {
 WiFiClient client = server.available(); 
  // wait for a client (web browser) to connect
  if (client)
  {
    Serial.println("\n[Client connected]");
    while (client.connected())
    {
      // read line by line what the client (web browser) is requesting
      if (client.available())
      {
        String line = client.readStringUntil('\r');
        Serial.print(line);
        analogWrite(RED, HIGH); 
       //  wait for end of client's request, that is marked with an empty line
        if (line.length() == 1 && line[0] == '\n')
        {
          client.println(200);
          break;
        }
      }
    }
    delay(1); // give the web browser time to receive the data

    // close the connection:
    client.stop();
    Serial.println("[Client disonnected]");

}
}`

Client
`#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>

ESP8266WiFiMulti WiFiMulti;

void setup() {
    Serial.begin(115200);
    delay(10);

    // We start by connecting to a WiFi network
    WiFiMulti.addAP("esp8266");

    Serial.println();
    Serial.println();
    Serial.print("Wait for WiFi... ");

    while(WiFiMulti.run() != WL_CONNECTED) {
        Serial.print(".");
        delay(500);
    }

    Serial.println("");
    Serial.println("WiFi connected");
    Serial.println("IP address: ");
    Serial.println(WiFi.localIP());

    delay(500);
}

void loop() {
    const uint16_t port = 80;
    const char * host = "192.168.4.2"; // ip or dns

    
    
    Serial.print("connecting to ");
    Serial.println(host);

    // Use WiFiClient class to create TCP connections
    WiFiClient client;

    if (!client.connect(host, port)) {
        Serial.println("connection failed");
        Serial.println("wait 5 sec...");
        delay(5000);
        return;
    }

    // This will send the request to the server
    client.print("Send this data to server");

    //read back one line from server
    String line = client.readStringUntil('\r');
    client.println(line);

    Serial.println("closing connection");
    client.stop();
    
    Serial.println("wait 5 sec...");
    delay(5000);
}`

bio-amr, 2016-12-25:
i had the same issue that i couldn't connect ESP8266 "as a client" to matlab " as a server " but i found a solution

sol666666, 2017-01-05:
i have the same issue. i used the exactly same code for client. it worked after a while the hole tcp connection lost. please help.tnx

league55, 2017-01-07:
guys, I don't have second esp now, but I guess I found solution. Try to use DNS library to work with link instead of IP address...

tanaydeshmukh1, 2017-07-07:
can anyone provide me link to get  <ESP8266WiFi.h> plzzzzz

devyte, 2017-10-06:
@league55 in your server, your printout:  
`Serial.printf("Web server started, open %s in a web browser\n", WiFi.localIP().toString().c_str());
`
Shouldn't that be:
`Serial.printf("Web server started, open %s in a web browser\n", WiFi.softApIP().toString().c_str());
`
?

The IP address in the SoftAP is usually 192.168.4.1, and I guess that holds for you, because you aren't specifying the softAP IP address. In your client, I see:
```
    const char * host = "192.168.4.2"; // ip or dns
...
    if (!client.connect(host, port)) {
...
```

I think that should be:
`    const char * host = "192.168.4.1"; // ip or dns
`

If you need further assistance, please discuss at a community forum.
Closing due to user error, and off topic, see #3655 .

closed 2017-10-06 by devyte

salmanulfaris, 2019-03-04:
You may not included the port (80 in your code) with IP. If didn't mention the Port with IP Request may refused.
For Example -> 192.168.1.106:80

miladfzn, 2022-07-22:
there is difference between WiFi.localIP() and WiFi.softAPIP()
you must use WiFi.softAPConfig (ip, gateway, subnet) to solve your problem