esp8266/Arduino #2760: Problem with WIFI WPA personal TKIP again

sixbacon, 2016-12-12:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.md

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.md

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			?ESP-12?
Core Version:      	?2.1.0-rc2?

### Description

Problem description

### Settings in IDE

Module:  			?Generic ESP8266 Module?
Flash Size: 		?4MB/1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
messages here
```

I have had the same program as in issue 2038, same error reports, when loading the sketches with version 2.3.0. either as nodeMCU 1.0 or generic ESP8266. The example programs would not connect to a Netgear DG834G - an old one - but would connect to various other wireless routers. The wifi scan program does work.

I too rolled back to version 2.0.0 and the sketches worked. I have various ESP12E and ESP01 units working which I developed on this earlier version and never had any problems. 

If there is anything I can do to illuminate the problem let me know, but I suspect it will only affect a few people.

Thanks for all your efforts is developing this software.

devyte, 2017-10-05:
@sixbacon is this issue still valid with latest git? There is also PR #3215 to test out, which has the latest sdk.
Please report back here.

sixbacon, 2017-10-06:
Unfortunately I had to redo my network and replace the router since I notified the issue. I now have no issues, but the router has changed.

closed 2017-10-06 by sixbacon