esp8266/Arduino #2763: Not working with MAX/MSP?

BLCKPSTV, 2016-12-13:
I managed to create the sender/receive example and make it work. 
Can read and send a package

Now I went to max/msp to read data from it and nothing happend. What was possible is that I could send data to the arduino, but the reply didn't came back in the max/msp console?

````
#include <ESP8266WiFi.h>
#include <WiFiUdp.h>

const char* ssid = "SSID";     //SSID
const char* password = "fancypass";    //Password
IPAddress ip(192,168,1,176); // Fixed IP address 
IPAddress gateway(192,168,1,1); // Gateway
IPAddress subnet(255,255,255,0); // Subnetmask

WiFiUDP Udp;
unsigned int localUdpPort = 4210;
char incomingPacket[255];
char  replyPacekt[] = "Hi there! Got the message :-)";

void setup() {
      Serial.begin(115200);
      Serial.println();

      // check static ip
      Serial.print(F("Setting static ip to :"));
      Serial.println(ip);
      WiFi.config(ip,gateway,subnet);
      
      
      // Connect to WiFi network
      Serial.println();
      Serial.println();
      Serial.print("Connecting to ");
      Serial.println(ssid);
      
      //WiFi.config(ip);
      WiFi.begin(ssid, password);
      
      while (WiFi.status() != WL_CONNECTED) {
      delay(500);
      Serial.print(".");
      }
      Serial.println("");
      Serial.println("WiFi connected");

      // Print the IP address
      Serial.println(WiFi.localIP());

      Udp.begin(localUdpPort);
      Serial.printf("Now listening at IP %s, UDP port %d\n", WiFi.localIP().toString().c_str(), localUdpPort);
      

}

void loop() {

  int packetSize = Udp.parsePacket();
  if (packetSize)
  {
    // receive incoming UDP packets
    Serial.printf("Received %d bytes from %s, port %d\n", packetSize, Udp.remoteIP().toString().c_str(), Udp.remotePort());
    int len = Udp.read(incomingPacket, 255);
    if (len > 0)
    {
      incomingPacket[len] = 0;
    }
    Serial.printf("UDP packet contents: %s\n", incomingPacket);

    // send back a reply, to the IP address and port we got the packet from
    Udp.beginPacket(Udp.remoteIP(), Udp.remotePort());
    Udp.write("hello");
    Udp.write(replyPacekt);
    Udp.endPacket();
  }
}

````

BLCKPSTV, 2016-12-14:
I think this is because I need to write to the specific port and not to the Udp.remotePort() because that's different one than the 4210 as in the code.

devyte, 2017-10-05:
@BLCKPSTV you seem to have a 1-off error: incomingPacket has size 255, and you read 255 bytes into it, but then you set incomingPacket[len] = 0, which seems to be an attempt to null-terminate. That is 1-passed-the end.
Closing due to usage error, and off-topic, see #3655 .

closed 2017-10-05 by devyte