esp8266/Arduino #2764: Nodemcu v3 can connecting wifi from Lua IDE, but can not connecting from Arduino IDE

memina, 2016-12-14:
Nodemcu v3 can connecting wifi from Lua IDE, but can not connecting from Arduino IDE
I tried from Mac and Windows. And Core version 2.1, 2.2, 2.3  but not connected. And I tried different Wifi, but nothing. Please help, thanks.

Hardware:			Nodemcu v3 ESP-12E
Core Version:      	2.3

Problem description

### Settings in IDE

Module:  			NodeMCU 1.0 (ESP8266 ESP12-E)
Flash Size: 		4MB
CPU Frequency:		80Mhz
Flash Mode:			not view
Flash Frequency:	80Mhz
Upload Using:		Serial
Reset Method:		

### Sketch

#include "ESP8266WiFi.h"

void setup() {
  Serial.begin(115200);

  // Set WiFi to station mode and disconnect from an AP if it was previously connected
  WiFi.mode(WIFI_STA);
  WiFi.disconnect();
  delay(100);

  Serial.println("Setup done");
}

void loop() {
  Serial.println("scan start");

  // WiFi.scanNetworks will return the number of networks found
  int n = WiFi.scanNetworks();
  Serial.println("scan done");
  if (n == 0)
    Serial.println("no networks found");
  else
  {
    Serial.print(n);
    Serial.println(" networks found");
    for (int i = 0; i < n; ++i)
    {
      // Print SSID and RSSI for each network found
      Serial.print(i + 1);
      Serial.print(": ");
      Serial.print(WiFi.SSID(i));
      Serial.print(" (");
      Serial.print(WiFi.RSSI(i));
      Serial.print(")");
      Serial.println((WiFi.encryptionType(i) == ENC_TYPE_NONE)?" ":"*");
      delay(10);
    }
  }
  Serial.println("");

  // Wait a bit before scanning again
  delay(5000);
}

### Serial Monitor

scan start
scan done
no networks found

scan start
scan done
no networks found

scan start
scan done
no networks found

scan start

Exception (29):
epc1=0x4020e85b epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000019 depc=0x00000000

ctx: sys 
sp: 3ffffd00 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffea0:  00000001 00001388 3ff20200 ffffffd3  
3ffffeb0:  3ffffed0 4020e96b 3ffececc 0000000e  
3ffffec0:  4022b6ee 0000000e 3ffed678 00000064  
3ffffed0:  00000018 00000000 3ffe83a8 3ffe83a8  
3ffffee0:  3fff022f 3fffc6fc 3ffefe2c 00000000  
3ffffef0:  402104df 3ffed5a0 3ffed678 066462a1  
3fffff00:  00000001 40228720 3ffecf50 4010549c  
3fffff10:  402101bc 40210190 0000001c 401006d8  
3fffff20:  40211f55 40210190 00000064 40107208  
3fffff30:  3ffecfec 40211f1c 3ffed5b8 3ffee3a0  
3fffff40:  40210186 3ffececc 3ffee3a0 00000000  
3fffff50:  4021012d 3ffed5b8 3ffee3a0 066462a1  
3fffff60:  402105e0 40222f29 3ffee378 3ffee3a0  
3fffff70:  40222f29 60000600 00000000 40222f29  
3fffff80:  40222f6e 3fffdab0 00000000 3fffdcb0  
3fffff90:  3ffee3b0 3fffdab0 00000000 40202e8f  
3fffffa0:  40000f49 40000f49 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld
Setup done
scan start

xhdix, 2016-12-16:
WiFi.softAPdisconnect(true);
WiFi.disconnect();
WiFi.mode(WIFI_STA);

memina, 2016-12-31:
Hi xhdix, 
I tested this codes but not working with arduino. 

New info:
Send example blink code it working,
then send example wifi code it working, connected to wifi,
but power off nodemcu devkit and power on not connect to wifi, it saying connecting...

Help for this issue.

devyte, 2017-10-05:
@memina instead of a huge delay after a scan, you should check elapsed time with millis(), and if it is over e.g.: 5000, then execute a scan. In general, only short delays should be used in code, because of timing requirements for servicing the wifi stack.
You are also not calling WiFi.begin() anywhere, so it's no wonder you aren't connecting to anything.
Closing as user error.

closed 2017-10-05 by devyte