esp8266/Arduino #2765: SPI SCK-Pin is always high

boaschti, 2016-12-14:
### Basic Infos

#### Hardware
Hardware:			ESP8266 ESP12 D1 Mini NodeMcu Lua WeMos
Core Version:      	2.3.0

http://www.ebay.de/itm/122220217781?_trksid=p2057872.m2749.l2649&ssPageName=STRK%3AMEBIDX%3AIT

### Description

Hi,

I've got a problem to comunicate via SPI to my RFM69 module.
My logic analyser tells me that there is no sck on Pin D5. The pin is alway high.
Mosi (D7) sends data, CS (D8) looks like CS...
I dont think that the pin is damaged because at startup it become low and then high again.
Can anybody help me?

Best regards S.Weisser

### Settings in IDE

Arduino 1.6.13
Module:  			WeMos D1 R2 & Mini
Flash Size: 		4MB/3MB
CPU Frequency:		80Mhz
Flash Mode:			?
Flash Frequency:	?
Upload Using:		SERIAL
Reset Method:		?

### Sketch

https://github.com/bbx10/nanohab

devyte, 2017-10-05:
@boaschti is this issue still valid? Have you tried latest git?

boaschti, 2017-10-05:
Hi I tested another module. It works. But I cannot explain the high low states of that pin.

HasanNO, 2018-04-10:
This can be taken LOW! Just write to it somthing like this: digitalWrite(D14, LOW);

earlephilhower, 2019-07-25:
Closing since it seems like a HW issue for a particular board several years ago.  If there are ongoing SPI issues with 2.5.2 or git head, please do open a new issue so we can look into it.

closed 2019-07-25 by earlephilhower