esp8266/Arduino #2768: ESP8266 Wake up issue after DEEP SLEEP (DEVICE HANGS)

KVK012, 2016-12-16:
Hello,

My ESP8266 12E is waking up properly most of the times but sometimes it is throwing some hex data continuously. Sometimes throwing garbage value too. I.e in Hang state.

I am thinking that is because of reset issue.

Hardware configurations are as below
I have shorted GPIO16 to reset pin.
GPIO15 is pulled down with 10K resistor
GPIO0 and GPIO2 are pulled up with 10K resistor.
One 100nf capacitor connected at reset pin. 
(As per hardware design of node mcu)

My device is running on two AA batteries. One 4.7uF and one 100nF capacitor connected across input voltage i.e at 3V and GND.

I am facing same issue most of times when I remove and insert battery again into socket.

Please guide.

vlast3k, 2016-12-16:
have a look on this video
https://www.youtube.com/watch?v=IYuYTfO6iOs
for running on batteries
i would try 300+uF cap
AA cells have high internal resistance and their voltage may fall below 2.4v during the initial power peaks

KVK012, 2016-12-19:
Thanks, very informative video.
I have tried with 470uf cap but still issue persist. Battery voltage is ~2.6 volt.

Datasheet state that operating voltage is 3 to 3.6V, Does it mean I cannot use AA battery(1.5+1.5)?
Which one is safe working voltage range?

vlast3k, 2016-12-19:
i believe 2.6 is about the minimum at which it can operate. So with two fresh AA batteries it should work well. I haven't tested personally but the video says that using some 1000 uF electorlytic cap would absorb the current spikes when the ESP is sending data. During those spikes the battery voltage may fall below 2.5v and cause the ESP to reset

jourjine, 2016-12-19:
range for work: range 2.0 - 4.2V (SOC work on 1.85 but flash not)
range for flash: 3 - 4.2V
Alkaline AAA|AA battery, voltage range = 0.8 - 1.5V
NiMH AAA|AA rechargeable battery, voltage range = 0.8 - 1.145 V
Li-Ion rechargeable battery, voltage range = 2.5 - 4.2V or 3 - 4.35V
LiFeRo4 rechargeable battery, voltage range = 2.0 - 3.6V or 2.5 - 3.6V

so, two AA has too low end voltage.

Best solution for power supply: 1x LiFeRo4 or 3x NiMH rechargeable battery

devyte, 2017-10-05:
@KVK012 during boot there will be power spikes that cause voltage dips. Try your code with a stable high power supple. If it works, the issue is obviously the batteries. I use a 5V20A supply for development to assure correct code functionality. That way I know that any resets during development are not due to power supply, and once development is done, odd resets with the final power supply are likely due to a power issue.
Off-topic, closing, see #3655 .

closed 2017-10-05 by devyte