esp8266/Arduino #2770: Crashes after a few seconds when using delay() in main loop along with timer0 ISR

AlessandroAU, 2016-12-17:
Crashes after a few seconds when using delay() in main loop along with timer0 ISR.

If there are is no delay(); in main loop it runs fine. 

Comment out delay(); and it stop crashing.

Decoding 24 results
0x40104084: lmacTxFrame at ?? line ?
0x40100599: calloc at C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\cores\esp8266\umm_malloc/umm_malloc.c line 1684
0x401014a6: spi_flash_read at ?? line ?
0x402164d8: pm_get_sleep_type at ?? line ?
0x40100f35: ets_timer_disarm at ?? line ?
0x402165e0: pm_get_sleep_type at ?? line ?
0x40100590: calloc at C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\cores\esp8266\umm_malloc/umm_malloc.c line 1684
0x401079ac: pvPortZalloc at C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\cores\esp8266/heap.c line 54
0x40216051: pm_set_sleep_time at ?? line ?
0x4021652e: pm_get_sleep_type at ?? line ?
0x402165f3: pm_get_sleep_type at ?? line ?
0x402094f5: ets_timer_handler_isr at ?? line ?
0x4020953a: ets_timer_handler_isr at ?? line ?

Exception (0):
epc1=0x40201ed4 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3ffffc70 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffe10:  40104084 3ffed138 3fffc248 4000050c  
3ffffe20:  40100599 00000030 3fffc248 4000050c  
3ffffe30:  40004b1c 00000030 00000016 ffffffff  
3ffffe40:  401014a6 003fc000 3fff0d7c 000002f4  
3ffffe50:  3feffe00 fffffeff 00000000 00e8a101  
3ffffe60:  fffdffff 3fffc6fc 3ffe8db4 3fff0d7c  
3ffffe70:  000002f4 003fc000 60000600 00000030  
3ffffe80:  402164d8 00000030 00000013 ffffffff  
3ffffe90:  00000000 00000000 0000001f 40100f35  
3ffffea0:  4000050c 3ffe95f4 3fffc248 4000050c  
3ffffeb0:  40000f68 00000030 0000001c ffffffff  
3ffffec0:  40000f58 00000000 00000020 00000000  
3ffffed0:  00000013 402165e0 3ffe8db4 00000000  
3ffffee0:  ffffffff 3ffe9694 3ffe8db4 3fffdab0  
3ffffef0:  00000000 3fffdcb0 3ffe8de0 00000030  
3fffff00:  ffffffff 00000000 3ffe99b1 60000600  
3fffff10:  0101da9b 000002f4 3ffe9540 40100590  
3fffff20:  3ffe9630 3ffe8dd0 3ffe9540 401079ac  
3fffff30:  40216051 3ffe9540 3ffe8dd0 0101da9b  
3fffff40:  3fff0d7c 00001000 4021652e 00000008  
3fffff50:  3ffe8dd0 3ffe95f4 402165f3 3ffe95f4  
3fffff60:  3ffe8dd0 0032f8e1 3ffe8dd0 60000600  
3fffff70:  402094f5 3ffe95f4 3ffe8dd0 0101a178  
3fffff80:  4020953a 3fffdab0 00000000 3fffdcb0  
3fffff90:  3ffe8de8 40000f65 3fffdab0 00000000  
3fffffa0:  40000f49 40000f49 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,7)

 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			WeMos D1 mini
Core Version:      	2.3.0-rc2 -- also crashes in other versions

### Description

Crashes after a few seconds when using delay() in main loop along with timer0 ISR.

If there are is no delay(); in main loop it runs fine. 

Comment out delay(); and it stop crashing.

### Settings in IDE

Module:  			WeMos D1 R2 & mini
Flash Size: 		4MB/3MB SPIFFS
CPU Frequency:		160/80Mhz

CODE:
///////////////////////////////////////////////////////////////////////
unsigned long timer0_count = 0; //gobal 32bit timer
///////////////////////////////////////////////////////////////////////
void timer0_ISR (void) {

  timer0_count = timer0_count + 1; //incr. count

  if (timer0_count == 100) {
    timer0_count = 0;
  }
  timer0_write(ESP.getCycleCount() + 2000);
}
////////////////////////////////////////////////////////////////////////
void setup() {
  Serial.begin(115200);
  ///////////Setup Timer0 for Sampling////////////////////////////////
  timer0_isr_init();
  timer0_attachInterrupt(timer0_ISR);
  timer0_write(ESP.getCycleCount() + 2000); // Adds a small delay  to the first timer0 call.
  /////////////////////////////////////////////////////////////////
}

void loop() {
  delay(1000);
  Serial.println("hello");
}
///////////////////////////////////////////////////////////////////////

devyte, 2017-10-05:
@AlessandroAU I'm not sure here, but I think the ISR function has to have the ICACHE_IRAM_ATTR decorator.
Please report back here if that works.

devyte, 2017-12-29:
Closing due to no feedback in almost 3 months, and age.

closed 2017-12-29 by devyte