esp8266/Arduino #2771: SSL error 40, SSL handshake failure

kidd86, 2016-12-17:
### Basic Infos

#### Hardware
Hardware:		Wemos D1 mini
Core Version:      	2.3.0-master

### Description

Problem description

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB
CPU Frequency:	80Mhz
Flash Mode:		qio
Flash Frequency:	80Mhz
Upload Using:		SERIAL
Reset Method:		nodemcu

### Sketch

```cpp

#include <ESP8266WiFi.h>
#include <WiFiClientSecure.h>

#define WLAN_SSID    "H********8"
#define WLAN_PASS    "T***********2"

#define SERVER_HOST  "apiv2.boltbutton.com"
#define SERVER_PORT  443

// WiFiFlientSecure for SSL/TLS support
WiFiClientSecure client;

//HTTPS Port fingerprint
const char* fingerprint = "‎07 59 1b 25 a7 36 7a 95 b0 ef 2d ae b2 58 d4 50 a0 89 d4 bb";

void setup() {
  Serial.begin(115200);
  delay(10);
  Serial.setDebugOutput(true);
 
  // Connect to WiFi access point.
  Serial.print("Connecting to ");
  Serial.println(WLAN_SSID);
  
  delay(1000);
  
  WiFi.begin(WLAN_SSID, WLAN_PASS);
  delay(2000);
  
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("WiFi connected");
  Serial.print("IP address: "); Serial.println(WiFi.localIP());  
}

void loop() {
  
  const char* host = SERVER_HOST;
  const int httpPort = SERVER_PORT;
  
  // create a secure connection using WiFiClientSecure
  WiFiClientSecure client;
  
  if (client.connect(host, httpPort)) {
    
    // verify the signature of the ssl certificate
    if (client.verify(fingerprint, host)) {
      Serial.println("ssl cert matches");
    } else {
      Serial.println("ssl cert mismatch");
    }
  }
  else {
    Serial.println("Connect fail before verification");
  }
  
  delay(60000);
}

```

### Debug Messages

```
WiFi connected
IP address: X.X.X.X
:ref 1
please start sntp first !
State:	sending Client Hello (1)
:sent 72
:rn 7
:rcl
:abort
:rd 5, 7, 0
:rdi 7, 5
:rd 2, 7, 5
:rdi 2, 2
:c0 2, 7
Alert: handshake failure
Error: SSL error 40
Alert: unexpected message
Alert: close notify
Connect fail before verification

```

JohnAntonusMaximus, 2017-01-09:
Any progress on this?

Daemach, 2017-01-17:
From a conversation with @igrr:

AxTLS library doesn't support cipher suites listed there. IIRC, the two cipher suites supported in 2.3.0 are TLS_RSA_WITH_AES_128_CBC_SHA and TLS_RSA_WITH_AES_256_CBC_SHA. Latest git version adds a couple other cipher suites, but also not the ones listed for this website (axTLS only does RSA).

You need to configure the webserver to allow one of these two cipher suites, or set up an HTTPS termination proxy if this server is outside of your control.

kylefoo, 2017-01-24:
Your server host already supported TLS_RSA_WITH_AES_128_CBC_SHA and TLS_RSA_WITH_AES_256_CBC_SHA. Maybe TLS_RSA_WITH_AES_256_CBC_SHA256 (0x3d) is the one you are missing

copercini, 2017-06-20:
Getting same error with AWS IoT

The most strange is Amazon seems support the AxTLS cipher suites (http://docs.aws.amazon.com/iot/latest/developerguide/iot-security-identity.html) like:

AES128-SHA
AES256-SHA​
AES128-SHA256
AES256-SHA256

sej7278, 2017-06-22:
So 0x3d is the best we can get? Is there a reason we're not using wolfssl - just the bsd vs gpl licensing?

copercini, 2017-06-22:
Maybe port Mbedtls from ESP32

sej7278, 2017-06-23:
oh is that what we're using on esp32? i might switch to that then, probably pushing the esp8266 a bit. is that part of the arduino core though, or just for the idf or whatever its called (sdk)?

wilburgersoftware, 2018-03-06:
Hi, i also get the the same error connecting wo amazonws.
`[hostByName] request IP for: xxx
[hostByName] Host: xxx IP: 52.11.111.33
:ref 1
State:	sending Client Hello (1)
:wr 96 96 0
:wrc 96 96 0
:sent 96
:rn 7
:rcl
:abort
:rd 5, 7, 0
:rdi 7, 5
:rd 2, 7, 5
:rdi 2, 2
:c0 2, 7
Alert: handshake failure
Error: SSL error 40
Alert: unexpected message
Error: SSL error 40
Alert: close notify
connection failed
:ur 1
:del
`

devyte, 2018-05-29:
BearSSL is merged in #4273 , with alternate BearSSL::WiFi* classes. Although axtls-based classes are still available and even the default, they are planned for deprecation and then retirement, hence won't be fixed. Any issues with BearSSL-based classes should be reported in new issues.
Closing.

closed 2018-05-29 by devyte

arnolde, 2018-09-19:
How can I actually use this newly merged version with BearSSL? I also have the "ssl error 40" problem, I only include ESP8266WiFi.h and ESP8266HTTPClient.h and nothing else, adding the line "using namespace BearSSL;" did not help, I have not installed any libraries via the AdruinoIDE library manager that relate to ssl or http, the esp8226httpclient.h seems to be "just there"... to to use the newest ones instead?

d-a-v, 2018-09-19:
When #4980 is merged, that will be easy.
An example will be provided: [link](https://github.com/esp8266/Arduino/pull/4980/files#diff-46402b56934d1fb76007c6229f811545)

arnolde, 2018-09-19:
I'm sorry for bothering you again but when will that happen and what steps exactly do I have to take to make sure the changes are effective on my development computer? Is there anything I can do right now? My whole project is stalled since 2 days because of this cipher mismatch issue.

d-a-v, 2018-09-19:
You can try this PR, it is up to date with current master.
You need to use git (how to obtain the git version is described in the documentation).
Please address your questions directly in #4980 until it is merged.
Please report (positive, negative) in #4980.

To use it right now:

```
git fetch origin pull/4980/head:ssl
git checkout ssl
```

Revert to master:

```
git checkout master
```

in case of updates in the PR, do this before restarting the above.

```
git checkout master
git branch -D ssl
```