esp8266/Arduino #2772: warning: espcomm_sync failed error: espcomm_open failed error: espcomm_upload_mem failed error: espcomm_upload_mem failed

hackernikhilraj, 2016-12-17:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.md

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.md

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			ESP8266
Core Version:      	?2.1.0-rc2?

### Description

Problem description

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB/1MB
CPU Frequency:		80Mhz
Flash Mode:			DIO
Flash Frequency:	40Mhz
Upload Using:	          DISBLE
Reset Method:		BOTH ck / nodemcu

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
messages here
```

jyotirajsharma, 2017-01-03:
Did you get it working now? I am gettinh similar errors. How to resolve it ? Most of the forums are saying we need to use 3.3v external power supply but i am using arduino 3.3v power supply? Can tjis be the reason?

hackernikhilraj, 2017-01-03:
arduino didn't supply that much of current that is required for  ESP8266 to work. arduino o/p current is less than that .

jyotirajsharma, 2017-01-03:
Was it flashed properly after external power supply ? I will also try then. And I also saw in some of the places, esp8266 should be flashed standalone using usb-ttl cable. I am connecting arduino to esp and directly uploading the code.
Wil following connections work ?
Arduion     esp
Tx--------------rx
Rx--------------tx
Vcc-(3.3v)---------vcc(note: will change to ext.power
Vcc(3.3v)---------ch_pd note: will use ext.power
Gpio_0---------GNd note:this will enter flash mode
Gnd-------------gnd

devyte, 2017-10-05:
@hackernikhilraj the espcomm open failed issue is documented. If what is in the docs doesn't work, you either have bad wiring, or a bad board.
Closing.

closed 2017-10-05 by devyte