esp8266/Arduino #2776: DS3231 and  BMP180 with esp8266

msbrigittte, 2016-12-18:
I understand the I2C bus, but I have trouble with connecting DS3231 and  BMP180 to eps8266 at the same time.

For the DS3231 I used this example:
http://domoticx.com/esp8266-wifi-rtc-tijdklok-ds1307-nodemcu/

And this one for the BMP180:
http://www.esp8266.com/viewtopic.php?p=24406

they work fine alone, but I have trouble connecting both divices to I2C at the same time. I think this is because of the differences in the init functions. And yes, I have changed the sda and scl pins to match (use one I2C).

freeck, 2016-12-19:
Both libraries call the wire.begin()-function. I removed the wire.begin()-function out of both libraries and put one in the setup() function.
I use both devices and they function as it should (so far :)). I would suggest not to use gpio0 and gpio2 as I2C wires, they could influence the behaviour of esp8266 at startup or during a watchdog-reset.....

devyte, 2017-10-05:
@msbrigittte The issue is a clash in initialization between the two libs. I think the proposed solution is correct: the begin() method should be called only once.
Closing due to 3rd party libs.

closed 2017-10-05 by devyte