esp8266/Arduino #2779: Unable to connect to wireless after reading the SSID and Pass using serial function

abdel-malek, 2016-12-19:
hello am having an issue with my ESP8266

when i burn the code in my sketch every thing works fine, but when i try to read it using the serial port the ESP will not connect .

this is my code : 

#include <ESP8266WiFi.h>
char ssid[28];
const char* password = "MyPassword";
const char* ssidPointer;
String ssidString;
void setup() {
  Serial.begin(9600);
  ////receive ssid as serial
  Serial.print("Enter user name: ");
  ssidString=Serial.readString();
  
  ///wait until enter ssid
  while(ssidString == "")  
  {
      ssidString=Serial.readString();
      delay(200);
  }
  ////convert serial to char array
  for(int i=0;i<4;i++)
  {
    ssid[i]=ssidString[i];
  }
  
  ///convert char array to pointer array
  ssidPointer = ssid;
 
  ////connect to wifi
  Serial.print("Connecting to ");
  Serial.println(ssidPointer);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssidPointer, password);
  int wifi_ctr = 0;
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("WiFi connected");
  Serial.println("IP address: " + WiFi.localIP());
}
void loop() {
}

tablatronix, 2016-12-26:
Make sure there are no CR LF or /0 in there ?

devyte, 2017-10-05:
@abdel-malek the loop that iterates only 4 times makes no sense, and even if it did, it doesn't null-terminate ssid.
Closing due to user error.

closed 2017-10-05 by devyte