esp8266/Arduino #2781: client.write() hangs indefinitely

sezeryalcin, 2016-12-20:
#### Hardware
Hardware:			ESP-12E Wemos D1 mini
Core Version:      	2.3.0-rc2/

### Description

New method client.write() looks hanging without a reason. Code and output is below.
This looks like a problem when I refresh the page. It works fine on initial loads .

### Settings in IDE

Module:  			Wemos D1 mini
Flash Size: 		4MB
CPU Frequency:		80Mhz
Flash Mode:			4M/3M FS
Flash Frequency:	?
Upload Using:		 SERIAL
Reset Method:		?

### Sketch
#if SERIAL_DEBUG == 1
     Serial.print("Static file: ");
     Serial.println(req);
     Serial.println(contentType);
#endif

           File f = SPIFFS.open(req, "r");
           Serial.printf("File Size=%d\r\n", f.size());

           client.print("HTTP/1.1 200 OK\r\nContent-Type: " + contentType + "\r\n\r\n");
           Serial.printf("Header sent");
           if (client.status() != CLOSED)
           {
              client.write(f);
              Serial.printf("File sent");
              client.flush();
           }
           else
              Serial.printf("File NOT sent");

           f.close();

### Debug Messages

Static file: /static/bootstrap.300.min.css
text/css
File Size=121200
Header sent
<NO MORE LINES>

sezeryalcin, 2016-12-21:
I see that css file is one line of 120KB+ data. I wonder if that has anything to do with it.

sezeryalcin, 2016-12-21:
Found that my code was not yielding enough

closed 2016-12-21 by sezeryalcin

johnnytolengo, 2019-01-16:
Hi @sezeryalcin how do you solve it?