esp8266/Arduino #2782: I can not connect the ESP8266 ESP-01 to the Wi-Fi network

Marcel2108, 2016-12-20:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.md

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.md

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:	ESP8266	 ESP-01
Core Version:   2.3.0

### Description

Problem description

### Settings in IDE

Module:  		        Generic ESP8266 Module
Flash Size: 		512K(64KSPIFFS)
CPU Frequency:	 80Mhz
Flash Mode:		DIO
Flash Frequency:	40Mhz
Upload Using:		WIFI 
Reset Method:		ck

### Sketch

#include <ESP8266WiFi.h> // Importa a Biblioteca ESP8266WiFi

#define DEBUG true //mostrar as mensagens do ESP8266 no Serial Monitor

// WIFI
const char* SSID = "TP-LINK_D834A8"; // SSID da Rede
const char* PASSWORD = "eneagrama"; // Senha da Rede

// Prototypes das funções utilizadas no decorrer do Firmware
void initWiFi();

void setup() {
  Serial.begin(115200);
}

// Função para iniciar a Conexão com a rede WiFi
void initWiFi() {
  delay(1000);
  Serial.println();
  Serial.print("Conectando-se em: ");
  Serial.println(SSID);

  WiFi.begin(SSID, PASSWORD); // Conecta na Rede Wireless
  while (WiFi.status() != WL_CONNECTED) {
    delay(100);
    Serial.print(".");
  }
  Serial.println();
  Serial.print("Conectado na Rede ");
  Serial.print(SSID);
  Serial.print(" | IP ");
  Serial.println(WiFi.localIP()); // Mostra o IP que foi atribuido para o ESP8266
}

void reconectar() {
  while (WiFi.status() != WL_CONNECTED) {
    delay(100);
    Serial.print(".");
  }
}
void loop() {
  reconectar(); // Caso o ESP perca conexão com a rede WiFi, ele tenta se reconectar a rede.
  Serial.print("Conectado na Rede: ");
  Serial.println(SSID);
  delay(1000);
}
```

### Debug Messages

.scandone
no SSID found, reconnect after 1s
wifi evt: 1
STA disconnect: 201
.reconnect

messages here

Não consigo ver no monitor serial do Arduino o IP que ele recebe do roteador. Eu so recebo a mensagem .scandone
no SSID found, reconnect after 1s
wifi evt: 1
STA disconnect: 201
.reconnect

Meu roteador usa a criptografia WPA2. Como posso resouver esse problema?

Marcel2108, 2016-12-20:
I can not see the Arduino serial monitor the IP it receives from the router. I just get the message .scandone
No SSID found, reconnect after 1s
Wifi evt: 1
STA disconnect: 201
.reconnect

My router uses WPA2 encryption. How can I solve this problem?

drkimbo, 2016-12-21:
may be tihs two factors
https://support.google.com/chromecast/answer/3213086?hl=es
AND
https://support.google.com/chromecast/answer/3213084?hl=es

devyte, 2017-10-05:
@Marcelphilippeandrade you are not calling initWifi() anywhere, which is the function that actually calls WiFi.begin().
Closing due to user error.

closed 2017-10-05 by devyte