esp8266/Arduino #2787: rom_i2c_writeReg_Mask was not declared in this scope

probonopd, 2016-12-22:
### Basic Infos

#### Hardware
Hardware:			WeMos D1 mini
Core Version:      	2.3.0 (Arduino-1.6.11.esp2.3.0-x86_64.AppImage)

### Description

`esp8266_peri.h:32:134: error: 'rom_i2c_writeReg_Mask' was not declared in this scope` when trying to compile [this sketch](https://gist.githubusercontent.com/probonopd/c114ea42ea6fef5b6b6b6118ae49ebf8/raw/0cb4f85513c5cf7338542c0b51bc7245443affc3/esp8266_bosecontrol.ino)

### Settings in IDE

Module:  			WeMos D1 mini
Flash Size: 		4MB/1MB
CPU Frequency:		80Mhz

### Sketch

[Here](https://gist.githubusercontent.com/probonopd/c114ea42ea6fef5b6b6b6118ae49ebf8/raw/0cb4f85513c5cf7338542c0b51bc7245443affc3/esp8266_bosecontrol.ino)

### Debug Messages

```
In file included from hardware/esp8266/esp8266/cores/esp8266/Arduino.h:38:0,
                 from sketch/TestEspApi.ino.cpp:1:
/home/me/Desktop/TestEspApi/TestEspApi.ino: In function 'void initI2S()':
hardware/esp8266/esp8266/cores/esp8266/esp8266_peri.h:32:134: error: 'rom_i2c_writeReg_Mask' was not declared in this scope
 #define i2c_writeReg_Mask(block, host_id, reg_add, Msb, Lsb, indata)  rom_i2c_writeReg_Mask(block, host_id, reg_add, Msb, Lsb, indata)
                                                                                                                                      ^
hardware/esp8266/esp8266/cores/esp8266/esp8266_peri.h:33:55: note: in expansion of macro 'i2c_writeReg_Mask'
 #define i2c_writeReg_Mask_def(block, reg_add, indata) i2c_writeReg_Mask(block, block##_hostid,  reg_add,  reg_add##_msb,  reg_add##_lsb,  indata)
                                                       ^
hardware/esp8266/esp8266/cores/esp8266/esp8266_peri.h:752:43: note: in expansion of macro 'i2c_writeReg_Mask_def'
 #define I2S_CLK_ENABLE()                  i2c_writeReg_Mask_def(i2c_bbpll, i2c_bbpll_en_audio_clock_out, 1)
                                           ^
/home/me/Desktop/TestEspApi/TestEspApi.ino:171:3: note: in expansion of macro 'I2S_CLK_ENABLE'
   I2S_CLK_ENABLE(); ////////////////////////// / same error
   ^
exit status 1
Error compiling for board Generic ESP8266 Module.
```

probonopd, 2016-12-22:
Additional information from https://gitter.im/esp8266/Arduino:

> @probonopd
it's unclear to me why i keep getting esp8266_peri.h:32:134: error: 'rom_i2c_writeReg_Mask' was not declared in this scope...
>
> @me-no-dev
maybe it was declared and now it's not. try adding one
>
> @me-no-dev
hmm you are correct it's no longer defined anywhere as what arguments it takes
>
> @probonopd
can we add that back?
>
> @me-no-dev
you know what... I can't find it either

cnlohr, 2016-12-22:
This is confusing.  I thought the Arduino environment used the standard ld script.

```
PROVIDE ( rom_i2c_writeReg = 0x400072d8 );
PROVIDE ( rom_i2c_writeReg_Mask = 0x4000730c );
```

probonopd, 2016-12-22:
Not sure what this all really means, but i find `0x4000730c` in two places in my Arduino installation, one of which seemingly related to the bootloader and the other to "eagle", whatever that means:

```
~$ grep -r 0x4000730c .
usr/bin/hardware/esp8266/esp8266/bootloaders/eboot/rom.ld:PROVIDE ( rom_i2c_writeReg_Mask = 0x4000730c );
usr/bin/hardware/esp8266/esp8266/tools/sdk/ld/eagle.rom.addr.v6.ld:PROVIDE ( rom_i2c_writeReg_Mask = 0x4000730c );
```

me-no-dev, 2016-12-22:
yes, but has no definition of what arguments it takes. I guess that might be the problem, though I did not see any removed definitions. Could that be defined in the toolchain?

@probonopd you find it in the linkers for the sketch and for the bootloader

cnlohr, 2016-12-22:
Oh, I have a sort-of definition in nosdk8266... but it's not that good https://github.com/cnlohr/nosdk8266/blob/master/include/nosdk8266.h

```
void rom_i2c_writeReg( int reg, int hosid, int par, int val ); 
```
I do not have a definition for the mask.

Aaland, 2016-12-27:
Hi
This is from  an old Espressif IR transmitter demo code but in the current demo they have removed it.
Maybe some smart person understand it. Their updated version can be found at:
[http://bbs.espressif.com/viewtopic.php?f=31&t=1199](url)

`#define i2c_bbpll         						0x67
#define i2c_bbpll_en_audio_clock_out        	4
#define i2c_bbpll_en_audio_clock_out_msb    	7
#define i2c_bbpll_en_audio_clock_out_lsb    	7
#define i2c_bbpll_hostid       					4

	//ENABLE I2S CLK SOURCE
	rom_i2c_writeReg_Mask(i2c_bbpll, i2c_bbpll_hostid, i2c_bbpll_en_audio_clock_out, i2c_bbpll_en_audio_clock_out_msb, i2c_bbpll_en_audio_clock_out_lsb, 1);
`

probonopd, 2016-12-28:
Where would I have to put this? The sketch? Somewhere else?

Aaland, 2016-12-28:
I just thought my entry would give some more leads to the parameters for the function. I used the function in my version of an IR transmitter and I think it worked even if I haven't used it. To be able to call the function in C++ I declared it as an external C function. 
`extern "C" {
void rom_i2c_writeReg_Mask(int, int, int, int, int, int);
}
The linker then uses the address from the ld script. If you write C code as the Espressif example you don't need any declaration. Specifying the parameters as integers was just guessing. See also the old example script in the attached file.

[ir_tx.zip](https://github.com/esp8266/Arduino/files/675803/ir_tx.zip)

`

probonopd, 2016-12-28:
@Aaland awesome, this seems to have done the trick, the sketch now compiles:

```
#ifdef ESP8266
extern "C" {
#include "user_interface.h"
#include "i2s_reg.h"
#include "slc_register.h"
#include "esp8266_peri.h"
void rom_i2c_writeReg_Mask(int, int, int, int, int, int);
}
#endif
```

closed 2016-12-28 by probonopd

probonopd, 2016-12-28:
While I am able to generate a signal at the desired frequency now (I checked with RTLSDR), I don't seem to be able to switch on and off this frequency fast enough for on-off-keying a NEC remote signal which requires 564µs "frequency on" pulses, at least using this code:

```
const int raw_NEC2_186_85_1[] = { 9024,-4512,564,-564,564,-1692,564,-564,564,-1692,564,-1692,564,-1692,564,-564,564,-1692,564,-1692,564,-564,564,-1692,564,-564,564,-1692,564,-564,564,-1692,564,-564,564,-1692,564,-564,564,-564,564,-564,564,-564,564,-564,564,-564,564,-564,564,-564,564,-1692,564,-1692,564,-1692,564,-1692,564,-1692,564,-1692,564,-1692,564,-38628 };

void loop()
{
   int i, *p;
    
for(i=0; i<(&raw_NEC2_186_85_1)[1]-raw_NEC2_186_85_1; i++){
  if(raw_NEC2_186_85_1[i] > 0) {
    SET_PERI_REG_MASK(I2SCONF, I2S_I2S_TX_START);
  } else {
    CLEAR_PERI_REG_MASK(I2SCONF,I2S_I2S_TX_START);
  }
  delay(abs(raw_NEC2_186_85_1[i])/550); 
  // I cannot seem to be able to make the delay shorter without the signal being lost.
  // When I change the divisor from 550 to 570 it stops working. It would have to be 1000
}

  // Wait one second
  CLEAR_PERI_REG_MASK(I2SCONF,I2S_I2S_TX_START);
  delay(1000);
```

I am sure I am doing something wrong since the NTSC signals require much finer grained timings...

Aaland, 2016-12-28:
When you divide an integer value of 564 with 570 the result is 0. You need to use delayMicroseconds to get us accuracy. Otherwise have a look at the IR transmitter example from Espressif to see how to use timers to handle the on/off switching. The link is in my first entry.

Aaland, 2016-12-28:
Maybe I also should mention that I think you will 'kill' the ESP if you use DelayMicroseconds in a loop because I don't think it will yield and give processor time for the background tasks handling WiFi.

probonopd, 2016-12-28:
@Aaland `delayMicroseconds` totally did the trick for me, at least for a proof-of-concept it is working now. But I understand I should use timers. I find the Espressif example hard to read though (as someone coming from the Arduino rather than ESP camp).