esp8266/Arduino #2789: HTTPClient with TLS : Exception 29

olileger, 2016-12-22:
### Basic Infos
I got `Fatal exception 29(StoreProhibitedCause)` when trying to connect to secured public endpoint over HTTPS.
The public endpoint is and Azure IoT Hub URI.

I don't have any error when connecting to my local secured endpoint (self-signed certificate) through HTTPS.
Unforetunately the ESP Exception decoder tool doesn't recognize my stack trace so I'm not able to understand the issue and fix it.
Is anyone able to decode it for me please ? Very appreciated :)

Is there any guidelines & FAQ related to the TLS support ? Something like cipher suite limitation, TLS version supported or any other limits.

#### Hardware
Hardware:			ESP8266 (WeMos D1 Mini)
Core Version:      	2.3.0

### Settings in IDE
Module:  			WeMos D1 Mini
Flash Size: 		4MB
CPU Frequency:		80Mhz

### Sketch
```
HTTPClient client;
client.begin(/* hostname */,  /* TLS Fingerprint */);
client.POST(/* payload str */);
/* exception happening here */
```

### Debug Messages & Stack trace
```
Fatal exception 29(StoreProhibitedCause):
epc1=0x4000e1b2, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000

Exception (29):
epc1=0x4000e1b2 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3fff4a50 end: 3fff5150 offset: 01a0

>>>stack>>>
3fff4bf0:  00000040 00000082 3fffae2c 40240c59  
3fff4c00:  3fff9c74 3fff708c 00000000 00000041  
3fff4c10:  00000040 0000003f 00000000 00000041  
3fff4c20:  3fffbd24 00000000 3fffae44 3fff7584  
3fff4c30:  3fffbe20 3fffae5c 3fff708c 3fff708c  
3fff4c40:  00000040 3fff708c 3fffae2c 3fff708c  
3fff4c50:  00000040 3fff708c 3fffae2c 4024178e  
3fff4c60:  3fff728c 0000003f 5be1fe63 00000020  
3fff4c70:  3fff9c6c 3fff708c 3fff75b4 00000020  
3fff4c80:  3fff9c6c 3fff708c 00000010 40241a74  
3fff4c90:  3fffadfc 3fff72bc 3fff75b4 00000001  
3fff4ca0:  00000001 3fff75b4 3fff7899 4024084c  
3fff4cb0:  00000100 3fff725c 3fff7897 00000000  
3fff4cc0:  00000100 3fff725c 3fff7897 40242a75  
3fff4cd0:  3fff4d00 00000000 000000d0 00000030  
3fff4ce0:  89b5fbab dfcb6a89 dd2b7e3a 00000004  
3fff4cf0:  00000004 3fff6e4c 3fff7891 4023eb80  
3fff4d00:  05f80203 cb1a7b3f feecd66d 1586188f  
3fff4d10:  936c5130 e6ddd4fd 54943175 fcca7ed8  
3fff4d20:  71f06a7a 82eb5f06 8320193c fbab7d1a  
3fff4d30:  00000004 3fff84e8 3fff6e4c 4023ef81  
3fff4d40:  c2109d2c 564cf7ff 59ce65ab f7932814  
3fff4d50:  dbb4ac6f 52b8a48f c423f09b 99632428  
3fff4d60:  c13fadcf 3fff84e8 3fff745c 00000b80  
3fff4d70:  00000c11 3fff84ca 0000001d 00000004  
3fff4d80:  3fff6e4c 3fff84e8 00000004 00000004  
3fff4d90:  00000004 3fff84e8 3fff6e4c 4023e978  
3fff4da0:  00000000 3fff7891 3fff6e4c 4023ed20  
3fff4db0:  0000000c 3fff6d24 0000000d 401004d8  
3fff4dc0:  3fff4e10 0000000e 00000010 00000000  
3fff4dd0:  3fff72d4 3fff6d24 3fff6e4c 01000000  
3fff4de0:  3fff4e10 3fff6de4 3fff6e4c 4023ead0  
3fff4df0:  4021a154 00000000 3fff4130 00001387  
3fff4e00:  0000c2e8 3fff6de4 3fff6d3c 4021a7e5  
3fff4e10:  000001bb 3fff6f54 3fff6de4 40219b7e  
3fff4e20:  2bac4f0d 00000000 3fff4e60 3fff5bf4  
3fff4e30:  000001bb 3fff6de4 3fff6d24 4021ab39  
3fff4e40:  3ffeaa80 2bac4f0d 3ffeaa80 2bac4f0d  
3fff4e50:  3fff4fa0 00000000 3fff4f30 4021b754  
3fff4e60:  00000000 3fff4ef0 00000000 4010053d  
3fff4e70:  00000014 00000039 3fff4f30 4021bc34  
3fff4e80:  3fff25ac 00000018 3fff4068 3fff5030  
3fff4e90:  00000040 3fff4ef0 3fff4ef0 40221548  
3fff4ea0:  3ffe9048 00000039 3fff4ef0 40221597  
3fff4eb0:  3fff4040 00000260 3fff4ef0 3fff5030  
3fff4ec0:  3fff4fa0 3fff4f30 00000001 4021bce4  
3fff4ed0:  3fff4fa0 3fff4f30 00000001 4021bd02  
3fff4ee0:  00000001 3fff5030 3fff4efc 4020cb85  
3fff4ef0:  3fff5bf4 0000003f 00000039 00000000  
3fff4f00:  00000000 00000000 00000000 00000000  
3fff4f10:  00000000 3fff5b5c 0000000f 00000004  
3fff4f20:  00000001 41880000 00000000 4020cc52  
3fff4f30:  3fff70f4 3fff6de4 3fff6d24 0000000f  
3fff4f40:  0000000c 410001bb 3f001388 3fff6e04  
3fff4f50:  0000003f 0000003b 3fff5b44 0000000f  
3fff4f60:  00000005 3fff77a4 000000df 000000d1  
3fff4f70:  3fff631c 0000001f 00000011 3fff6f3c  
3fff4f80:  0000000f 00000000 00000000 00000000  
3fff4f90:  00000000 ffffffff 00000000 00000000  
3fff4fa0:  3fff6d54 0000003f 00000039 40203451  
3fff4fb0:  3fff2038 00000003 00000001 3fff25ac  
3fff4fc0:  0000c259 3fff1c70 3fff5030 4020cdec  
3fff4fd0:  3ffeb458 00000000 3fff5058 3fff25ac  
3fff4fe0:  3fff1c0c 3fff1968 3fff5030 4020f9c9  
3fff4ff0:  00000000 3fff5030 3fff5058 4022161a  
3fff5000:  3ffeb458 00000000 0000001f 3fff25ac  
3fff5010:  00000004 00000000 3ffeb458 3fff25ac  
3fff5020:  3fff24c2 3fff1c70 00000010 4020fb34  
3fff5030:  00000000 00000001 00000002 fffffffe  
3fff5040:  ffffffff 3fffc6fc 00000001 3fff63dc  
3fff5050:  0000000f 00000000 3fff6304 0000000f  
3fff5060:  00000000 00000000 00000000 00000000  
3fff5070:  00000000 00000000 0000010b 4010020c  
3fff5080:  3fff3c0c 3fff1c08 3fff50b0 4010068c  
3fff5090:  3fff4040 00000104 00000104 4010020c  
3fff50a0:  3fffdad0 00000000 3fff510c 4010068c  
3fff50b0:  3fff264c 00000000 3fff31f0 4021ad36  
3fff50c0:  3fffdad0 00000000 00000000 00000001  
3fff50d0:  3fff2a34 3fff1c20 00000000 4020fc3c  
3fff50e0:  6c696146 00000000 3fff4130 00000000  
3fff50f0:  3fffdad0 3fff4124 4022236c 3fff4124  
3fff5100:  3fffdad0 00000000 3fff1964 40212d44  
3fff5110:  3fffdad0 00000000 00000000 3fff4124  
3fff5120:  3fffdad0 00000000 3fff1964 40213258  
3fff5130:  3fffdad0 00000000 3fff411e 402223b8  
3fff5140:  feefeffe feefeffe 3fff4130 40100718  
<<<stack<<<
```

yaohaizh, 2016-12-23:
@olileger  
For ESP8266 work with Azure IoT Hub library(V1.0.7), there is a fix available(https://github.com/Azure/azure-iot-arduino/commit/32af3b65bcef653b6d83b4a4297d8dc4694330d4) but not published yet. 

You can use this change to make the code work at this time.  I'm also working on self-signed CA to azure iot hub but also failed.

olileger, 2016-12-23:
Hi :)

Of course I could rely on the Azure IoT Hub library but I would like to only rely on the ESP8266 lib.
Since I'm using only HTTP it should be pretty easy : HTTP POST with the right headers (Content-Type & Authorization) + the body.
I built a simulated device that works very well through PowerApps & Azure Functions but now I'd like to use my real devices.

By the way I also got an exception when trying to connect to a moackable.io endpoint so I guess it's not related to the endpoint itself but more on the type of certificate.

Olivier.

olileger, 2017-01-05:
I finally got my answer :)
Regarding this thread `https://github.com/esp8266/Arduino/issues/2075#issuecomment-222881980` where I could read from @igrr 

> Having ESP8266WebServer and HTTPS client and the same time likely means you are going to run out of memory.

This is the case on my configuration : a local webserver to setup my MCU and also trying to connect to an HTTPS endpoint.

@igrr : is there any way to have insight about how much RAM is used by the WiFiClientSecure library ?

pieman64, 2017-01-05:
@olileger from what I have seen if you turn on all ESP debugging in the IDE and select debug port of Serial it will show you all manner of things including RAM being used by the WiFiClientSecure library etc.

olileger, 2017-01-06:
@pieman64 interesting, do you have any pointers to article about that please ?

pieman64, 2017-01-06:
@olileger presumably you are using the Arduino IDE as this GitHub specifically relates to that.
I have just started to get my ESP to update Google Sheets and I was struggling with fingerprint mismatches.
Setting up debugging in the IDE gave me the details I required and I noticed the RAM data was also provided. It shows how much is available and how much is need for cert approval etc.

It's accessed from Tools menu in the IDE.

olileger, 2017-01-06:
@pieman64 I guess you are talking about the ESP Exception Decoder tool ?
Unfortunately it doesn't work on my station :(

SummerSun, 2017-03-16:
Print heap size when set up and after send a https request, turns out it takes about ~18K to make a https request.

devyte, 2017-10-05:
@olileger is this issue still valid with latest git?

supersjimmie, 2017-11-22:
So I guess I am running into the same issue.
I have a small setup working to fetch calendar items from Google calendar, using the WiFiClientSecureRedirect client. Free heap about 30k before the client starts. Works fine until now.

But when I add this same working code to my complete setup, which also contains several other components like a small webserver, I get a Exception (29) between client.connect() and waiting for it to become connected(). Free heap about 11k before the client starts.  :(

devyte, 2018-05-29:
BearSSL is merged in #4273 , with alternate BearSSL::WiFi* classes. Although axtls-based classes are still available and even the default, they are planned for deprecation and then retirement, hence won't be fixed. Any issues with BearSSL-based classes should be reported in new issues.
Closing.

closed 2018-05-29 by devyte