esp8266/Arduino #2792: How to get IP address from String

hoacvxd, 2016-12-24:
I have esp8266 with webserver. Now I want changr the IP address from web.

How to convert: String strIP = "192.168.4.1" to IPAddress ip(192, 168, 4, 1)

Thank.

SoylentGraham, 2016-12-24:
Something along these lines should work.
```
int Parts[4] = {0,0,0,0};
int Part = 0;
for ( int i=0; i<strIP.length(); i++ )
{
	char c = strIP[i];
	if ( c == '.' )
	{
		Part++;
		continue;
	}
	Parts[Part] *= 10;
	Parts[Part] += c - '0';
}
IPAddress ip( Parts[0], Parts[1], Parts[2], Parts[3] );
```

oqibidipo, 2016-12-24:
```
IPAddress addr;
if (addr.fromString(strIP)) {
  // it was a valid address, do something with it 
}
```

devyte, 2017-10-05:
@hoacvxd not the right place to ask these questions.
Closing due to off-topic, see #3655 .

closed 2017-10-05 by devyte