esp8266/Arduino #2793: HTTPClient begin does not work with varaible

AF340, 2016-12-24:
Hi, 
 If i try to connect to my server using hard coded string i.e "192.168.0.10", it works, but if i create a String variable i.e String url="192...."  and pass it to .begin(url) it fails.

Any suggestions will be appreciate.

Thks and regards

Palatis, 2016-12-25:
print that string, you might just have a wrong url.

AF340, 2016-12-25:
http.begin("192.168.0.20/action.php?check_test&state=1"); works fine but
String url="192.168.0.20/action.php?check_test&state="
url+=state;
http.begin(url); returns bad request 400
state comes from console input without \n

AF340, 2016-12-25:
Hum, actually, seams a \r is added to the string and cause the issue, should be great the library cleans the argument, issue can be closed, 

Thanks

Palatis, 2016-12-25:
print that string, you might just have wrong URL.

400 is server side error, nothing to do with client code.

Palatis, 2016-12-25:
windows does send \r, the client code shouldn't touch your input, thats by design.

devyte, 2017-10-05:
@AF340 no MCVE, no basic info, no IDE settings provided.
I tried testing similar code:
```
String state = "1";
String url="192.168.0.122/digiout?state=" + state;
http.begin(url);
```

and I can't reproduce. I suspect state has trailing chars in OP's code.
Closing due to lack of info and most likely usage error.

closed 2017-10-05 by devyte