esp8266/Arduino #2794: String functions in IRAM causing `.text' will not fit in region `iram1_0_seg

hreintke, 2016-12-25:
### Basic Infos

#### Hardware
Hardware:			Wemos d1 mini
Core Version:      	git master

### Description

I added the code below to my application and that causes  a "`.text' will not fit in region `iram1_0_seg'" error.
Increased the size in the ld to force an elf file to be created.
Analysing this with objdump -x output shows that f.e. 

_ZNKSs13find_first_ofEPKcjj and 
_ZNKSs17find_first_not_ofERKSsj

are added to IRAM1
Seems standard functions from std:string is that correct ? 

Is there a way to force these functions to flash ?

Also see that some std:map functions are in IRAM, same question for that.

### Settings in IDE

Module:  			Wmos
Flash Size: 		4MB/1MB
Upload Using:		SERIAL

### Sketch

```cpp
std::vector<std::string> split(const std::string& text, const std::string& delims)
{
    std::vector<std::string> tokens;
    std::size_t start = text.find_first_not_of(delims), end = 0;

    while((end = text.find_first_of(delims, start)) != std::string::npos)
    {
        tokens.push_back(text.substr(start, end - start));
        start = text.find_first_not_of(delims, end);
    }
    if(start != std::string::npos)
        tokens.push_back(text.substr(start));

    return tokens;
}
```

igrr, 2016-12-25:
Yes, you can force all non-inlined STL functions into IROM if you add something like
```
    *libstdc++.a:(.literal .text .literal.* .text.*)
```
to the `.irom0.text` section in the linker script (tools/sdk/ld/eagle.app.v6.common.ld).

hreintke, 2016-12-25:
@igrr 
Thanks, first tests show that it works as expected.

Do you want me to make a PR for that to include in Arduino library if it also works after more testing ?

hreintke, 2016-12-28:
Submitted #2804

closed 2016-12-28 by hreintke

bhaap, 2016-12-31:
Thank you for this, this made me able to build this:
https://github.com/ge0rg/WiFI-RGBWW-module
when i added libpwm.a to the ld file :D