esp8266/Arduino #2795: Unable to connect to some Wi-Fi access points

simonvanderveldt, 2016-12-26:
### Hardware
Hardware: ESP-01
Flash Size: 512KB
CPU Frequency: 80Mhz
Core Version: 2.3.0 (assuming this is the version of the Arduino framework for the esp8266)

### Description
I'm just doing a simple thing, namely connection to an access point, which used to work fine when using lua/nodemcu with this access point, but refuses to work using this framework.
The strange thing is, it does work when I connect to my phone's Wi-Fi hotspot, but doesn't work when connecting to my regular access point (a Asus WL-500gP running Tomato).

### Sketch
I've simplified it to just the basic steps/example code. Added the network scan so it shows that the SSID does exist and is found by the esp8266. So at the very least the debug message `no SimonAP found, reconnect after 1s` is wrong.

```
void setup() {
  Serial.begin(115200);
  Serial.setDebugOutput(true);
  Serial.println();
  Serial.println("Starting setup");

  Serial.print("Scan start ... ");
  int n = WiFi.scanNetworks();
  Serial.print(n);
  Serial.println(" network(s) found");
  for (int i = 0; i < n; i++)
  {
    Serial.println(WiFi.SSID(i));
  }
  Serial.println();

  WiFi.disconnect(true);
  WiFi.setAutoConnect(false);
  WiFi.setPhyMode(WIFI_PHY_MODE_11G);
  WiFi.begin("SimonAP", "thisisapassword");
  WiFi.printDiag(Serial);
  Serial.println(WiFi.getPhyMode());

  Serial.print("Connecting");
  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.print(".");
  }
  Serial.println();
  Serial.print("Connected, IP address: ");
  Serial.println(WiFi.localIP());

  delay(2000);
}
```

### Debug Messages
```
setup
Scan start ... scandone
f r0, scandone
14 network(s) found
SimonAP
... (<other 13 APs redacted>)

del if0
usl
mode : null
mode : sta(18:fe:34:9d:2b:d6)
add if0
f r0, Mode: STA
PHY mode: G
Channel: 1
AP id: 0
Status: 1
Auto connect: 1
SSID (7): SimonAP
Passphrase (15): thisisapassword
BSSID set: 0
2
Connecting....scandone
state: 0 -> 2 (b0)
.state: 2 -> 3 (0)
state: 3 -> 0 (19)
..reconnect
f 0, ....scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 0 (19)
..reconnect
f -180, ....scandone
no SimonAP found, reconnect after 1s
reconnect
f r0, ....scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 0 (19)
..reconnect
f 0, .....scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 0 (19)
..reconnect
f -180, ....scandone
no SimonAP found, reconnect after 1s
reconnect
f r0, ....scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 0 (19)
..reconnect
f 0, ..
```

kapyaar, 2016-12-26:
I am also Interested in this. I have had the same issue. I use ESP12, and it fails to connect to some APs. Works fine with 95% of the routers I guess.

tablatronix, 2016-12-26:
I saw another issue posted where connections were dropping out, and it was due to bad beacons, maybe related?

simonvanderveldt, 2016-12-26:
@tablatronix do you mean this one? #1997
I'm not seeing any exceptions though, so not sure if it's the same issue.

tablatronix, 2016-12-27:
Oh yeah thats it, my bad. Hmm yeah not sure if the exception is related or a side effect i did find it strange that there was a problem or incompatibility that went away when the router was changed.

sixbacon, 2017-01-01:
I raised a similar issue. I rolled back the IDE to version 2.0.0 and the problem went away. Not the idea solution but I have lost no other functionality in what I am doing.

simonvanderveldt, 2017-01-02:
> I raised a similar issue. I rolled back the IDE to version 2.0.0 and the problem went away. Not the idea solution but I have lost no other functionality in what I am doing.

@sixbacon thanks for the pointer. Do you mean version 2.0.0 of this project (i.e. https://github.com/esp8266/Arduino/releases/tag/2.0.0)?
I just built the above code with that version (with some items that didn't exist back then removed) but that still gives me the same error.

I might've done something wrong though since I'm using platformio and I'm not 100% sure how to replace the framework with an older version, though after some fiddling it compiled fine.

sixbacon, 2017-01-02:
Hi, 
I used the library manager in the Arduino IDE to list the available versions of the ESP8266 libarires under Info and selected 2.0.0.

simonvanderveldt, 2017-01-02:
> Hi,
> I used the library manager in the Arduino IDE to list the available versions of the ESP8266 libarires under Info and selected 2.0.0.

OK, thanks for the extra info! I'll give that a try this week.

simonvanderveldt, 2017-01-03:
Just tried the above code using v2.0.0 from this project using the Arduino IDE, I had to comment out a couple of lines because the functions didn't exist (`setAutoConnect `, `setPhyMode` and `getPhyMode`), erased the flash and then flashed it, but unfortunately still the same result: `no SimonAP found, reconnect after 1s`

dpeters1, 2017-08-24:
Ran into this problem as well today. Interestingly enough, the esp8266 could connect to the router until I uploaded a sketch including the setPhyMode(WIFI_PHY_MODE_11B) function. After that, it would no longer connect to the router, but would still connect to a different hotspot created by my phone. I flashed another chip with the same sketch, and it has no issues connecting to the router. So far I have tried flashing the factory AT firmware on the chip and factory resetting the router- neither have worked. Please let me know if anyone else has made any progress.

pedrolug, 2017-09-05:
i have exactly the same issue, when trying to connect to any HotSpot (created from my phone or using portable hotspot) the module connect to the hotSpot, acquire an IP but don't have any connection to Internet, using my router at home i don't have any problem... Any ideas ?

i am using #include <ESP8266WiFi.h>
...
WiFiClient client;
...
WiFi.begin(MY_SSID, MY_PWD);

underwoodblog, 2017-09-06:
For a more stable connection, I use fixed parameters for IP, GATEWAY and NETMASK.  Getting them via DHCP seems not to work with every Router reliable and stable.

devyte, 2017-10-05:
@simonvanderveldt is this still an issue with latest git? there is also #3215 . Did you ever erase the entire flash before flashing your sketch?
I see that you also had issues with the nodemcu firmware, per your comments in their repo. This makes me suspect an SDK issue, in which case there wouldn't be any possible fix to implement in this repo.

simonvanderveldt, 2017-10-05:
> @simonvanderveldt is this still an issue with latest git? there is also #3215 . Did you ever erase the entire flash before flashing your sketch?

@devyte I tried pretty much everything, using multiple devices, entirely nuking them, etc. Nothing worked.

> I see that you also had issues with the nodemcu firmware, per your comments in their repo. This makes me suspect an SDK issue, in which case there wouldn't be any possible fix to implement in this repo.

That could very well be the case. Is it somehow possible to identify if this is really and SDK issue?

dpeters1, 2017-10-05:
So here's what I did to allow the esp8266 to reconnect to the router:

Reflash stock firmware using the nodeMCU flashing software and ESP8266_NONOS_SDK-2.1.0 .bin files

Under 'Config' in the flash software, set these files to their corresponding memory locations

| File Path  | Memory Location  |
| ------------- | ----- |
| ESP8266_NONOS_SDK-2.1.0\ESP8266_NONOS_SDK-2.1.0\bin\esp_init_data_default.bin | 0xFC000 | 
| ESP8266_NONOS_SDK-2.1.0\ESP8266_NONOS_SDK-2.1.0\bin\blank.bin | 0xFE000  |
| ESP8266_NONOS_SDK-2.1.0\ESP8266_NONOS_SDK-2.1.0\bin\boot_v1.7.bin | 0x00000 | 
| ESP8266_NONOS_SDK-2.1.0\ESP8266_NONOS_SDK-2.1.0\bin\at\512+512\user1.1024.new.2.bin | 0x01000 | 

Now, plug in the ESP8266 via FTDI, and hit flash.

Once completed, you can unplug the ESP, exit the flash software, and plug it back in.

Open Putty or some other serial comm software and set your Serial address. For me, the baud rate was 115200 but you may have to experiment.

Start by typing in `AT` to verify that communication works.
To send any command, type your command, followed by enter, followed by ctrl+J. Don't know why putty requirs ctrl-J after enter, but knowing this will save you some headache.

If you get a response, you can type in `AT+RST` to fully reset the device.

Next, I don't know if this is necessary, but I changed the device's hard ip and mac address. This is supposed to stick after a reboot but didn't for me. Regardless, it may have had an effect on letting the chip reconnect to the router.

`AT+CIPSTAMAC_DEF="18:fe:35:98:d3:7b"`  (example values)

`AT+CIPSTA_DEF="192.168.0.70","192.168.6.1","255.255.255.0"` (example values)

[Further commands are documented here](https://www.espressif.com/sites/default/files/documentation/4a-esp8266_at_instruction_set_en.pdf)

After all this is done, flash a normal arduino sketch and see if you can connect!

devyte, 2017-12-05:
@simonvanderveldt are you _sure_ you tried with latest git? your original post said 2.3.0, which is ancient.
Some comments:
 - I have seen several issues between the ESP and Asus+Tomato. Not sure if the issue is Asus or tomato, but... *shrug*
 - I have personally seen no issues with dd-wrt, ever, on any router, including a couple of Asus (don't remember the models). Off the top of my head, I also don't remember seeing any online. I don't know if that's luck or not. 
 - I only use WPA2/AES.
 - A few days ago I had an issue with symptoms similar to this one, just not on an Asus+tomato router. Turned out that the router had wide wifi channel (40MHz) enabled by default. Changing that to 20MHz channel after a reset fixed it. I suppose other router wifi settings could also have impact.

electrosaeed, 2017-12-31:
I had same problem to connect to a specific Access Point (a Chinese D-LINK!), while my ESP connects normally to other APs. In my case the problem solved by changing AP security settings, this AP was set on WPA2-PSK (AES) security mode (that I could connect to it by my PC), in this mode my ESP was unable to connect to AP. By changing this to WPA-PSK (TKIP) (or set in open mode) the ESP connected to AP. I have to mention that my ESP connects to other APs in WPA2 mode and this problem is for just this kind of AP.
In summary I could say:
- My ESP (ESP07S) connects to All tested APs (mobile hotspot or a real modem) in all security modes.
- My PC connects to this specific AP (D-LINK DI-624+A G) in all security modes.
- My ESP can not connect to this AP in WPA or WPA2 in AES mode.
- My ESP connects to this AP in open or in WPA (TKIP) modes.

gabonator, 2017-12-31:
Hello guys, I am facing the same problem. I have two APs at home - both are Asus wl500-gP. One has stock firmware, the second one was flashed with alternative "oleg's firmware". My iphone can connect to both of them without any problem. Some android phones refuse to connect to the AP with alternative firmware, as well as the wemos D1 board equipped with ESP-12 module.  So it is probably related to the wifi configuration in your router. Unfortunately, I cannot login to the problematic AP, so I cannot verify this by changing the wifi settings.

devyte, 2018-02-28:
@simonvanderveldt we're now on SDK 2.2+. Does this still happen?

simonvanderveldt, 2018-02-28:
@devyte I've long ago left the esp behind because it didn't work, maybe someone that also experienced the issues can verify if it still happens?

devyte, 2018-02-28:
@gabonator @electrosaeed any feedback from you?

kapyaar, 2018-02-28:
@devyte I am using 2.3.0 and recently had issues with Ruckus ZoneDirector 3000 Series and Ruckus R500 WAPs. Could not get it to connect. Had to giveup.

gabonator, 2018-02-28:
I cannot verify it right now, will try new SDK during weekend and report whether it works

pwrsoft, 2018-03-09:
Changed my WiFi AP to WPA2-Auto (AES+TKIP) and it works now!
Thank you @electrosaeed very much!

nibblerrick, 2018-03-19:
I had the same problem using UBNT-APs. AES Only won't work, Auto (AES/TKIP) works. But I don't think TKIP should be used anymore.
With ESPEasy_v2.0-20180319 I wasn't able to connect to the AP at all. R120 works.

jaatbuddhi, 2018-06-07:
I had been facing the issue that my ESP12E module would connect to some routers and not connect to others, 

I really don't know whether this works for others or not, I tried with all 4 of the stand alone ESP12E modules that I have and through trial and error figured out that none of them would connect if the SSID was 4 characters or less, the moment I changed the SSID to 5 or more characters, the same ESP12E Modules would connect to the same routers without a hitch.

If any body has any knowledge about this, kindly share your information

RohitJangra, 2018-10-08:
changing mac address does no good.  My ESP12 still not connect to one of my router TP-LINK with encryption WPA_WPA2_PSK. Changing this to WPA2 or WPA solves the problem any my esp can now connect to router. This should not be the only solution.  looking forward for this bug fix soon.

underwoodblog, 2018-10-08:
I had problems that the ESP sometimes disconects, since I've added

#define PHY_MODE WIFI_PHY_MODE_11G

the connection is more stable.

seinecle, 2018-11-07:
For those still struggling, this did the trick:

WiFi.persistent(false);
WiFi.disconnect(true);
WiFi.begin(ssid, password);

source: http://blog.flynnmetrics.com/uncategorized/esp8266-exception-3/

morla10111, 2019-01-06:
I had problems connecting to my Linksys WRT3200acm running OpenWRT (SNAPSHOT r7925-b6adfde).
Changing the channel, mode or width did not help. Neither did changing the SSID or encryption key.
For me turning off `WMM` mode did the trick. 

My code looks like this:
```
#include <ESP8266WiFi.h>

void setup()
{
  Serial.begin(115200);
  Serial.println();
    
  uint8_t macAddr[6];
  WiFi.macAddress(macAddr);
  Serial.printf("mac address: %02x:%02x:%02x:%02x:%02x:%02x\n", macAddr[0], macAddr[1], macAddr[2], macAddr[3], macAddr[4], macAddr[5]);

  WiFi.begin("myssid", "mypasswurdz");

  Serial.print("Connecting");
  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.printf("Connection status: %d\n", WiFi.status());
  }
  Serial.println();
  WiFi.printDiag(Serial);
  Serial.print("Connected, IP address: ");
  Serial.println(WiFi.localIP());
}

void loop() {}
```
My Wifi settings are
```
Mode: N
Channel: 6
Width: 40 MHz
Transmit Power: auto
WMM Mode: disabled!!!!!!

Encryption: WPA2-PSK
Cipher: Force CCMP (AES)
Short Preamble: checked
```

Thanks for the great work!

therault, 2019-01-23:
Hi, I have the same configuration: Linksys WRT3200acm, running OpenWRT (LEDE 17.01.4, distrib revision r3560-79f57e422d), and multiple clones of ESP8622 claiming to be ESP12E. All the ESP8622 clones worked well with other routers, but with this version of the router, no packet would be received by the router after the (successful) connection to the WiFi network.

Disabling WMM mode on the interface solves the problem. I don't know if it is a bug of ESP8622 or of OpenWRT. I couldn't test if WMM was enabled on the other routers I tested.

jaromanda, 2019-02-09:
reading https://github.com/kaloz/mwlwifi/issues/278 seems ESP's have an issue connecting to some (in this case Marvell 88W8x64 based) routers - one (not very satisfactory) solution is to turn WMM off, thus disabling 802.11n - another (not very safe) solution is to turn wifi encryption off

Although that issue I linked to is with the current OpenWrt firmware, I can confirm, at least with the Linksys WRT32X, that the issue is also present in the factory firmware (which is based on an older OpenWrt anyway)

The debate continues if this is a bug in ESP or Marvell driver

MiguelPita, 2019-02-11:
> 
> 
> For those still struggling, this did the trick:
> 
> WiFi.persistent(false);
> WiFi.disconnect(true);
> WiFi.begin(ssid, password);
> 
> source: http://blog.flynnmetrics.com/uncategorized/esp8266-exception-3/

This worked fine for me... thank you very much....

tablatronix, 2019-02-11:
I am fairly certain there is a new bug causing connect issues. I have not been able to nail it down but the fix is as above to disconnect and begin, there seems to be a problem with the wifi station status sync, it imagine it could also be a race condition. I will post more details later or somewhere else

Khalid7777777, 2019-03-04:
Hi All:
our product currently suffering from the above issue. our device stops connecting ti the WiFi router. some of those worked for 4 weeks and then enters the state of unsuccessful WiFi connection. the resetting the device sometimes resolves the issue but always. resetting the router resolves the issue.
I really need an explanation of the source of this problem. 
I will apply the Fix and retry again using the above fix and you all know.

Note: I have 12 000 devices in stock with that code. :)

Keltere, 2019-03-04:
@Khalid7777777 what wifi router did you use? Did you tried disabling WMM?

tablatronix, 2019-03-04:
@Khalid7777777 you do not mention the last lib version you tested with or are running on these devices.

jaromanda, 2019-03-04:
The current dev branch seems to "hold" the wifi connection much better than when I last commented 23 days ago - however, OTA updates take 5+ minutes compared to tens of seconds - so there's still something odd going on

Khalid7777777, 2019-03-05:
Hi Thanks for responding:
the router that we are currently testing with and can reproduce is BT Hub6. lib 2.42.
however, I am currently trying using version 2.5 and still having the same problem.

the bellow fix made no change.
WiFi.persistent(false);
WiFi.disconnect(true);
WiFi.begin(ssid, password);

Khalid7777777, 2019-03-08:
to save the day I put a retry if the connection fails with bt hub however the failure is still current but more reliable. As for the root cause we are still waiting :)

chmpe, 2019-04-27:
The problem is with DHCP.
The only way to fix the problem was with static settings
IPAddress ip(192, 168, 29, 15); 
IPAddress gateway(192, 168, 29, 1);
IPAddress subnet(255, 255, 255, 0);
IPAddress dns(8,8,8,8);
WiFi.config(ip, dns, gateway, subnet)
WiFi.begin(ssid, password);

oivinds, 2019-05-13:
I can connect my nodemcu ESP 12E dev board to my android device access point but not to my (new) home router (I belive the old router worked).
Suggestions above to fix connection has not helped. 
-resetting router
- persistent / disconnect 
- fixed ip

Khalid7777777, 2019-05-13:
check the mode your ESP is set to and make sure it is using n,g or b modes. hope that helps

oivinds, 2019-05-14:
WiFi.setPhyMode( WIFI_PHY_MODE_11B ); // WIFI_PHY_MODE_11G /  WIFI_PHY_MODE_11N /
It returns the correct integer for the variables above when calling: 
WiFi.getPhyMode() 

So i assume it sets the mode correctly, but I does not help. 
Should I try this in combination with some of the fixes previously mentioned? Any particular?

jaromanda, 2019-05-14:
> 
> 
> I can connect my nodemcu ESP 12E dev board to my android device access point but not to my (new) home router (I belive the old router worked).

So, what is the new router? There's some that have an issue with ESP's and the like (see my previous comments)

None of the fixes here have any effect if you have one of the routers mentioned

oivinds, 2019-05-14:
The router is a combined PVR / router. After some googling It seems the router part is based on Openrg from the manufactor Arris.

oivinds, 2019-05-16:
Please ignore my previous post. Now Wifi client code works, without any changes to the helloserver sketch, and on my Arris router.

gangalal, 2019-06-16:
> check the mode your ESP is set to and make sure it is using n,g or b modes. hope that helps

I have ASUS router. i changed every setting mentioned above still couldn't connect to it. while i can connect to TP-LInk and netgear router. is there any new trick or setting i can use?

aloft38, 2019-09-27:
I was having the same symptoms.  I believe my issue may have been a simpler one to fix, but I haven't seen anyone comment on this potential cause here.  I will add this in hopes that it may help someone else coming across this thread as I had.

Some routers are setup to take hexadecimal values as the "password" and others are setup to take a password of characters.  https://www.arduino.cc/en/Reference/WiFiBegin shows that the begin function is overloaded.  You call different versions of it depending upon what type of security settings are configured on your router.  All of the examples I have seen online only use one version of this function call making it appear, at a cursory glance, that there is only one version of the function.

Syntax
WiFi.begin();
WiFi.begin(ssid);
WiFi.begin(ssid, pass);
WiFi.begin(ssid, keyIndex, key);

Parameters
ssid: the SSID (Service Set Identifier) is the name of the WiFi network you want to connect to.
keyIndex: WEP encrypted networks can hold up to 4 different keys. This identifies which key you are going to use.
key: a hexadecimal string used as a security code for WEP encrypted networks.
pass: WPA encrypted networks use a password in the form of a string for security.

My ESP8266 could see my router and my phone's hotspot with the scan code above, but could only connect to the phone until I used a different version of the begin function.  Now I have gotten everything to work!

TD-er, 2019-09-29:
> My ESP8266 could see my router and my phone's hotspot with the scan code above, but could only connect to the phone until I used a different version of the begin function. Now I have gotten everything to work!

What version do you now use?

tablatronix, 2019-09-30:
This seems like a very specific result, and probably not related to most of this issue.

aloft38, 2019-09-30:
> 
> 
> > My ESP8266 could see my router and my phone's hotspot with the scan code above, but could only connect to the phone until I used a different version of the begin function. Now I have gotten everything to work!
> 
> What version do you now use?

Each version of the *begin* function is used to connect to different WiFi Access Point configurations.  I use different versions of the function depending upon what I am trying to connect to.  You have to change your code and re-flash the ESP8266 each time you want to change your ESP8266 to connect to a different Access Point using the *begin* function.

You use
WiFi.begin(ssid);
if you are trying to connect to an Access Point that doesn't have a password

You use
WiFi.begin(ssid, pass);
if you are trying to connect to a WPA encrypted network that has a password in the form of a string

You use
WiFi.begin(ssid, keyIndex, key);
if you are trying to connect to a WEP encrypted network that uses a set of hexadecimal values as the password.

devyte, 2019-10-17:
This issue is old and the thread wandered off topic. If you're still affected, I suggest retesting with latest git, because there have been reports that boards that couldn't connect now work after #6484 .
I am closing. If the issue isn't fixed for you, please open a new issue and fill in the required details.

closed 2019-10-17 by devyte

InQuize, 2025-03-24:
Was having compatibility issues as well, found nothing, but managed to solve, posting for search engines.

Issue turned out to be ESP WiFi stack specific, and hardware irrelevant, but listing it just for context.
- AP: HostAPd v2.11 x86 (Mikrotik R11e-2HPnD) configured to 802.11g+n, WPA2, AES only (`rsn_pairwise=CCMP`)
- STA: ESP8266EX embedded into 303WIFILC01 module running [dCLC](https://github.com/maarten-pennings/303WIFILC01) Arduino firmware.

Behavior:
- ESP connects successfully when AP set to 20/40MHz channel mode forcing AP to downgrade channel 40 -> 20 Mhz
- ESP authenticates, but gets rejected AP association when the latter is set to 40MHz only mode (`require_ht=1`), goes into reconnect loop that (due to handshake data rates) causes significant throughput decrease (almost to 20Mhz channel levels) for entire AP and all STAs

Hostapd logs and config flexibility was a blessing during debug.

Solution:
Tweak ESP8266 board settings in Arduino IDE in a hope to fix HT40 support in WiFi stack.
Problematic firmware was compiled with default (ESP8266 v3.1.2 package) IDE settings, I only changed 3 of them and it solved the problem. Haven't confirmed which one specifically, since it takes too much time to reflash+reset+config this module. But here is the screenshot of modified settings comparison.

[image]

Betting that `lwIP Variant` was cutting out WiFi features to save flash space.
If so, _this could potentially fix most of the similar issues_ with AP compatibility, not only channel width.

BTW, there is a quirk with HT40 even after fix: `require_ht=1` hostapd option still prevents ESP from joining, saying that its PHY does not support mandatory HT, even though it is capable of 20/40MHz mode (`require_ht=0`) and no longer downgrades channel width.

P.S. I was using CH431A based programmed to flash EEPROM directly. Not sure that Erase "All Flash Contents" option applies here.