esp8266/Arduino #2796: This works on ESP8266-DEV but not in ESP8266-01

gusantor, 2016-12-26:
The code here https://github.com/gusantor/ESP8266-indice-uv6 run in a ESP8266-DEV but not in a ES8266-01

I would like to know why or how to improve the code to get it to work on a ESP8266-01

Any guide or comments are very appreciated

### Basic Infos

I've add many yield() statements, and move some string (css,javascript) to PROGMEM, but cant get a stable performance on the ESP8266-01

Capacitors are there to improve stability

#### Hardware
Hardware:			ESP8266-01
Core Version:      	don't know, just a black ESP8266-01, 1Mb

### Description

The code run on a ESP8266-DEV but not in a ESP8266-01. At the ESP8266-01 it runs just sometimes and with very poor stability. 

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		1MB
CPU Frequency:		80Mhz
Flash Mode:			qio
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		ck

### Sketch

please see https://github.com/gusantor/ESP8266-indice-uv6

### Debug Messages

There are many, really don't know which of them have valuable info

SuperJerry69, 2016-12-26:
Hi, esp8266-01 are mostly 512k device , DEV have 4mbit flash ....
juste news esp8266-01 'S' with mbit flash
On your Github <TimeAlarms.h> is missing to check ...
Maybe you can put a lot of const-strings in PROGMEM ...

gusantor, 2016-12-27:
Hello, thank for taking time to give this info, but your comment appears to be trimmed ... ?

`On your Github <TimeAlarms.h> is missing to check ...`

and yes, I'm using a black ESP8266-01 which seems to have 1Mb flash

When compile, I can see:

> El Sketch usa 293.681 bytes (58%) del espacio de almacenamiento de programa. El máximo es 499.696 bytes.
Las variables Globales usan 53.368 bytes (65%) de la memoria dinámica, dejando 28.552 bytes para las variables locales. El máximo es 81.920 bytes.

I wonder now, how much memory the device have ?. Doesn't it should say "1,048,576 bytes" as maximum ? ... I'm using 1M (512K SPIFFS) at Arduino IDE -> Falsh size configuration 

Should I look forward to reduce that 58% or that 65% or both ... ?

so, it seems it is a matter of free space  / memory ... ?

thanks for any help

pieman64, 2016-12-27:
@gusantor by allocating 50% of the memory to SPIFFS you are restricting the program memory and the ability to do OTA updates. Unless you need 512K SPIFFS I would suggest you try with 64K.

SuperJerry69, 2016-12-27:
@pieman64 : gusantor didn't use OTA on his project.
65% of ram is a lot but there is no diff between an esp8266 with an other ... ( on the amount of ram ... )
Gusantor: what is your flash chip on esp8266-01 ?

gusantor, 2016-12-27:
thank you @pieman64 , I did, but no improvement in actual issue, the board still is unstable

as far I can see, the problem occurs when I try to connect to my Wifi, and trying to deliver (through a server.send()) the new IP of the module

what I'm trying to do is:

> start in AP mode (IP: 192.168.4.1) -> give SSID&PASS of my Wifi -> return to the webbrowser the new IP obtained from DHCP -> kill AP mode -> stay STA mode connected to Wifi

trying to do that, the module just start to spit off a lot of garbage through serial, doesn't stop, becomes stuck

is there a recipe to do that ?, I searched for it but can't found

thanks

gusantor, 2016-12-27:
@pieman64 , the chip is 
> BergMicro
25Q80ASSIG
1524

SuperJerry69, 2016-12-27:
After a 5V touch on IO pin of the ESP : i already have the same behavior ... >> esp on trash :-(
I already have new ESP8266-01 but with fake flash chip ... switching between QIO / DIO sometimes make it works, but i prefer to take a new module than lost time to change the chip ...

gusantor, 2016-12-27:
@SuperJerry69 thank you

I discard the 5v issue, this module is, since ever, mounted on a 3.3v development board I arrange on a breadboard

also, It work normally with small sketches

thanks

pieman64, 2016-12-27:
@gusantor when I compile your project it gives the following details:

Sketch uses 290,721 bytes (27%) of program storage space. Maximum is 1,044,464 bytes.
Global variables use 52,408 bytes (63%) of dynamic memory, leaving 29,512 bytes for local variables. Maximum is 81,920 bytes.

Are you sure you have 1M and not 512Kb ESP-01?
There is a memory checker sketch on GitHub that can confirm memory details.

duncan-a, 2016-12-27:
According to the webpage for the flash chip... 80A = 8Mbit (1024kx8)

gusantor, 2016-12-27:
thank you all

so 8Mbit appears to be good news

I've been able to define Flash size -> 1Mb (64K SPIFFS) ,it still get stuck in a cycle of error / reboot

problem remains

the sketch runs on ESP8266-DEV but not on ESP8266-01 (8Mb)

so it seems it is no a matter of memory ... ?

@pieman64 did you run the sketch -> login (admin&admin) and connect to a Wifi ?, did you get a smooth operation ?

thanks

gusantor, 2016-12-27:
Hello

I run this sketch to get the actual size

>https://github.com/esp8266/Arduino/tree/master/libraries/esp8266/examples/CheckFlashConfig

results were

>Flash ide  size: 1048576
Flash ide speed: 40000000
Flash ide mode:  DIO
Flash Chip configuration ok.
Flash real id:   001440E0
Flash real size: 1048576

this confuses me, so I tryed also

>gusantor@polilla:~/esptool-master$ esptool.py -p /dev/ttyUSB0 flash_id
Connecting...
Manufacturer: e0
Device: 4014

and, no matter what Flash size I choose, above 1M, I get following error (only in ESP8266-01)

>WS:av
:ref 2
:ur 2
:ref 2
:ur 2
:c0 1, 384
ets Jan  8 2013,rst cause:4, boot mode:(3,6)
wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0xad
csum 0xad
v09f0c112
Fatal exception (0): 
epc1=0x400ef424, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000

with last two lines multipled to infinite

please any help is welcome, thanks

devyte, 2017-10-05:
@gusantor don't confuse Mbit with Mbyte. Your ESP-01 seems to be 1Mbyte = 8Mbit. Your first compilation message says 

> El Sketch usa 293.681 bytes (58%) del espacio de almacenamiento de programa. El máximo es 499.696 bytes.

which to me looks like a wrong flash chip size config.

The sketch you link to is too big for me to take a quick look for obvious mistakes. My advice:

- Be careful with what you declare on the stack. The ESP has a max stack size of only 4KB. If you go above that, you'll get mem corruption and instability.
- Be careful with timings. There are timing guidelines due to requirement of servicing the wifi stack.
- Be careful with the power supply. The wifi radio causes spikes in power requirements, which can cause dips in supply voltage. If the power supply is not good, it can result in instabilities. I use a 5V 20A power supply during development to make sure that any resets are NOT due to voltage dips.
- Recheck your code, break it up into pieces, test the pieces individually.
- Instad of the EEPROM, use the SPIFFS, if you can. The ESP doesn't really have an EEPROM, it's really emulated over a particular sector of the flash. If you trash it by writing frequently, you'll wear out the sector, and once it's damaged you could read garbage.

I'm sorry I can't help further, but in any case this is not the right place to request help for your particular project.
Closing as off-topic, see #3655 .

closed 2017-10-05 by devyte

gusantor, 2017-10-05:
@devyte thank you, I'll take in accouont the tips you give