esp8266/Arduino #2799: MFRC522 library interaction causes Soft WDT reset

Humancell, 2016-12-27:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.md

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.md

----------------------------- Remove above -----------------------------

### Basic Info

#### Hardware
Hardware:			?ESP-12F?
Core Version:      	?2.3.0?

### Description

I'm using the ESP-12F, and am using the Arduino IDE v1.6.5, with Arduino/ESP8266 v2.3.0, and MFRC522 v1.1.7

Everything compiles, and I've got the connections correct, but when calling PCD_Init, I get a Soft WDT reset and stack trace.  :-(

### Settings in IDE

Module:  			?Generic ESP8266 Module?
Flash Size: 		?2MB/1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?DIO?
Flash Frequency:	?40Mhz?
Upload Using:		?SERIAL?
Reset Method:		?ck?

### Sketch

```cpp
#define DEBUG

#include <SPI.h>
#include "MFRC522.h"

#define RST_PIN         5          // Configurable, see typical pin layout above
#define SS_PIN          4          // Configurable, see typical pin layout above

MFRC522 mfrc522(SS_PIN, RST_PIN);  // Create MFRC522 instance

void setup(void) {
	Serial.begin(9600);		// Initialize serial communications with the PC
  delay(10);
  
  #ifdef DEBUG
  /* Print our boot message and version */
  Serial.println(F(""));
  Serial.print(F("RC522 RFID Sensor - GUID: "));
  Serial.printf("%06X\n", ESP.getChipId());
  Serial.print (F("Free memory = "));
  Serial.println (ESP.getFreeHeap());
  #endif

  Serial.print(F("SPI.begin ... "));
	SPI.begin();			// Init SPI bus
  Serial.println(F("done!"));
 
  Serial.print(F("mfrc522.PCD_Init ... "));
	mfrc522.PCD_Init();		// Init MFRC522
  Serial.println(F("done!"));
 
	//mfrc522.PCD_DumpVersionToSerial();	// Show details of PCD - MFRC522 Card Reader details
	Serial.println(F("Scan PICC to see UID, type, and data blocks..."));
}

void loop() {
	// Look for new cards
	if ( ! mfrc522.PICC_IsNewCardPresent()) {
		return;
	}

	// Select one of the cards
	if ( ! mfrc522.PICC_ReadCardSerial()) {
		return;
	}

	// Dump debug info about the card; PICC_HaltA() is automatically called
	mfrc522.PICC_DumpToSerial(&(mfrc522.uid));
}
```

### Debug Messages

```
RC522 RFID Sensor - GUID: D546E6
Free memory = 47856
SPI.begin ... done!
mfrc522.PCD_Init ... 
Soft WDT reset

ctx: cont 
sp: 3ffef300 end: 3ffef530 offset: 01b0

>>>stack>>>
3ffef4b0:  00241001 00000001 3ffee510 3ffee468  
3ffef4c0:  00000002 3ffee430 3ffee448 402022c7  
3ffef4d0:  402010ae 00000001 00000032 402010a3  
3ffef4e0:  3fffdad0 00000010 3ffee430 40202445  
3ffef4f0:  3fffdad0 3ffef554 3ffee430 402024d6  
3ffef500:  3fffdad0 3ffef554 3ffef574 40201c93  
3ffef510:  feefeffe 00000000 3ffee500 402033a8  
3ffef520:  feefeffe feefeffe 3ffee510 40100114  
<<<stack<<<
ü!ÿü!
```

Kruziolo, 2017-01-01:
I had some issues while defining the SS_PIN and RST_PINs without the "D" next to the pin number. So D4 / D5 instead of using 4 / 5 worked for me.

Jorgen-VikingGod, 2017-01-02:
@Kruziolo he is using ESP-12F SoC and not the WeMos or NodeMCU board...
@Humancell try to compare your code with my example: https://github.com/Jorgen-VikingGod/ESP8266-MFRC522

Humancell, 2017-01-02:
I'll try both solutions.  I just found a post where someone said that using the D5/GPIO5 for RST will not work.  When I disconnected that pin I got things working.

@Jorgen-VikingGod I have looked at your code, etc. and you seem to be using GPIO5 in your code ... except in the wiring where you say to use GPIO2 and GPIO15.

https://github.com/Jorgen-VikingGod/ESP8266-MFRC522/blob/master/MFRC522-ESP8266_wiring.png

I'm going to try and move RST to GPIO15 and see if that works better.

Jorgen-VikingGod, 2017-01-02:
@Humancell sorry I did not update the wiring diagram. This worked for me in the beginning; but it seams that not all ESP chips are same. So some people had trouble of using the wiring from the diagram.
Please try it out - maybe it worked for you; but change the SS pin to GPIO15 and RST to GPIO02 or other free pin.

LucaUrbinati44, 2017-08-10:
@Humancell : @Jorgen-VikingGod 's solution worked for me, but I used RST to GPIO0, instead GPIO2. I have a Wemos D1 R2. Thanks for the solution.

tomrit, 2017-08-18:
@Humancell , @Jorgen-VikingGod   A further potential problem using the MFRC522 library arises from the baud rate. In your code, you're using 9600 which is way to slow to empty the buffer created by the "DumpToSerial"-Routines.  For me, increasing the baud rate to 115200 worked out (totally fine with your ESP board) and avoided any soft wdt resets....

devyte, 2017-10-05:
Don't confuse the D numbering with GPIO numbering. And different boards have different pins free for user applications.
And yes, the ESP usually has trouble with low baud rates, because the WDT tends to trigger (sending data takes way too long).
This is a 3rd party lib, and 3rd party hardware. In addition, several comments state success in various ways. 
Closing.

closed 2017-10-05 by devyte

TLR1000, 2018-07-11:
Just get de baudrate of the serial comms up.