esp8266/Arduino #2801: Unable to load the code from arduino ide to esp8266

jyotirajsharma, 2016-12-27:
Hi,

I am unable to load the code from arduino ide to esp8266
Sketch uses 232,505 bytes (53%) of program storage space. Maximum is 434,160 bytes.
Global variables use 32,504 bytes (39%) of dynamic memory, leaving 49,416 bytes for local variables. Maximum is 81,920 bytes.
error: failed reading byte
warning: espcomm_send_command: cant receive slip payload data
error: failed reading byte
warning: espcomm_send_command: cant receive slip payload data
error: failed reading byte
warning: espcomm_send_command: cant receive slip payload data
error: failed reading byte
warning: espcomm_send_command: cant receive slip payload data
error: failed reading byte
warning: espcomm_send_command: cant receive slip payload data
error: failed reading byte
warning: espcomm_send_command: cant receive slip payload data
error: failed reading byte
warning: espcomm_send_command: cant receive slip payload data
error: failed reading byte
warning: espcomm_send_command: cant receive slip payload data
error: failed reading byte
warning: espcomm_send_command: cant receive slip payload data
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed

Please help me to resolve this issue.

mcolinj, 2017-01-01:
I've just recently installed the Arduino version 1.8.0 IDE and loaded the third party ESP8266 extensions.    I've seen these same messages as well.   However, it does seem that the code is being downloaded *and* is running on the board.

There is likely a bug related to these messages, and, perhaps they are an indicator that you should retry the download.

But, if you just want to continue to make progress, you should look to see if your board is actually running the software.  (Perhaps you can ignore the messages and get on with it!)

matthew

P.S.  If you want more attention given to this bug report, you should include specific details about what versions of the IDE, what board you are using, etc...

jyotirajsharma, 2017-01-01:
@mcolinj ,
You mean esp8266 what we got does not have software flashed into it and we need to flash it externally using esptool ?
How to see if my board is actually running software or not.
I am using arduino-1.6.12 IDE and esp8266 board.

jyotirajsharma, 2017-01-01:
**Connection Diagram 1:**
Arduino Uno             Esp8266
TX---------------------RX
RX---------------------TX
3.3V-------------------VCC
3.3V--------------------CH_EN
GND--------------------GND

I have various forums and they suggested that 
1. GPIO_0 of esp8266 should also be connected to GND.
OR
Also seen some diagram that they connected RESET pin of Arduino is connected to GND.

Please suggest what should be the exact connection diagram.

mcolinj, 2017-01-01:
What I was saying is that you were seeing that error message when you were trying to flash the software
onto your esp8266 board, correct?

Several days ago, I saw the same messages as well, but I was able to try it again (even seeing similar messages)
and my software was flashed onto the esp8266 board.      Just to be sure, I loaded the blinky
LED software (which blinks once every 2 seconds).  I modified it to blink rapidly, and, even though the I received
those error messages.  This is how I determined that the resulting new code has been flashed and was running
on the board.

Yesterday, I chose a menu item under the Tools menu to set the “Upload Speed” to 115200, and it seems that the
upload/flash process is more reliable.     I don’t see those messages at all any ore.  I don’t know if you have that
option (I think it is board specific, but I have my board set to WeMos D1 (Retired), as that is the device configuration
recommended for developing with the NodeLink ESP8266 D1 board.

good luck,

matthew

jyotirajsharma, 2017-01-01:
Yes, while trying to flash the esp8266 (connect GPIO_0 of esp8266 to GND). I put more verbose logs and got following errors.
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	serialport_receive_C0: F8 instead of C0
resetting board
trying to connect

warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed

cyyamazaki, 2017-01-01:
I tried to compile and upload a code to esp-12f (ch340g) and got the same message when I didn´t put in flash mode and upload speed was 256000. Setting both mistakes right, the software was uploaded.

**Arduino: 1.8.0 (Windows 7), Placa:"Generic ESP8266 Module, 80 MHz, 40MHz, DIO, _256000_, 512K (64K SPIFFS), ck, Disabled, None"
...
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	serialport_receive_C0: F8 instead of C0
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed**

jyotirajsharma, 2017-01-03:
Do i need to supply 3.3v from external source to esp8266 ? For now, I provided it from Arduino 3.3v to esp8266?

jyotirajsharma, 2017-01-03:
Please suggest if possible.

cyyamazaki, 2017-01-03:
Sorry for my english. I speak portuguese and English is a second language for me.

Yes, it´s better supply a good 3.3 v to the board, like the forums say.  As I used a integrated programmer (like this<https://www.aliexpress.com/item/ESP8266-serial-WIFI-Witty-cloud-Development-Board-ESP-12F-module/32557886226.html?spm=2114.01010208.3.292.dxkXg8&ws_ab_test=searchweb0_0,searchweb201602_5_10065_10068_10000009_10084_10083_10080_10082_10081_10060_10062_10056_10055_503_10054_10059_10099_10078_501_10079_426_10103_10073_10102_10096_10052_10053_10050_10107_10051_10106-501_503,searchweb201603_4,afswitch_5&btsid=2c86b91c-85ea-40f8-9189-054123f75ac2>), it supplies my board just by an USB cable. I just used a LM1117 for this too and a cell phone adapter with a CP2102 and it works, too.

In the thread, I said I set the speed to 115.200 and keep the flash button pressed during the upload (i think the GPIO00 is used to this) and the upload was fine.

To repeat identically your error message, i set the speed to 256000 and did´nt press the flash buttton. But may be another errors to cause the same message.

I hope it could be usefull for you.

rousnay, 2017-01-16:
Try to connect The GPIO0 with a 10k resistor to ground and the CH_PD to VCC and it works :)

darthm0e, 2017-01-25:
I ran into same Problems with flashing esp-01 modules.
While i need a bunch of them (>50 pcs) i ordered some CH340G-USB-Programmer from aliExpress.
they work well for flashing the esp-01 modules, but i still have to connect GPIO-0 -> GND and CH_PD -> VCC.

But then it works like a charm. I used both Arduino IDE (1.8.0) and esptool.
I'm running arch Linux .

When you use a simple UART-USB adapter to flash the ESP-01, take care of the power supply while most UART Programmers didn't supply the needed ~200mA for flashing the ESP-01.

Sephok, 2017-04-10:
Seems like the following thread is basically the same problem.

[https://github.com/esp8266/Arduino/issues/770](url)

Did this work? Just signed up yesterday, but there should be a link to a similar problem above.
However, I'll also copypaste my post from there into here, because I had the same problem and got it working now (even if there is still the problem of the esp seemingly hanging up after a couple minutes):

TL,DR: I solved my problems (warning: espcomm_sync failed etc.) by replacing the 10k resistor between GPIO00 and GND I used first with a 3.9k resistor.

Long Version:
I was also starting to feel a little irritated after sitting in front of this problem last night and today. But just about an hour ago I managed to solve my problem.
Initially my Setup is the ESP-12S on the typical white adapter board. VCC to 3.3V, GND to GND, CH_PD (or EN on other chips) and GPIO02 pulled high with 10k (although CH_PD should be high anyways with the layout of the adapter board, turning the two parallel 10k resistors effectively into a 5k), GPIO15 and GPIO00 pulled low with 10k (although GPIO15 should also be pulled low by the 10k on the board thus making the duo 5k as well). 0.1uf and 47uf are placed across VCC/GND. Power is supplied by the PSU from the Elegoo "The Most Complete Starter Kit". USB-to-Serial is done with one of these
http://www.ebay.de/itm/272461294802?_trksid=p2060353.m1438.l2649&ssPageName=STRK%3AMEBIDX%3AIT
, connecting the TX/RX/GND pins. For programming I use the Arduino IDE 1.6.12 with the esp-library.

The problem: I get the
"
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed
"
messages. Turning up the verbosity during upload reveals these lines as well:
"
trying to connect
flush start
setting serial port timeouts to 1 ms
setting serial port timeouts to 1000 ms
flush complete
espcomm_send_command: sending command header
espcomm_send_command: sending command payload
espcomm_send_command: receiving 2013 bytes of data
read 0, requested 1
error: failed reading byte
warning: espcomm_send_command: cant receive slip payload data
"

It is the Serial Monitor that will give me an idea what might be the problem:
On 115200 Baud it shows me something like this:
"
garbage_data (This is actually the bootloader reporting stuff on 74880 Baud)
AI-Thinker something something

ready
"
On 74880 Baud it reports a bunch of stuff, but most interesting is the very first line:
ets Jan 8 2013,rst cause:2, boot mode:(3,6).

Testing reveals: When connecting GPIO00 to high for normal boot - the Serial Monitor shows exactly the same.

Oh.

Ahhhh!

No wonder I can't program the chip if it isn't even in the right mode!
The 10k resistor seems to be too much for GND to properly register at GPIO00, thus not starting the UART Mode (programming mode) but Flash Mode (normal boot) instead.

I replaced the 10k (probably, I mean, I didn't measure it, but the other ones from the same batch from China seem to be 10k) resistor between GPIO0 and GND with a 3.9k. Voila,
"
writing flash
................................................................................ [ 34% ]
................................................................................ [ 69% ]
..................................................................... [ 100% ]
starting app without reboot
" (and some more lines before and after).

Also, the proper behavior of the Serial Monitor when starting the esp in UART Mode seems to be to show only one line of garbage data on 115200 Baud which is revealed as
"
ets Jan 8 2013,rst cause:2, boot mode:(1,7)
"
on 74880 Baud. Dates might differ, of course.

j0hnsmith, 2017-04-30:
So I got exactly these error messages on a ESP8266 mounted on a dev board. To resolve the problem I had to keep the flash button pressed while flashing (output below). Here's a screenshot of [my board](https://www.amazon.co.uk/gp/product/B01GCK3J40/) with the flash button highlighted

[image]

```
Archiving built core (caching) in: /var/folders/90/ppjzg_ws355456nzj843tvgh0000gn/T/arduino_cache_494362/core/core_esp8266_esp8266_generic_CpuFrequency_80,FlashFreq_40,FlashMode_dio,UploadSpeed_115200,FlashSize_512K64,ResetMethod_ck,Debug_Disabled,DebugLevel_None_____77a656e245c3f7119ecc0a74f86b36b1.a
Sketch uses 238685 bytes (54%) of program storage space. Maximum is 434160 bytes.
Global variables use 33444 bytes (40%) of dynamic memory, leaving 48476 bytes for local variables. Maximum is 81920 bytes.
Uploading 242832 bytes from /var/folders/90/ppjzg_ws355456nzj843tvgh0000gn/T/arduino_build_265999/firmata.ino.bin to flash at 0x00000000
................................................................................ [ 33% ]
................................................................................ [ 67% ]
..............................................................................   [ 100% ]
```

WolfCreations, 2017-05-20:
For me, under Microsoft Windows, I've just changed the driver in _Ports (COM and LPT)_ from **RAMPS** to **USB-SERIAL CH340** (with update driver) and all works fine now ...

nafizb, 2017-05-28:
[image]
Just change Reset method to nodemcu.

maxalone, 2017-06-13:
@nafizb thanks, your settings solved to my ch340 esp8266 based
[image]

zivanov, 2017-06-23:
What fixed it for me was uploading the bare minimum sketch to the arduino

vikibaarathi, 2017-06-26:
I had the same problem and fixed with the following 3 steps:

- I connected D3 to GND and D8 to GND with simple jumper cables.
- I took out every single USB device connected to my machine. (Keyboard, Mouse, Monitor extension)
- Upload the Bare Minimum Sketch

devfilipe, 2017-07-23:
I had the same issue using NodeMCU Amica. I got it fixed just starting the upload process keeping FLASH button pressed. After upload starts the button can be released. No external supply. No external resistors.

On Arduino IDE (assuming package esp8266 is installed), select Board NodeMCU 1.0, baud rate 115200.

darrenolson, 2017-08-19:
I had tried all of your suggestions and what made my set up work - ensure good connections and proper VCC level

I have 55 Esp8266 to set up and all work now thanks to fixing poor wiring connections.
GPIO0 is grounded all the time, directly, not through any resistor. 

TX---------------------RX
RX---------------------TX
3.3V-------------------VCC
3.3V--------------------CH_EN
GND--------------------GND
GNO--------------------GPIO0

KaneIbrahima, 2017-09-18:
I solved mine by connecting the reset pin to the arduino's reset. I believe connecting to gnd and a button would also solve it. I restarted and it worked. I tried another program and it did not work at first but when I reset it it worked again.

devyte, 2017-10-05:
The espcomm open failed error is documented. If what the doc says doesn't work, you either have bad wiring, or a bad board.
Closing.

closed 2017-10-05 by devyte

vanzavishal95, 2017-11-25:
error: Failed to open COM1
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed

how to solve this error

Pablo2048, 2017-11-25:
Select proper COM port :-(

mflf, 2017-11-26:
i have the same problem. I have done all the above methods but problem is still occured.
error: Failed to open COM1
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed

mflf, 2017-11-26:
This is the error. plz help me fast.!
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed

Pablo2048, 2017-11-26:
Yeah, another "help me fast" guy... What did you try to solve your problem?

devyte, 2017-11-26:
As I stated above, that error is documented. If you follow the instructions and still get the error, you either have bad wiring, or a bad module, or you're doing something wrong.
If you need further assistance, please refer to a community forum, as this is an issue tracker for the core. Also, see our issue policy doc or #3655 .

dmeylman, 2017-12-23:
Thanks, nafizb 
[image]

psatheesh18, 2018-02-22:
The following error can be rectified by ,,
Connections:
**tx -->rx(esp)
rx-->tx(esp)
3.3v -->ch_pd +vcc(esp)
GND -->Gnd +Gpi0(esp)**

Better to make a **switch** for **ch_pd** pins this helps to enable **flashing** and **rectifies** the below error.

**baudrate** supports for me is **9600,57600 t**hat works for me,     

**error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed**

svatos-jirka, 2018-03-03:
As many people wrote: basic problem with Arduino UNO is, that usb programmer isnt enough strong power. You have use external power supply (3,3V and wirted GND with usb programmer)

irtesamh, 2018-03-26:
I solved this error by changing upload speed to 115200

svatos-jirka, 2018-03-27:
I had similar issue, because I used wrong programmer.
It is better to use https://www.aliexpress.com/item/OPEN-SMART-USB-to-ESP8266-ESP-01-Wi-Fi-Adapter-Module-w-CH340G-Driver/32795568611.html?spm=a2g0s.9042311.0.0.8vCLBN instead of https://www.aliexpress.com/item/ESP8266-ESP-01S-WIFI-Module-Adapter-Download-Debug-Link-Kit-for-Arduino-IDE-USB-to-ESP8266/32838195520.html?spm=a2g0s.9042311.0.0.8vCLBN. Very important is switch (UART <-> PROG)

psatheesh18, 2018-03-28:
You have no need of usb programmer to buy you can use arduino to program esp8266

Prasadgiot, 2018-04-09:
I have wasted my 4 days to check below error, but still not succeded. Tried every solution, can someone help me please

I am using ESP8266, Driver installed properly for CH340G & COM port properly reflected... Whle uploading it is throwing me error -

warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed

Need URGENT Help, someone can directly contact me on prasadgiot@gmail.com

Please HELP here...

lrmoreno007, 2018-04-09:
Google this "esp8266 arduino espcomm_sync failed" and click in "I'm feeling lucky"

Prasadgiot, 2018-04-09:
Sorry, but already browsed all these, but no luck....

Actually Serial Monitor is also throwing "ERROR" when sending command as
"AT"

I checked PORT & Baud, it is fine, not getting exact rror

Prasadgiot, 2018-04-12:
> [image: image]
> <https://user-images.githubusercontent.com/5265771/38638204-ed0742f8-3da3-11e8-8e35-110711d5734f.png>
> my config on lolin new mcu v3 using Arduino 1.8.2 and updating driver from
> RAMPS to USB-SERIAL CH340 just like @WolfCreations
> <https://github.com/WolfCreations> said.

Hi

Thanks for reply

Much Appreciate it!!!

Good News - I can able to upload the code by pressing & holding the FLUSH
button n device... It works for me

Thanks & keep supporting to me :)

shekarpathi, 2018-04-26:
Prasadgiot where is the FLUSH button?
I have an Lenoardo and trying to program the ESP8266. I continue getting the following error.

warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed

msingh-dminc, 2018-05-21:
I was getting the same error.
I pressed the nodeMCU FLASH button, while uploading the code using Arduino IDE.
It worked absolutely fine

cyyamazaki, 2018-05-21:
About above comments: NodeMCU 1.0 doesn't need Flash button to be pressed on upload. On the others models, if there is a FLASH button, it must be pressed on upload until Dots begin to apear, when the FLASH button can be released. 

If there isn't a FLASH button, GPIO00 must be connected to GND.

shinuza, 2018-05-31:
I have Lolin NodeMCU V3, I followed, [these instructions](http://www.instructables.com/id/Getting-Started-With-ESP8266LiLon-NodeMCU-V3Flashi/) and to upload use NodeMCU firmware programmer, you have to hold down the flash button and then press the reset button. That's the only way I go the NODEMCU Firmware programmer to work.

From then I was able to upload a sketch.

wirthual, 2018-06-03:
I have the same problem: I am using Ubuntu 17.10 with linux 4.13.0-37-generic kernel.

I dmesg I found this line, so I assume it has something to do with the usb driver:
ch341-uart ttyUSB0: usb_serial_generic_read_bulk_callback - urb stopped: -32

When I use Ubuntu 16.04 with the same NodeMcu 1.0, everything works for me.

kyoung90, 2018-06-20:
I feel like I've tried just about everything here and it still doesn't work. There must be something I've not seeing. I am using an arduino uno and an ESP8266 wifi module. The arduino is connected to the laptop, and I'm using 2 AA batteries as external source (they are brand new). My connection is the following:

ESP GND --> Arduino GND + AA batteries GND + GPIO 0
ESP TX --> Arduino RX
ESP RX --> Arduino TX
(have tried switching these but still nothing)
CH_PD --> ESP VCC which is also connected to the batteries
 
[image]

I have tried changing the upload speed on every change I make but still same error. I have even tried wiring the reset on the ESP to the rest of the arduino. I just don't know what else to try. 

[image]

I still keep getting the same error. 
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed

ServisOficial, 2018-07-13:
i have the same problem :(
Arduino:1.8.5 (Linux), Tarjeta:"NodeMCU 1.0 (ESP-12E Module), 80 MHz, 4M (1M SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

Archiving built core (caching) in: /tmp/arduino_cache_148097/core/core_esp8266_esp8266_nodemcuv2_CpuFrequency_80,FlashSize_4M1M,LwIPVariant_v2mss536,Debug_Disabled,DebugLevel_None____,FlashErase_none,UploadSpeed_115200_183312692e25616796233f7b6af8a20d.a
El Sketch usa 246259 bytes (23%) del espacio de almacenamiento de programa. El máximo es 1044464 bytes.
Las variables Globales usan 32224 bytes (39%) de la memoria dinámica, dejando 49696 bytes para las variables locales. El máximo es 81920 bytes.
/home/necorom/.arduino15/packages/esp8266/tools/esptool/0.4.13/esptool -vv -cd nodemcu -cb 115200 -cp /dev/ttyUSB0 -ca 0x00000 -cf /tmp/arduino_build_449254/Blink.ino.bin 
esptool v0.4.13 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to nodemcu
	setting baudrate from 115200 to 115200
	setting port from /dev/ttyUSB0 to /dev/ttyUSB0
	setting address from 0x00000000 to 0x00000000
	espcomm_upload_file
	espcomm_upload_mem
opening port /dev/ttyUSB0 at 115200
	tcgetattr
	tcsetattr
	serial open
opening bootloader
resetting board
trying to connect
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	serialport_receive_C0: 08 instead of C0
trying to connect
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	serialport_receive_C0: 08 instead of C0
trying to connect
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
resetting board
trying to connect
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
resetting board
trying to connect
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed

Este reporte podría tener más información con
"Mostrar salida detallada durante la compilación"
opción habilitada en Archivo -> Preferencias.
[image]

ServisOficial, 2018-07-13:
I have the same problem :(

Arduino:1.8.5 (Linux), Tarjeta:"NodeMCU 1.0 (ESP-12E Module), 80 MHz, 4M (1M SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

Archiving built core (caching) in: /tmp/arduino_cache_148097/core/core_esp8266_esp8266_nodemcuv2_CpuFrequency_80,FlashSize_4M1M,LwIPVariant_v2mss536,Debug_Disabled,DebugLevel_None____,FlashErase_none,UploadSpeed_115200_183312692e25616796233f7b6af8a20d.a
El Sketch usa 246259 bytes (23%) del espacio de almacenamiento de programa. El máximo es 1044464 bytes.
Las variables Globales usan 32224 bytes (39%) de la memoria dinámica, dejando 49696 bytes para las variables locales. El máximo es 81920 bytes.
/home/servisoficial/.arduino15/packages/esp8266/tools/esptool/0.4.13/esptool -vv -cd nodemcu -cb 115200 -cp /dev/ttyUSB0 -ca 0x00000 -cf /tmp/arduino_build_449254/Blink.ino.bin 
esptool v0.4.13 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to nodemcu
	setting baudrate from 115200 to 115200
	setting port from /dev/ttyUSB0 to /dev/ttyUSB0
	setting address from 0x00000000 to 0x00000000
	espcomm_upload_file
	espcomm_upload_mem
opening port /dev/ttyUSB0 at 115200
	tcgetattr
	tcsetattr
	serial open
opening bootloader
resetting board
trying to connect
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	serialport_receive_C0: 08 instead of C0
trying to connect
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	serialport_receive_C0: 08 instead of C0
trying to connect
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
resetting board
trying to connect
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
resetting board
trying to connect
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed

Este reporte podría tener más información con
"Mostrar salida detallada durante la compilación"
opción habilitada en Archivo -> Preferencias.

[image]

lrmoreno007, 2018-07-13:
Read this instructions:
https://arduino-esp8266.readthedocs.io/en/latest/faq/a01-espcomm_sync-failed.html

In spanish (no official):
http://esp8266-arduino-spanish.readthedocs.io/es/latest/faq/a01-espcomm_sync-failed.html

luxarts, 2018-09-21:
For those looking for a solution, I had he same problem and I solved it with the following config:

Hardware:
- Button between GPIO0 and GND
- 10k resistor between EN and VCC
- 10k resistor between GPIO15 and GND

Software (Arduino):
- Board: Generic ESP8266 module
- Flash mode: QIO
- Flash size: 4M (1M SPIFFS) <- This may vary for your proyect/module
- Reset method: none
- Upload Speed: 115200

Procedure:
1) Keep pressing the button (GPIO0 will be connected to GND)
2) Connect the module to the power supply (VCC to 3.3v - GND to GND)
3) Press Upload in Arduino IDE
4) When the uploading start (you will see dots filling the console window) you can leave the button

FelipeTesta, 2018-11-27:
> Try to connect The GPIO0 with a 10k resistor to ground and the CH_PD to VCC and it works :)

Worked! Thanks.

daknadnane, 2019-03-13:
I  encountered the same issue using MCUNode8266  (failed reading byte )  , the solution is pretty simple , push flash button when uploading file

prajakta2, 2019-03-22:
C:\Users\LENOVO\Documents\Arduino\libraries\TelegramBot-Library-master\src\TelegramBot.cpp:39:11: error: StaticJsonBuffer is a class from ArduinoJson 5. Please see arduinojson.org/upgrade to learn how to upgrade your program to ArduinoJson version 6

    StaticJsonBuffer<JSON_BUFF_SIZE> jsonBuffer;

           ^

C:\Users\LENOVO\Documents\Arduino\libraries\TelegramBot-Library-master\src\TelegramBot.cpp:87:11: error: StaticJsonBuffer is a class from ArduinoJson 5. Please see arduinojson.org/upgrade to learn how to upgrade your program to ArduinoJson version 6

   StaticJsonBuffer<JSON_BUFF_SIZE> jsonBuffer;

           ^

C:\Users\LENOVO\Documents\Arduino\libraries\TelegramBot-Library-master\src\TelegramBot.cpp:102:11: error: StaticJsonBuffer is a class from ArduinoJson 5. Please see arduinojson.org/upgrade to learn how to upgrade your program to ArduinoJson version 6

   StaticJsonBuffer<JSON_BUFF_SIZE> jsonBuffer;

           ^

C:\Users\LENOVO\Documents\Arduino\libraries\TelegramBot-Library-master\src\TelegramBot.cpp: In member function 'message TelegramBot::getUpdates()':

C:\Users\LENOVO\Documents\Arduino\libraries\TelegramBot-Library-master\src\TelegramBot.cpp:39:20: error: expected primary-expression before '<' token

    StaticJsonBuffer<JSON_BUFF_SIZE> jsonBuffer;

                    ^

C:\Users\LENOVO\Documents\Arduino\libraries\TelegramBot-Library-master\src\TelegramBot.cpp:39:37: error: 'jsonBuffer' was not declared in this scope

    StaticJsonBuffer<JSON_BUFF_SIZE> jsonBuffer;

                                     ^

C:\Users\LENOVO\Documents\Arduino\libraries\TelegramBot-Library-master\src\TelegramBot.cpp:44:12: error: 'ArduinoJson::JsonObject' has no member named 'success'

    if(root.success()){

            ^

C:\Users\LENOVO\Documents\Arduino\libraries\TelegramBot-Library-master\src\TelegramBot.cpp: In member function 'String TelegramBot::sendMessage(String, String)':

C:\Users\LENOVO\Documents\Arduino\libraries\TelegramBot-Library-master\src\TelegramBot.cpp:87:19: error: expected primary-expression before '<' token

   StaticJsonBuffer<JSON_BUFF_SIZE> jsonBuffer;

                   ^

C:\Users\LENOVO\Documents\Arduino\libraries\TelegramBot-Library-master\src\TelegramBot.cpp:87:36: error: 'jsonBuffer' was not declared in this scope

   StaticJsonBuffer<JSON_BUFF_SIZE> jsonBuffer;

                                    ^

C:\Users\LENOVO\Documents\Arduino\libraries\TelegramBot-Library-master\src\TelegramBot.cpp:93:8: error: 'ArduinoJson::JsonObject' has no member named 'printTo'

   buff.printTo(msg);

        ^

C:\Users\LENOVO\Documents\Arduino\libraries\TelegramBot-Library-master\src\TelegramBot.cpp: In member function 'String TelegramBot::sendMessage(String, String, TelegramKeyboard&, bool, bool)':

C:\Users\LENOVO\Documents\Arduino\libraries\TelegramBot-Library-master\src\TelegramBot.cpp:102:19: error: expected primary-expression before '<' token

   StaticJsonBuffer<JSON_BUFF_SIZE> jsonBuffer;

                   ^

C:\Users\LENOVO\Documents\Arduino\libraries\TelegramBot-Library-master\src\TelegramBot.cpp:102:36: error: 'jsonBuffer' was not declared in this scope

   StaticJsonBuffer<JSON_BUFF_SIZE> jsonBuffer;

[... 75 lines cut ...]

C:\Users\LENOVO\Documents\Arduino\libraries\TelegramBot-Library-master\src\TelegramBot.cpp:81:2: warning: control reaches end of non-void function [-Wreturn-type]

  }

  ^

C:\Users\LENOVO\Documents\Arduino\libraries\TelegramBot-Library-master\src\TelegramBot.cpp: In member function 'String TelegramBot::sendMessage(String, String)':

C:\Users\LENOVO\Documents\Arduino\libraries\TelegramBot-Library-master\src\TelegramBot.cpp:98:1: warning: control reaches end of non-void function [-Wreturn-type]

 }

 ^

Multiple libraries were found for "TelegramBot.h"
 Used: C:\Users\LENOVO\Documents\Arduino\libraries\TelegramBot-Library-master
 Not used: C:\Users\LENOVO\Documents\Arduino\libraries\TelegramBot-1.2.5
Using library TelegramBot-Library-master at version 1.2.5 in folder: C:\Users\LENOVO\Documents\Arduino\libraries\TelegramBot-Library-master 
Using library ArduinoJson-6.x at version 6.9.1 in folder: C:\Users\LENOVO\Documents\Arduino\libraries\ArduinoJson-6.x 
Using library ESP8266WiFi at version 1.0 in folder: C:\Users\LENOVO\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\libraries\ESP8266WiFi 
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

Please help me solve this errors

mcolinj, 2019-03-25:
> reply_markup.set("one_time_keyboard", one_time_keyboard);
> reply_markup.set("resize_keyboard", resize_keyboard);
> reply_markup.set("selective", false);

I ran into this problem a number of times already.     It seems the ArduinoJson library changes
from 5 to 6 are not compatible.   Whatever you are trying to build is *using* version 5, and you
have version 6 installed.

Use the library manager to load the 5.x version of ArduinoJson and you should be fine.

good luck,

matthew

furtnerthomas, 2019-05-09:
i ran into the same problem. the solution in my case was to put a 100uF capacitor between power supply (3.3 V) and ground.

3PACK, 2019-05-16:
I have the same problem , i try every forum solution but no success !
I try your tips here to and i start to wonder me , this is a cheap board maybe the solder  of the wifi module are faulty , so i bring my iron and resolder it and add some more solder ..and now works like a charm no resistor no jumper cables.. just the usb cable  !

Aibana17, 2020-03-23:
I have de same error but my board is: UNO+WiFi R3 ATmega328P+ESP8266, 32Mb flash, USB-TTL CH340G, Micro-USB
I try every solutions in the internet... no success...
I am give up. Help please!

daknadnane, 2020-03-23:
Hi , Try to push the reset button the moment when you load the code