esp8266/Arduino #2802: Why isn't logical operator short circuit evaluation working?

JamesNewton, 2016-12-28:
### Basic Infos
I'm flabergasted... short circuit evaluation of logic operators doesn't appear to be working... I must be missing something here... 

#### Hardware
Hardware:			ESP-12E
Core Version:      	?2.1.0-rc2?

#### Software
IDE: 1.6.9
ESP 2.3.0

### Description

Short circuit evaluation is a mandatory part of C/C++... 34 || 1 can't be 1. It has to be 34.

http://www.open-std.org/JTC1/SC22/WG14/www/docs/n1256.pdf
C99 spec, section 6.5.14 Logical OR operator says

> (4) Unlike the bitwise | operator, the || operator guarantees left-to-right evaluation; there is a sequence point after the evaluation of the first operand. If the first operand compares unequal to 0, the second operand is not evaluated.

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		512K (64K SPIFFS)
CPU Frequency:		80Mhz
Flash Mode:			DIO
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		nodemcu

### Sketch

```cpp

#include <Arduino.h>

#define BAUD_RATE 38400

void setup() {
  Serial.begin(BAUD_RATE);
  Serial.println( 0 || 1); // expect 1
  Serial.println( 34 || 1); // expect 34
  Serial.println( ((String)"34").toInt() || 1); // expect 34
}

void loop() {

}
```

### Debug Messages

```
load 0x4010f000, len 1384 ...
1
1
1

```

igrr, 2016-12-28:
Suggest trying `Serial.println(34 || 2);` — this may give a hint ;)

JamesNewton, 2016-12-28:
No, || is a logical not a bitwise or. I did not write 34 | 1, I wrote 34 || 1.

igrr, 2016-12-28:
Okay, seemingly it didn't give you a hint. You are missing the fact that implicit type conversions are happening:
http://ideone.com/4FfeNX

JamesNewton, 2016-12-28:
:facepalm: Everything is getting converted to bool. So you can't pull that trick at all. 

This is what happens when a programmer has worked in many different languages and starts mashing them all together in his head. In e.g. Perl, or JavaScript, or LUA, or Python the "|| defaultValue" pattern works nicely, e.g.
http://docstore.mik.ua/orelly/perl3/prog/ch03_14.htm
http://its-a-kind-of-magic.com/2016/09/02/expression-evaluation-and-short-circuiting-tricks/

And C/C++ do have short circuit evaluation, but... I totally forgot that they type convert everything to 0 or 1 boolean values. 

Thanks for setting me straight.

closed 2016-12-28 by JamesNewton