esp8266/Arduino #2803: Handling HTTPS requests on ESPWebServer

Devesh24, 2016-12-28:
Hi Guys,
               I am using the ESPWebServer example from the master branch, and i want to add TLS support onto it, i have gone through some codes and forum but it seems TLS integration with Client only, I want to add it on my ESPWebServer , i have no idea how to get started with it.

Thanks

devyte, 2017-10-05:
@Devesh24 I don't know what the current status for a secure server is, there are a lot of issues open for secure sockets. However, latest git is supposed to support TLS1.2, so I suggest looking at that.
It has been a long time since you opened this, have you made any progress? If you don't intend to work on it, please close the issue.

devyte, 2017-12-05:
Closing due to lack of feedback in over 2 months, and due to age.

closed 2017-12-05 by devyte