esp8266/Arduino #2805: Have support for TLS 1.2 in SDK v2.0?

yuridiniz, 2016-12-28:
Hi, i see some posts on the internet saying that the new version of the SDK (ESP8266_NONOS_SDK_V2.0.0_16_07_19) supports TLS 1.2, but in the official documentation says it only has support for TLS 1.1

https://github.com/esp8266/Arduino/blob/master/doc/esp8266wifi/readme.md#client-secure

Thanks

partofthething, 2016-12-29:
In #2537 (35ee060) it looks like the axTLS library was upgraded to a version supporting TLS 1.2. I think the documentation might be behind.

bantaj, 2017-01-05:
Curious if you were able to confirm support for TLS 1.2 by getting it to work?

tedder, 2017-04-03:
With version 2.3.0 installed, using fancyssl.hboeck.de (which is tls1.2-only), it appears TLS1.2 is NOT supported. @igrr, does this make sense? It appears it should work, per the 2.3.0 release notes and #2537.

Further, the release notes give two git hashes for axTLS binary inclusions. The 2.3.0 tag has the older release hash, not [the newer one](https://github.com/esp8266/Arduino/blame/2.3.0/doc/changes.md#L44):
https://github.com/esp8266/Arduino/blob/2.3.0/tools/sdk/lib/libaxtls.a

[The commit of the updated binary](https://github.com/esp8266/Arduino/commit/c2414a2252d14afaf29860b11664a1d64e0e8dc1) appears to be on master, not in the 2.3.0 tag. I don't know where the git hashes of the axtls binary come from, so it's hard to be certain that's the problem.

FWIW, using 'fancyssl' means SNI works, which was the other addition that should be in axtls2.0.

jfktrey, 2017-05-15:
Did you configure the TLS version appropriately when attempting to connect? https://github.com/micropython/micropython/issues/2343

tedder, 2017-05-15:
Hmm, no. Looks like it needs to be `ssl_version: TODO`. But what are the constants? I don't understand how it gets mapped from C to Python; the two values I see in C are `SL_SO_SEC_METHOD_TLSV1_2` or `MP_QSTR_PROTOCOL_TLSv1_2`.

igrr, 2017-05-16:
TLS1.2 is supported in git version, to be released in 2.4.0.

zdewitt, 2017-10-24:
Hey all, just curious if anyone was able to get TLS 1.2 working with the git version? I'm trying to connect to AWS IoT and it seems to keep rejecting the connection at the SSL stage. I've successfully connected to the AWS endpoint using openssl on my laptop and the same certificates/key (they are DER versions) but I keep getting this on the debug output:
```Connected, IP address: 192.168.1.23
SPIFFSImpl: allocating 512+240+1400=2152 bytes
SPIFFSImpl: mounting fs @100000, size=2fb000, block=2000, page=100
SPIFFSImpl: mount rc=0
:ref 1
please start sntp first !
State:	sending Client Hello (1)
:wr 129 129 0
:wrc 129 129 0
:sent 129
:rn 1460
:rch 1460, 1460
:rch 2920, 119
:rd 5, 3039, 0
:rdi 1460, 5
:rd 3034, 3039, 5
:rdi 1455, 1455
:c 1455, 1460, 3039
:rdi 1460, 1460
:c 1460, 1460, 1579
:rdi 119, 119
:c0 119, 119
State:	receiving Server Hello (2)
State:	receiving Certificate (11)
State:	receiving Certificate Request (13)
State:	receiving Server Hello Done (14)
State:	sending Certificate (11)
:wr 877 877 0
:wrc 256 877 0
:wrc 256 621 0
:wrc 256 365 0
:wrc 109 109 0
:sent 877
State:	sending Client Key Exchange (16)
:wr 267 267 0
:wrc 256 267 0
:wrc 11 11 0
:sent 267
State:	sending Certificate Verify (15)
:wr 269 269 0
:wrc 256 269 0
:wrc 13 13 0
:wr 6 6 0
:wrc 6 6 0
State:	sending Finished (16)
:wr 85 85 0
:wrc 85 85 0
:wr 69 69 0
:wrc 69 69 0
Alert: close notify
Secure connection to host failed!
SPIFFS_close: fd=3
SPIFFS_close: fd=2
SPIFFS_close: fd=1
:sent 269
:sent 160
:rn 160
:rcl
:abort
```
A snippet of my code for this connection test is here:
```SPIFFS.begin();
File CACertFile = SPIFFS.open("/CA-certificate.der.crt", "r");
File certFile = SPIFFS.open("/fb24dcba28-certificate.der.crt", "r");
File pkFile = SPIFFS.open("/fb24dcba28-private.der.key", "r");

if (!wifiClient.loadCertificate(certFile)) Serial.println("certFile failed!");
if (!wifiClient.loadCACert(CACertFile)) Serial.println("CACert failed!");
if (!wifiClient.loadPrivateKey(pkFile)) Serial.println("pkFile failed!");

wifiClient.connect(HOST_NAME, 8883);
if (wifiClient.connected())
{
  Serial.println("Established secure connection to host.");
} else {
  Serial.println("Secure connection to host failed!");
}
```
I don't know if it means anything, but it takes a while (a couple seconds) after `State:	sending Finished (16)` before the `Alert: close notify` message comes up. It seems as though it's trying to negotiate but unable to. Is there any way to get any more verbose messages out of the TLS system? I don't fully understand the comment above about micropython, is there something I need to configure to use v1.2?

closed 2018-01-02 by igrr

tedder, 2018-01-02:
Closed sans comment, @igrr?

igrr, 2018-01-02:
I have commented above: https://github.com/esp8266/Arduino/issues/2805#issuecomment-301657669