esp8266/Arduino #2813: String(negativeInteger, HEX) doesn't return the same thing than with an Arduino UNO.

pierrepoulpe, 2016-12-29:
### Basic Infos

#### Hardware
Hardware:			NodeMCU ESP-12E
Core Version:      	2.3.0

### Description

String(negativeInteger, HEX) doesn't return the same thing than with an Arduino UNO.

### Settings in IDE

Module:  			NodeMCU 1.0
Flash Size: 		4MB/3MB
CPU Frequency:		80Mhz
Upload Using:		SERIAL
Reset Method:		nodemcu

### Sketch
```
long a;

void setup() {
  // put your setup code here, to run once:
  a = -100;
  delay(500);
  Serial.begin(9600);
  Serial.println(String(a,HEX));
  

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

### Serial Output
`-64`

### Exact same code, uploaded to UNO
`ffffff9c`

Strange, isn't it? I supposed String library was platform independant... I was wrong.

Uno version sounds better to me....

devyte, 2017-10-05:
@pierrepoulpe is this still an issue in latest git? I think the correct output should be what you get for the UNO.

earlephilhower, 2019-01-16:
This is a bug.

Our ltoa doesn't behave correctly for negative, non base10 #s.  It should be safe to replace the shim functions in `core_esp8266_noniso.c` with calls to `itoa` and `utoa` since `long` and `int` are the same size.

earlephilhower, 2019-01-21:
Fixed with #5625

closed 2019-01-21 by earlephilhower