esp8266/Arduino #2817: Reading SSID and Password from EEPROM to use in WiFi.begin() - does not work

MisterBE2, 2016-12-31:
### Basic Infos
I'm trying to read SSID and password to connect with Wi-Fi. Everything seems fine but ESP can't connect. I don't know why.

### Description

I had written little debugger here is what it outcomes:

	Starting check
		Value of byte at adress 1 =10

	Connecting to Wi-Fi
		Value of byte at adress 2 =16
		Readed ssid =MY SSID
		Value of byte at adress 3 =12
		Readed password =MY PASSWORD
	Connecting
		 ...............................	Could not connect

	Starting Access Point
		Value of byte at adress 4 =8
		Readed name =MRJW-ESP

AP IP address: 192.168.4.1
Port: 23
Max users: 1
New client connected - ID: 0
	Sending: Host: You are connected!

### Settings in IDE

Module: ESP8622MOD - lolin nodemcu v3
Flash Size: 4MB
CPU Frequency:	80Mhz
Flash Mode:	?
Flash Frequency:	?
Upload Using:	SERIAL
Reset Method:	nodemcu

### Sketch

```cpp

// -------------- Zalaczone biblioteki --------------

#include <ESP8266WiFi.h> // Biblioteka odpowiedzialna za obsługe wi-fi
#include <WiFiClient.h> // Biblioteka odpowiedzialna za Acces Pointa
#include <EEPROM.h> // Obsluguje pamiec EEPROM
#include <ESP8266WebServer.h> // Biblioteka odpowiedzialna za odpowiedzi na wezwania serwera

// -------------- Podstawowe dane --------------

#define defName "ESP-HUB"// Domyslna nazwa urzadzenia
#define defPort 23 // Domyslny port urzadzenia
#define maxClients 1 // Ile urzadzen na raz moze obslurzyc esp - na razie nie moznqa obluzyc wiecej niz 1
#define acceptPin 2 // Pin D4 pozwalajacy na akceptacje wybranego trybu przez resetPin
#define resetPin 5 // Pin D1 pozwalajacy na restart lub reset urzadzenia
#define LED 12 // Pin do ktorego podlaczony jest LED informacyjny
#define rst 13 // pin do którego podpięty jest reset

// -------------- Globalne dane --------------

const char* command; // Odebrane komendy z Telent
char eRead[25];
byte len;
bool byteRead;
byte tray =0;
bool success = true;

// -------------- Functions --------------

// Saves string do EEPROM
void SaveString(int startAt, const char* id)
{
  for (byte i = 0; i <= strlen(id); i++)
  {
    EEPROM.write(i + startAt, (uint8_t) id[i]);
  }
  EEPROM.commit();
}

// Reads string from EEPROM
void ReadString(byte startAt, byte bufor)
{
  for (byte i = 0; i <= bufor; i++)
  {
    eRead[i] = (char)EEPROM.read(i + startAt);
  }
  len = bufor;
}

//Saves byte to EEPROM
void SaveByte(int startAt, byte val)
{
  EEPROM.write(startAt, val);
  EEPROM.commit();
}

//Reads byte from EEPROM
byte ReadByte(byte startAt)
{
  byte Read = -1;
  Read = EEPROM.read(startAt);
  return Read;
}

// WI-Fi
WiFiServer server(23); // Sets communication port
WiFiClient serverClients[maxClients]; // Sets clients ammount

void setup()
{
  EEPROM.begin(512);
  Serial.begin(115200);
  delay(500);

/*
  Serial.println("\tWriting Data:");
  SaveByte(1, 10);
  Serial.println("\t\tSaved byte 1 to value of 10");
  SaveByte(2, 16);
  Serial.println("\t\tSaved byte 2 to value of 16");
  SaveString(5, "MY SSID");
  Serial.println("\t\tSaved string MY SSID to address 5 ");
  SaveByte(3, 12);
  Serial.println("\t\tSaved byte 3 to value of 12");
  SaveString(31, "MY PASSWORD");
  Serial.println("\t\tSaved string MY PASSWORD to address 31");
  SaveByte(4, 8);
  Serial.println("\t\tSaved byte 4 to value of 8");
  SaveString(57, "MRJW-ESP");
  Serial.println("\t\tSaved string MRJW-ESP to address 57");
  */

  delay(1000);
  Serial.println("\n\tStarting check");
  byte x = -2;
  x = ReadByte(1);
  String tempR;

  Serial.print("\t\tValue of byte at adress 1 ="); Serial.println(x);
  if (x == 10)
  {
    Serial.println("\n\tConnecting to Wi-Fi");
    x = ReadByte(2);
    Serial.print("\t\tValue of byte at adress 2 ="); Serial.println(x);
    const char* ssid;
    ReadString(5, x);
    tempR = "";
    for (byte i = 0; i < len; i++)
    {
      tempR += eRead[i];
    }
    ssid = tempR.c_str();
    Serial.print("\t\tReaded ssid ="); Serial.println(ssid);
    x = ReadByte(3);
    Serial.print("\t\tValue of byte at adress 3 ="); Serial.println(x);
    const char* password;
    ReadString(31, x);
    tempR = "";
    for (byte i = 0; i < len; i++)
    {
      tempR += eRead[i];
    }
    password = tempR.c_str();
    Serial.print("\t\tReaded password ="); Serial.println(password);

    WiFi.begin(ssid, password);
    Serial.println("\tConnecting");
    Serial.print("\t\t ");
    while (WiFi.status() != WL_CONNECTED) {
      delay(1000);
      Serial.print(".");
      tray++;
      if(tray > 30)
      {
        success = false;
        Serial.println("\tCould not connect");
        break;
      }
    }
    if(success)
    {
    Serial.println("\n\tConnected!");
    server.begin();
    server.setNoDelay(true);
    }
  }
  else
  {
    WiFi.disconnect();
    Serial.println("\n\tStarting Access Point");
    x = ReadByte(4);
    Serial.print("\t\tValue of byte at adress 4 ="); Serial.println(x);
    const char* Name;
    ReadString(57, x);
    tempR = "";
    for (byte i = 0; i < len; i++)
    {
      tempR += eRead[i];
    }
    Name = tempR.c_str();
    Serial.print("\t\tReaded name ="); Serial.println(Name);
    WiFi.softAP(Name, "123456789"); // Ustawia nazwe urzadzenia
    server.begin();
    server.setNoDelay(true);
  }

  if(!success)
  {
   WiFi.disconnect();
        Serial.println("\n\tStarting Access Point");
    x = ReadByte(4);
    Serial.print("\t\tValue of byte at adress 4 ="); Serial.println(x);
    const char* Name;
    ReadString(57, x);
    tempR = "";
    for (byte i = 0; i < len; i++)
    {
      tempR += eRead[i];
    }
    Name = tempR.c_str();
    Serial.print("\t\tReaded name ="); Serial.println(Name);
    WiFi.softAP(Name, "123456789"); // Ustawia nazwe urzadzenia
    server.begin();
    server.setNoDelay(true);
  }

  Serial.print("\nAP IP address: "); Serial.println(WiFi.softAPIP());
  Serial.print("Port: "); Serial.println(defPort);
  Serial.print("Max users: "); Serial.println(maxClients);
}

```

MisterBE2, 2016-12-31:
Manual reading from EEPROM helped.

closed 2016-12-31 by MisterBE2

rezaajdarkosh, 2017-09-08:
Hi MisterBE2!
change your Code with this one:

[code]
// -------------- Libraries --------------

#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <EEPROM.h>
#include <ESP8266WebServer.h>

//-------------- Main data --------------

#define defPort 1234 // What port
#define maxClients 1 // how many cilents

// -------------- Global --------------

const char* command;
char eRead[25];
byte len;
bool byteRead;
byte tray = 0;
bool success = true;

// -------------- Functions --------------

// Saves string do EEPROM
void SaveString(int startAt, const char* id)
{
  for (byte i = 0; i <= strlen(id); i++)
  {
    EEPROM.write(i + startAt, (uint8_t) id[i]);
  }
  EEPROM.commit();
}

// Reads string from EEPROM
void ReadString(byte startAt, byte bufor)
{
  for (byte i = 0; i <= bufor; i++)
  {
    eRead[i] = (char)EEPROM.read(i + startAt);
  }
  len = bufor;
}

//Saves byte to EEPROM
void SaveByte(int startAt, byte val)
{
  EEPROM.write(startAt, val);
  EEPROM.commit();
}

//Reads byte from EEPROM
byte ReadByte(byte startAt)
{
  byte Read = -1;
  Read = EEPROM.read(startAt);
  return Read;
}

// WI-Fi
WiFiServer server(1234); // Sets communication port
WiFiClient serverClients[maxClients]; // Sets clients ammount

void setup()
{
  EEPROM.begin(512);
  Serial.begin(115200);
  delay(500);

  /*
    Serial.println("\tWriting Data Begin...");
    SaveByte(1, 10);
    SaveByte(2, strlen(ssid)); // Length of ssid
    SaveString(5, ssid);
    SaveByte(3, strlen(password)); // Length of password
    SaveString(31, password);
    SaveByte(4, strlen(Soft_AP); // Length of SoftAP ssid
    SaveString(57, Soft_AP);
    Serial.println("\tWriting Data Complated!");
  */

  delay(1000);
  Serial.println("\n\tStarting check");
  byte x = -2;
  x = ReadByte(1);

  Serial.print("\t\tValue of byte at adress 1 ="); Serial.println(x);
  if (x == 10)
  {
    Serial.println("\n\tConnecting to Wi-Fi");
    x = ReadByte(2);
    Serial.print("\t\tValue of byte at adress 2 ="); Serial.println(x);
    const char* ssid;
    ReadString(5, x);
    String tempS = "";
    for (byte i = 0; i < len; i++)
    {
      tempS += eRead[i];
    }
    ssid = tempS.c_str();
    Serial.print("\t\tReaded ssid ="); Serial.println(ssid);
    x = ReadByte(3);
    Serial.print("\t\tValue of byte at adress 3 ="); Serial.println(x);
    const char* password;
    ReadString(31, x);
    String tempP = "";
    for (byte i = 0; i < len; i++)
    {
      tempP += eRead[i];
    }
    password = tempP.c_str();
    Serial.print("\t\tReaded password ="); Serial.println(password);
    WiFi.disconnect();
    WiFi.begin(ssid, password);
    Serial.print("\nConnecting to ");
    Serial.print(ssid);
    while (WiFi.status() != WL_CONNECTED) {
      delay(1000);
      Serial.print(".");
      tray++;
      if (tray > 30)
      {
        success = false;
        Serial.println("\n\tCould not connect");
        break;
      }
    }
    if (success)
    {
      Serial.println("\n\tConnected!");
      server.begin();
      server.setNoDelay(true);
      Serial.print("\nAP IP address: "); Serial.println(WiFi.localIP());
      Serial.print("Port: "); Serial.println(defPort);
      Serial.print("Max users: "); Serial.println(maxClients);
    }
  }
  if (x != 10 || !success)
  {
    WiFi.disconnect();
    Serial.println("\n\tStarting Access Point");
    x = ReadByte(4);
    Serial.print("\t\tValue of byte at adress 4 ="); Serial.println(x);
    const char* Name;
    ReadString(57, x);
    String tempN = "";
    for (byte i = 0; i < len; i++)
    {
      tempN += eRead[i];
    }
    Name = tempN.c_str();
    Serial.print("\t\tReaded name ="); Serial.println(Name);
    WiFi.softAP(Name, "123456789"); // Ustawia nazwe urzadzenia
    server.begin();
    server.setNoDelay(true);
    Serial.print("\nAP IP address: "); Serial.println(WiFi.softAPIP());
    Serial.print("Port: "); Serial.println(80);
    Serial.print("Max users: "); Serial.println(maxClients);
  }
}

void loop() {
}

[/code]