esp8266/Arduino #2818: Telent server Exception(3) & (9)

MisterBE2, 2016-12-31:
### Basic Infos
I've got this error while connecting to Wi-Fi. I'm a bit confused, because moments ago it was working. But now no matter what I do it crashes.

#### Hardware

Module: ESP8266MOD - lolin nodemcu v3
Flash Size: 4MB
CPU Frequency:	80Mhz
Upload Using:	SERIAL
Reset Method:	nodemcu

### Description

Exception (3):
epc1=0x4010010e epc2=0x00000000 epc3=0x00000000 excvaddr=0x4006ff01 depc=0x00000000

ctx: cont 
sp: 3ffef4b0 end: 3ffef700 offset: 01a0

>>>stack>>>
3ffef650:  3ffee5a4 00000001 3ffee6a4 402038bd  
3ffef660:  3ffee684 0000fffc 0000fffc 4010020c  
3ffef670:  3ffee5a4 3ffee578 3ffef6a0 4010068c  
3ffef680:  3ffee5a4 3ffee578 3ffee6a4 40203b64  
3ffef690:  4954454e 3ffee578 3ffee6a4 402023a4  
3ffef6a0:  3ffeff00 0000000f 00000000 feefeffe  
3ffef6b0:  feefeffe feefeffe feefeffe feefeffe  
3ffef6c0:  3ffef680 3ffef6a0 0000000c feefeffe  
3ffef6d0:  feefeffe feefeffe feefeffe 3ffee6d0  
3ffef6e0:  3fffdad0 00000000 3ffee6c8 40203efc  
3ffef6f0:  feefeffe feefeffe 3ffee6e0 40100718  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09f0c112
~ld

It looks like program crasches when it's entering void loop(){}

###Sketch

,,,cpp

// -------------- Libraries --------------

#include <ESP8266WiFi.h> // Biblioteka odpowiedzialna za obsługe wi-fi
#include <WiFiClient.h> // Biblioteka odpowiedzialna za Acces Pointa
#include <EEPROM.h> // Obsluguje pamiec EEPROM
#include <ESP8266WebServer.h> // Biblioteka odpowiedzialna za odpowiedzi na wezwania serwera

// -------------- Main Data --------------

#define maxClients 1
#define rst 13
// -------------- Global Data --------------

const char* command;
int port = 23;
byte len = -1;
bool byteRead = -1;
byte tray = 0;
bool success = true;

// -------------- Functions--------------

[... 325 lines cut ...]
      if (serverClients[i].available())
      {
        Mryg();
        String temp = "";
        while (serverClients[i].available())
        {
          char c = serverClients[i].read();
          if (c == '\n')
          {
            command = temp.c_str();
            delay(2);
            menu();
            break;
          }
          else
          {
            temp += c;
          }
          delay(10);
        }
      }
    }
  }
}
,,,

MisterBE2, 2017-01-02:
@igrr please look at These. Now all the sudden all of my previous working version of this code also do not work. I do not know what to do :/

devyte, 2017-10-05:
@MisterBE2 you are storing strings in the EEPROM (which by itseld is not a very good idea, you should use the SPIFFS instead), and your loop for that is based on strlen. That means you don't store the null-terminator. Then, when you read the strings back, I don't see a null terminator either.
Most likely, you have garbage in your buffs, and then when trying to print, you crash due to the missing null terminator.
In any case, this is not the right place to ask for assistance for your particular project.
Closing as off-topic and language error, see #3655 .

closed 2017-10-05 by devyte