esp8266/Arduino #2820: BUG: The SPI library does not handle HW SC for other bit lengths than 8.

tclarsson, 2017-01-02:
BUG: The SPI library does not handle HW SC for other bit lengths than 8.
For example: There is a bug in the routine "transfer16".

Hardware:			WEMOS ESP8266

If one uses transfer16 for 16 bit SPI transfer using HW CS, the SS/CS pin goes high after each 8 bit transfer. the 16bit word is therefore transferred as 2 x 8 bit bytes.

For the below to work:
  SPI.setHwCs(true);  //Use (automatic) HW SS/CS  (GPIO 15 , D8)
  SPI.setBitOrder(MSBFIRST);
  SPI.transfer16(0x1234);

I had to write my own alternative "_transfer16" routine:

uint16_t _bs(uint16_t data) {
  return(((data&0xFF)<<8)|((data>>8)&0xFF));
}

void _setDataBits(uint16_t bits) {
    const uint32_t mask = ~((SPIMMOSI << SPILMOSI) | (SPIMMISO << SPILMISO));
    bits--;
    SPI1U1 = ((SPI1U1 & mask) | ((bits << SPILMOSI) | (bits << SPILMISO)));
}

uint16_t _transfer16(uint16_t data) {
    if(!(SPI1C & (SPICWBO | SPICRBO))) data=_bs(data);
    while(SPI1CMD & SPIBUSY) {}
    _setDataBits(16);
    SPI1W0 = data;
    SPI1CMD |= SPIBUSY;
    while(SPI1CMD & SPIBUSY) {}
    data=SPI1W0;
    if(!(SPI1C & (SPICWBO | SPICRBO))) data=_bs(data);
    return (data);
}

devyte, 2017-10-04:
@tclarsson is this issue still valid in latest git?

UlliBien, 2019-08-12:
The bug is still there (Arduino/ESP8266, version 2.5.2). write16() does it correctly. I am not familiar with github and unfortunately can not make the change myself.

Change code to:
uint16_t SPIClass::transfer16(uint16_t data) {
  while (SPI1CMD & SPIBUSY) {}
  // Set to 16Bits transfer
  setDataBits(16);
  // Change byte order for MSBFIRST
  if (!(SPI1C & (SPICWBO | SPICRBO))){
    // MSBFIRST Byte first
    SPI1W0 = (data >> 8) | (data << 8);
  }
  else {
    // LSBFIRST Byte first
    SPI1W0 = data;
  }
  // Start the transmission
  SPI1CMD |= SPIBUSY;
  // Wait for transmission to complete
  while (SPI1CMD & SPIBUSY) {}
  data = SPI1W0;
    // Change byte order for MSBFIRST
  if (!(SPI1C & (SPICWBO | SPICRBO)))
    return (data >> 8) | (data << 8);
  else
    return data;
}

thangktran, 2019-08-13:
@UlliBien I'll be working on this issue