esp8266/Arduino #2822: Arduino IDE

fredericplante, 2017-01-02:
You should really consider using other IDE then the Arduino IDE, make one or Use other one. In fact you should dissociate ESP8266 From Arduino all together.

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.md

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.md

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			?ESP-12?
Core Version:      	?2.1.0-rc2?

### Description

Problem description

### Settings in IDE

Module:  			?Generic ESP8266 Module?
Flash Size: 		?4MB/1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
messages here
```

vicnevicne, 2017-01-02:
Err, you do realize you are opening an issue to "dissociate ESP8266 From Arduino" on a project whose rationale is to "bring support for ESP8266 chip to the Arduino environment", right ?
It's a bit like posting on a cooking forum that people should go paragliding instead :-).
Seriously, there are many other ways to program and use the ESP8266: plain C using Espressif's native SDK,  Lua, Javascript, and many more. If I'm not mistaken, the Arduino environment is one of the latest in fact. 
However, the fact is that the learning curve for the Arduino IDE is almost flat and that's why it's so popular,... but feel free not to use it of course :-)

fredericplante, 2017-01-02:
They are other IDE that does the same, and that are happy to have new user in their forum. Unlike arduino, their are other platform where the people are not arrogante and ignoble to new user.

If you feel like being treated like shit by racists, well no problem, have fun in the arduino world my friend.

fredericplante, 2017-01-02:
[image]

Of coarse, when I pointed out that this comment is racist I got my IP ban for 10 day. So I created and still do, many account, to get all the IP of my internet service providers ban, so they will hurt them self, cause. We are now many to do it, so Soon no one in Québec will be able to reach their site.

Note that Mister Mike is a bretaine member what ever this mean, he can ban people.

Arduino does not deserve to be able to reach my children, since they are Racist, fascist, and very bad model

This being sayed, wish you a good day mate

closed 2017-01-03 by fredericplante