esp8266/Arduino #2823: Can I connect to home modem from esp8266 ?

jyotirajsharma, 2017-01-03:
Hi,

Can I connect from esp8266 to home modem using ssid and password directly?

Or do we need to configure wireless router separately ?

jyotirajsharma, 2017-01-03:
Also can i use mobile hotspot as ad-hoc so that esp8266 connects to it ?

MisterBE2, 2017-01-03:
No need for configuration.
ESP connects to any Wi-Fi suorce.

You just need to do this:

```cpp

#include <ESP8266WiFi.h>

const char* ssid = "Your SSID";
const char* pass = "Your password";

void setup()
{
Serial.begin(115200);
WiFi.begin(ssid, pass);

uint8_t i = 0;
    while (WiFi.status() != WL_CONNECTED && i++ < 20)
    {
      Serial.print(".");
      delay(500);
    }

    if (i == 21) {
      Serial.print("\nCould not connect to"); Serial.println(ssid);
    }
    else
    {
      Serial.println("\nConnected!");
      Serial.print("Network IP address: "); Serial.println(WiFi.localIP());
    }

}

```

devyte, 2017-10-04:
Closing as off-topic, see #3655 .

closed 2017-10-04 by devyte