esp8266/Arduino #2825: Should esp8266 be flashed with USB TTL cable and external power supply and not with Arduino ?

jyotirajsharma, 2017-01-03:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.md

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.md

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			?ESP-12?
Core Version:      	?2.1.0-rc2?

### Description

Problem description

### Settings in IDE

Module:  			?Generic ESP8266 Module?
Flash Size: 		?4MB/1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
messages here
```

vicnevicne, 2017-01-03:
You asked similar questions in issues #2801 #2772 and #1471 . That doesn't help developers and won't help you in the end. 
May I kindly suggest you to ask this question too on the esp8266.com community forums, in the "ESP8266 Arduino" section.

jyotirajsharma, 2017-01-03:
OK..sure.

closed 2017-01-03 by jyotirajsharma