esp8266/Arduino #2826: How to set host name?

MisterBE2, 2017-01-03:
### Basic Infos
Is there any command which sets ESP host name?
I mean name which is displayed on router.

Right now it is something like this:

[Router panel screen](http://www.mediafire.com/convkey/9f46/f5sc7nnvfg6lp2xzg.jpg)

liquidfalcon, 2017-01-03:
Yes, if you use the search facility of GitHub to search this repository, you will find the code and examples on how to set it.

MisterBE2, 2017-01-03:
I found this:

wifi_station_set_hostname( "Test" );

But it does not work :/

MisterBE2, 2017-01-04:
After more digging i found it:

WiFi.hostname("Name");

anitech-embedded-systems, 2017-03-11:
It worked for me 👍

rhdw, 2017-07-22:
Can you inform me where is the right Position of   `WiFi.hostname("Name");`   in sourcecode ?
I use the following libraries and code, but it does not work
```
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
...
    WiFi.mode(WIFI_STA);
    WiFi.begin(ssid, passwort);
    if (WiFi.status() == WL_CONNECTED) {
      WiFi.hostname("WebPool");
      ...

```

mcmillion, 2017-07-22:
You need to call `WiFi.hostname()` before `WiFi.begin()`

devyte, 2017-07-22:
Since we're on this topic, what is the correct way to change the hostname
later on? Would the wifi need to be restarted or something?

kdbqlb, 2017-07-28:
Doesn t work

petely, 2017-08-12:
Thank you.
This works:
`
Serial.begin(115200);
  delay(10);
  Serial.print("Connecting to ");
  Serial.println(ssid);

  WiFi.hostname("relay");
  WiFi.begin(ssid, password);
 
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("WiFi connected");
 
  // Start the server
  server.begin();
  Serial.println("Server started");
 
`

**Serial monitor log:**
Connecting to astro-net
..
WiFi connected
Server started
Use this URL to connect: http://192.168.1.54/

**from Linux box on the same subnet:**
corv-root: ping 192.168.1.54
PING 192.168.1.54 (192.168.1.54) 56(84) bytes of data.
64 bytes from 192.168.1.54: icmp_seq=1 ttl=128 time=6.47 ms
64 bytes from 192.168.1.54: icmp_seq=2 ttl=128 time=1.98 ms
64 bytes from 192.168.1.54: icmp_seq=3 ttl=128 time=1.92 ms

**and then ...**
corv-root: ping relay
PING relay (192.168.1.54) 56(84) bytes of data.
64 bytes from relay (192.168.1.54): icmp_seq=1 ttl=128 time=2.66 ms
64 bytes from relay (192.168.1.54): icmp_seq=2 ttl=128 time=2.53 ms
64 bytes from relay (192.168.1.54): icmp_seq=3 ttl=128 time=1.89 ms

kdbqlb, 2017-08-13:
And how do you See the Hostname ???

Thanks
Klaus-Dieter 🚴🏊🏼⛵️⛷🎾😊

BRMateus2, 2017-08-30:
You can see in the router or by nslookup <ip>. There seems to be no way of pushing in the hostname to the code, other than settings yourself an array of char and WiFi.hostname(array)

By the way, this might help lots of people who needs more detailed WiFi report after connect:
```
#include <ESP8266WiFi.h>
void setup(){
	setup_connect_to_WiFi(); //Call for the WiFi Connection
}

void setup_connect_to_WiFi(){
	unsigned int WiFi_connect_delay_try = 500; //Delay of the retry to connect to WiFi
	WiFi.hostname(WiFi_hostname);
	WiFi.mode(WIFI_STA); //Indicate to act as wifi_client only, defaults to act as both a wifi_client and an access-point.
	WiFi.begin(WiFi_SSID, WiFi_password); //Try to connect now
	Serial.print("Connecting to \"" + String(WiFi_SSID) + "\" with " + String(WiFi_connect_delay_try) + "ms interval with mode WIFI_STA"); Serial.printf("%s", WiFi.mode(WIFI_STA) ? "" : ", mode failed!!");
	while(WiFi.status() != WL_CONNECTED){
		/*Note: if connection is established, and then lost for some reason, ESP will automatically reconnect. This will be done automatically by Wi-Fi library, without any user intervention.*/
		delay(WiFi_connect_delay_try);
		Serial.print(".");
	}
	Serial.print("\nConnected to " + String(WiFi_SSID) + " with IP Address: "); Serial.print(WiFi.localIP()); Serial.print("\n");
}
```

devyte, 2017-10-04:
WiFi.hostname() is currently working in latest git. Closing as resolved.

closed 2017-10-04 by devyte

lbottero, 2017-10-20:
/* in some cases, (I had this problem) need to put after WiFi.begin, this MDNS.begin("hostname"), dontt forget to include this #include <ESP8266mDNS.h>
This works:
*/
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>
...
    WiFi.mode(WIFI_STA);
    WiFi.hostname("WebPool");
    WiFi.begin(ssid, passwort);
    MDNS.begin("WebPool");
    if (WiFi.status() == WL_CONNECTED) {
      
      ...

brianjmurrell, 2018-01-26:
```
  WiFi.hostname("foobar");
  WiFi.begin(ssid, password);
```

does not seem to be setting the name here.  It still uses it's `ESP_XXXXXX` when requesting a DHCP lease and also uses `esp8266-XXXXXX.local` in the mDNS name.

I'm not sure I'm using version 2.4.0 of esp8266/Arduino (yet) but not sure how to find that out.  I'd also be happy to upgrade to 2.4.0 if I am not, but not sure how to do that either.

lucaspiller, 2018-02-12:
I'm also seeing the same behaviour @brianjmurrell. I'm using PlatformIO, platform espressif8266 1.6.0, which includes the 2.4.0 release of this project. My code is as follows:

```
WiFi.hostname("fan");                                                                                                                                                                                                                                    
WiFi.mode(WIFI_STA);                                                                                                                                                                                                                                     
WiFi.begin(WIFI_SSID, WIFI_PASSWORD);  
```

lrmoreno007, 2018-02-12:
Try to restart your router.

brianjmurrell, 2018-02-12:
> Try to restart your router.

Why?  What does my router have to do with setting the hostname on my ESP?

lucaspiller, 2018-02-12:
@lrmoreno007 Well that is embarrassing, but yes it worked. I'm not sure why though, as I am able to change the hostname of virtual machines without having to do that. I guess the DHCP implementation in the Espressif SDK isn't 100% spec compliant. Thanks!

lrmoreno007, 2018-02-12:
It's not embarrassing. Depending on the configuration of your network, the router can make the DNS server function, was a possibility and that is why I said that you should try it.

d-a-v, 2018-02-13:
WiFi.hostname is broken in 2.4.0 with lwip-v2. Please update to latest git or use lwip-v1.4.

tokra, 2018-03-30:
Not working for me:
```c
void wifiSetup() {
  char* WiFi_hostname = "ESP8266-2";
  WiFi.config(ip, dns, gateway, subnet_mask);
  WiFi.hostname(WiFi_hostname);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  Serial.println("");

  // Wait for connection
  while (WiFi.status() != WL_CONNECTED) {
    delay(10);
    Serial.print(".");
  }
  Serial.println("");
  Serial.print("Connected to : ");
  Serial.println(ssid);
  Serial.print("IP address   : ");
  Serial.println(WiFi.localIP());
  Serial.print("GW address   : ");
  Serial.println(WiFi.gatewayIP());
  Serial.print("MAC address  : ");
  Serial.println(WiFi.macAddress());
  if (mdns.begin(WiFi_hostname)) {
    Serial.println("mDNS responder started !");
  }
}
```

BRMateus2, 2018-03-30:
You should not set config if you are not using an static IP routing; your router has DHCP activated? If yes, some routers simply block any data from static users.

mindquanta, 2018-06-16:
It worked for me...

void setup(void){
  pinMode(led, OUTPUT);
  digitalWrite(led, 0);
  **Serial.begin(115200);
  WiFi.mode(WIFI_STA);
  WiFi.hostname("IotONE");
  WiFi.begin(ssid, password);**
  Serial.println("");

  // Wait for connection
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");

duncan-a, 2018-06-16:
Using hypen ("-") in a hostname seems to cause problems - it shouldn't, but
it did for me...

rayvt, 2018-06-18:
It doesn't work right.  Sometimes it gets the name I set in WiFi.hostname(), and sometimes it gets the "ESP_xxx" name.

manulsan, 2018-11-27:
Above comments are all right.

Setting host name works at LwIP V 1.4 Above
Set as following way

  Sketch menu ==> Tool ==> LwIP Variant xxxxx==> "V1.4 Higher Bandwidth"

Then it will work.

[image]

Yonni20140812, 2019-05-03:
If doesn't work, try wash flash, may help

ewaldc, 2019-08-01:
I tried many of the posted suggestions, but none of them worked for me using 2.5.2 or 2.6_dev.
So far the only code sequence that worked reliably is:
```
 wifi_station_set_hostname(myhostname);
 WiFi.begin(ssid, pwd);
 DbgPrintln("hostname::", WiFi.hostname());
```
If you put certain Wifi commands in between the setting of the hostname and WiFi.begin (e.g. WiFi.enableSTA(true)) it no longer works and the hostname becomes again the default "ESP_<id>".

It also does not work for soft AP mode.
```
wifi_station_set_hostname(myhostname);
WiFi.softAP(APssid, pwd);
DbgPrintln("hostname::", WiFi.hostname());
```

produces an empty hostname.  Adding multiple WiFi.hostname(myhostname) or wifi_station_set_hostname(myhostname) statements before or after WiFi.softAP, does not change the result.  Nor does adding MDNS.begin(myhostname);   Another question is of course: does it really matter if the hostname is not correctly set?  In station mode, clients can use the name provided by the router's DNS (or DHCP) service, while for AP mode, mDNS can be used.

BRMateus2, 2019-08-01:
> I tried many of the posted suggestions, but none of them worked for me using 2.5.2 or 2.6_dev.
> produces an empty hostname. Adding multiple WiFi.hostname(myhostname) or wifi_station_set_hostname(myhostname) statements before or after WiFi.softAP, does not change the result. Nor does adding MDNS.begin(myhostname); Another question is of course: does it really matter if the hostname is not correctly set? In station mode, clients can use the name provided by the router's DNS (or DHCP) service, while for AP mode, mDNS can be used.

You might have found a pretty bug, I cannot test what you meant because it's been some 7 months since I stopped with Arduino for now - it might be important to open a new ticked with your code, if those functions in the [documentation](https://arduino-esp8266.readthedocs.io/en/latest/esp8266wifi/station-class.html?highlight=hostname#hostname) really don't work.

About the hostname, it does not matter if you set it or don't set, other than being able to call the device by name as it is probably automagically linked to a DNS server - and even so, there are some routers like MikroTik, which do not link the DHCP leases to the DNS cache, making the device hostname useless outside the router management interface.

d-a-v, 2019-08-01:
Without MCVE we can't help.
The one in OP works:
```
#include <ESP8266WiFi.h>
void setup() {
  Serial.begin(115200);
  WiFi.hostname("moon");
  WiFi.begin(STASSID, STAPSK);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println(WiFi.hostname());
}

void loop () {
}
```
console:
```
19:37:42.546 -> SDK:2.2.2-dev(38a443e)/Core:2.5.2-102-g8ee720b9=20502102/lwIP:STABLE-2_1_2_RELEASE/glue:1.1-8-g2314329/BearSSL:89454af
19:37:42.546 -> wifi evt: 2
19:37:42.613 -> scandone
19:37:43.109 -> .....scandone
19:37:46.422 -> state: 0 -> 2 (b0)
19:37:46.422 -> .state: 2 -> 3 (0)
19:37:46.422 -> state: 3 -> 5 (10)
19:37:46.422 -> add 0
19:37:46.422 -> aid 1
19:37:46.422 -> cnt 
19:37:46.455 -> 
19:37:46.455 -> connected with xxxxxx, channel 1
19:37:46.455 -> dhcp client start...
19:37:46.455 -> wifi evt: 0
19:37:46.488 -> ip:10.0.1.166,mask:255.255.255.0,gw:10.0.1.254
19:37:46.488 -> wifi evt: 3
19:37:46.919 -> .
19:37:46.919 -> WiFi connected
19:37:46.919 -> moon
19:37:56.429 -> pm open,type:2 0
```
on openwrt router:
```
root@xxxxx:~# ping moon
PING relay (10.0.1.166): 56 data bytes
64 bytes from 10.0.1.166: seq=0 ttl=128 time=1.903 ms
```

Closing

M1DLG, 2020-05-10:
> Another question is of course: does it really matter if the hostname is not correctly set?

It does matter if you are setting up a cluster of sensors and each have their own ID and name. It gets hard to track them by random strings when an actual name is easy to read and display.

jordanadania, 2023-01-23:
WiFi.config does not allow setting a custom hostname. At least not in the router control panel list of attached devices.

govego, 2023-07-15:
Try after connected to router, first  connect to esp8266 to SSID router after change hostname:

  while (WiFi.status() != WL_CONNECTED)
{
  Serial.printf("Default hostname: %s\n", WiFi.hostname());  // Current Hostname
  WiFi.hostname("SuMotoru");  //Set new hostname
  delay(2000);
  Serial.printf("New hostname: %s\n", WiFi.hostname());// new Hostname
..........................