esp8266/Arduino #2829: rst cause:4,boot mode(1,6)

sarwadenj, 2017-01-04:
problem coming after uploading program....
 
rst cause:4,boot mode(1,6)

GPIO's status after uploading code
CH_PD--VCC
GPIO 15--GND
GPIO 0-Connected to DC MOTOR via N-MOSFET
GPIO 2-Connected to interrupt pin for sensor.....

previously it is not giving problem with this setup.....

devyte, 2017-10-04:
@sarwadenj no sketch, no basic info, no IDE settings.
Closing due to lack of info, see #3655 . If the problem still exists, please open a new issue, and provide the details requested in the issue template.

closed 2017-10-04 by devyte

adavidzh, 2017-12-11:
I just started getting exactly the same thing after flashing:
```
1 30 03 03 22 a3 ff c0 22 01 30 03[...removed for brevity...]
 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset
```
It happens with any sketch, including
```
void setup() {
  // put your setup code here, to run once:
}
void loop() {
  // put your main code here, to run repeatedly:
}
```
I'm using the Arduino IDE 1.8.5, and `Generic ESP8266 Module` on a Ledunia. This Ledunia was working perfectly. I looked through #1013 without success.

If I reset by shorting `RES` to `GND`, I then get an infinite loop of
```
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v3de0c112
~ld
[...byte code...]
```
This seemed like #2414, but I am powering from the same place as when it was working (a USB port).

The whole thing started after uploading a sketch that was setting pin 13 as output. Could that be the reason? Any thoughts are welcome.

Dynamite-7, 2018-04-05:
Hey everyone..
I'm getting this in serial monitor of arduino... please suggest some solutions, I'm unable to check my desired output.

Exception (9):
epc1=0x402027ae epc2=0x00000000 epc3=0x00000000 excvaddr=0x0000001f depc=0x00000000

ctx: cont
sp: 3ffefe40 end: 3fff0090 offset: 01a0

stack>>>
3ffeffe0: 0000004b 00000050 3ffeed48 402027a8
3ffefff0: 3ffe8cbc 3ffeefa0 3ffe8cb8 a0640534
3fff0000: 3ffe8994 3ffeefa0 0000004b 402032cc
3fff0010: 40104e5a 3ffeefa0 3ffeed48 402021cd
3fff0020: 00000000 00000000 00000000 402032a1
3fff0030: 3ffe8cbc 3ffeed6c 3fff15b4 0000003f
3fff0040: 00000038 3ffef060 3ffeefa0 402032cc
3fff0050: 3fff068c 40202578 3ffeefa0 3ffef060
3fff0060: 3fffdad0 00000000 3ffef058 402022c3
3fff0070: 3fffdad0 00000000 3ffef058 402038a0
3fff0080: feefeffe feefeffe 3ffef070 40100710
<<<stack<<<

ets Jan 8 2013,rst cause:2, boot mode:(1,6)

ets Jan 8 2013,rst cause:4, boot mode:(1,6)

wdt reset

Please help ASAP..!!!

Thanks.

adavidzh, 2018-05-20:
For some reason, after a few weeks of not touching the board, I no longer have the problem.

dangsonbk, 2018-07-29:
I got same error when my code looped in while(1);. Seem that the board was reset by watchdog.

not4unco, 2019-02-07:
I'm having the same issue with my NodeMCU...

09:48:45.503 -> Sdk version: 3.0.0-dev(c0f7b44)
09:48:45.503 -> Core Version: 2_5_0
09:48:45.503 -> Boot Version: 31
09:48:45.503 -> Boot Mode: 1
09:48:45.503 -> CPU Frequency: 80 MHz

```
11:19:04.627 -> Soft WDT reset
11:19:04.627 -> 
11:19:04.627 -> >>>stack>>>
11:19:04.627 -> 
11:19:04.627 -> ctx: sys
11:19:04.627 -> sp: 3fffed40 end: 3fffffb0 offset: 01b0
11:19:04.627 -> 3fffeef0:  3ffe8d00 64e525a5 60000600 60000600  
11:19:04.627 -> 3fffef00:  00000000 3ffed318 00000000 00000000  
11:19:04.627 -> 3fffef10:  4022d51c 3ffed318 3ffee2d0 60000600  
11:19:04.627 -> 3fffef20:  4022fe65 3ffed318 3ffee2d0 447e425a  
11:19:04.627 -> 3fffef30:  4022feaa 3fffdab0 00000000 3fffdcb0  
11:19:04.906 -> 3fffef40:  3ffee2f8 3fffdab0 00000000 4020644b  
11:19:04.906 -> 3fffef50:  40000f49 40000f49 3fffdab0 40000f49  
11:19:04.906 -> 3fffef60:  40000e19 40001878 00000002 00000000  
11:19:04.906 -> 3fffef70:  3fffff10 ffffffff ffffffff 40105694  
11:19:04.906 -> 3fffef80:  4010569a 00000002 00000000 173243cc  
11:19:04.906 -> 3fffef90:  4010000d 5797346d 62d64d98 f57879d7  
11:19:04.906 -> 3fffefa0:  40101868 3fffef3c 40101815 3fffff48  
11:19:04.906 -> 3fffefb0:  3fffffc0 00000000 00000000 feefeffe  
11:19:04.906 -> 3fffefc0:  feefeffe feefeffe feefeffe feefeffe  
11:19:04.906 -> 3fffefd0:  feefeffe feefeffe feefeffe feefeffe  
11:19:04.906 -> 3fffefe0:  feefeffe feefeffe feefeffe feefeffe  
11:19:04.906 -> 3fffeff0:  feefeffe feefeffe feefeffe feefeffe  
11:19:04.906 -> 3ffff000:  feefeffe feefeffe feefeffe feefeffe  
11:19:04.906 -> 3ffff010:  feefeffe feefeffe feefeffe feefeffe  
11:19:04.906 -> 3ffff020:  feefeffe feefeffe feefeffe feefeffe  
11:19:04.906 -> 3ffff030:  feefeffe feefeffe feefeffe feefeffe  
11:19:04.906 -> 3ffff040:  feefeffe feefeffe feefeffe feefeffe  
11:19:04.906 -> 3ffff050:  feefeffe feefeffe feefeffe feefeffe  
11:19:04.906 -> 3ffff060:  feefeffe feefeffe feefeffe feefeffe  
11:19:04.906 -> 3ffff070:  feefeffe feefeffe feefeffe feefeffe  
11:19:04.906 -> 3ffff080:  feefeffe feefeffe feefeffe feefeffe  
11:19:04.906 -> 3ffff090:  feefeffe feefeffe feefeffe feefeffe  
11:19:04.906 -> 3ffff0a0:  feefeffe feefeffe feefeffe feefeffe  
11:19:04.906 -> 3ffff0b0:  feefeffe feefeffe feefeffe feefeffe  
11:19:04.906 -> 3ffff0c0:  feefeffe feefeffe feefeffe feefeffe  
11:19:04.906 -> 3ffff0d0:  feefeffe feefeffe feefeffe feefeffe  
11:19:04.906 -> 3ffff0e0:  feefeffe feefeffe feefeffe feefeffe  
11:19:04.906 -> 3ffff0f0:  feefeffe feefeffe feefeffe feefeffe  
11:19:04.906 -> 3ffff100:  feefeffe feefeffe feefeffe feefeffe  
11:19:04.906 -> 3ffff110:  feefeffe feefeffe feefeffe feefeffe  
11:19:04.906 -> 3ffff120:  feefeffe feefeffe feefeffe feefeffe  
11:19:04.906 -> 3ffff130:  feefeffe feefeffe feefeffe feefeffe  
11:19:04.906 -> 3ffff140:  feefeffe feefeffe feefeffe feefeffe  
11:19:04.906 -> 3ffff150:  feefeffe feefeffe feefeffe feefeffe  
11:19:04.906 -> 3ffff160:  feefeffe feefeffe feefeffe feefeffe  
11:19:04.906 -> 3ffff170:  feefeffe feefeffe feefeffe feefeffe  
11:19:04.906 -> 3ffff180:  feefeffe feefeffe feefeffe feefeffe  
11:19:04.906 -> 3ffff190:  feefeffe feefeffe feefeffe feefeffe  
11:19:04.906 -> 3ffff1a0:  feefeffe feefeffe feefeffe feefeffe  
11:19:04.906 -> 3ffff1b0:  feefeffe feefeffe feefeffe feefeffe  
11:19:04.906 -> 3ffff1c0:  feefeffe feefeffe feefeffe feefeffe  
11:19:04.906 -> 3ffff1d0:  feefeffe feefeffe feefeffe feefeffe  
11:19:04.906 -> 3ffff1e0:  feefeffe feefeffe feefeffe feefeffe  
11:19:04.906 -> 3ffff1f0:  feefeffe feefeffe feefeffe feefeffe  
11:19:04.906 -> 3ffff200:  feefeffe feefeffe feefeffe feefeffe  
11:19:04.906 -> 3ffff210:  feefeffe feefeffe feefeffe feefeffe  
11:19:04.906 -> 3ffff220:  feefeffe feefeffe feefeffe feefeffe  
11:19:04.906 -> 3ffff230:  feefeffe feefeffe feefeffe feefeffe  
11:19:04.906 -> 3ffff240:  feefeffe feefeffe feefeffe feefeffe  
11:19:04.906 -> 3ffff250:  feefeffe feefeffe feefeffe feefeffe  
[... 197 lines cut ...]
11:19:05.917 -> 3ffffeb0:  00000027 00000001 3ffee9e4 3fff0118  
11:19:05.917 -> 3ffffec0:  00000051 3ffee928 3ffee9e1 40205390  
11:19:05.917 -> 3ffffed0:  00000059 3ffee928 3ffee9d0 402053b8  
11:19:05.917 -> 3ffffee0:  40100cec 00000051 3ffee928 40204ee5  
11:19:05.917 -> 3ffffef0:  00000001 3ffee928 00000051 40204f42  
11:19:05.917 -> 3fffff00:  00000001 3ffee928 00000051 40204f6c  
11:19:05.917 -> 3fffff10:  00000000 00000650 3ffee928 40204fa6  
11:19:05.917 -> 3fffff20:  3ffee928 00000003 3fff0117 40204fc3  
11:19:05.917 -> 3fffff30:  40105fb0 64fb6275 3ffeed2c 00000000  
11:19:05.917 -> 3fffff40:  3ffee2d0 3ffeed2c 3ffe8590 3ffeed2c  
11:19:05.917 -> 3fffff50:  3fffdad0 3ffeec54 40206490 3fffefa0  
11:19:05.917 -> 3fffff60:  3ffeed2c 3fffdad0 00001388 402019f7  
11:19:05.917 -> 3fffff70:  3fffdad0 00001388 3ffee93c 40203d93  
11:19:05.917 -> 3fffff80:  00000000 00000000 00000000 00000000  
11:19:05.917 -> 3fffff90:  00000000 00000000 00000001 402064b1  
11:19:05.917 -> 3fffffa0:  3fffdad0 00000000 3ffeec24 4020653c  
11:19:05.917 -> <<<stack<<<
11:19:05.917 -> 
11:19:05.917 ->  ets Jan  8 2013,rst cause:2, boot mode:(1,6)
11:19:05.917 -> 
11:19:10.020 -> 
11:19:10.020 ->  ets Jan  8 2013,rst cause:4, boot mode:(1,6)
11:19:10.020 -> 
11:19:10.020 -> wdt reset
```

SkomisLTU, 2019-05-09:
cause:4, boot mode:(1,6)

I got this when used DS18B20 sensor with U8g2lib.h display. :(

faelpinho, 2020-01-03:
Have no solution?
I got only this message when GPIO_0 is GND:
" ets Jan  8 2013,rst cause:2, boot mode:(3,6)

ets_main.c "

Nothing more. 
And I got this message when GPIO_0 is VCC:
" ets Jan  8 2013,rst cause:2, boot mode:(1,7)"
Nothing more.

I have clicked "Erase" button on ESP8266 Download Tool V.3.6.8, and the prompt shows this message:
CONNECT BAUD: 115200
==========
.Uploading stub...
Running stub...
Stub running...
------------------
crc_efuse_4bit: 0
crc_calc_4bit: 8
48bit mac
"

[image]

Bighoneypot, 2021-05-11:
@Archy404 very well!!!